openbmc/openbmc #3593: Firmware command line update

Niubbo, 2019-08-09:
Hi, i'm looking for a simple command line tool to update the BMC and host PNOR firmware from the host command line
example: ./update-op-firmware -b <bmc-image>
               ./update-op-firmware -p <pnor-image>

Hoping you could work asap on this feature because i think it's a must have for a 2019 system

Thank you

madscientist159, 2019-08-09:
This is a feature request we have seen a number of times now, and I think it's reasonable to provide a host-based (not out of band  / remote system based) mechanism for BMC and PNOR firmware update.

Ideally an intermediate library would be made available that could then be tied into a basic command line tool, a GUI application, and automated firmware update tools (we would contribute code for the latter two).  The intermediate library would deal with the (Redfish?) API communication to the BMC from the host side.

geissonator, 2019-08-09:
We do have https://github.com/openbmc/openbmc-tools/blob/master/thalerj/openbmctool.py which has a --flash option for flashing bmc and pnor firmware. It currently uses OpenBMC custom REST api's but will eventually be moved over to Redfish based interfaces. This could be run from the host but it's all over out of band network to the BMC (i.e. not inband). 

@pstrinkle has been working on an inband tool over IPMI for firmware update.

madscientist159, 2019-08-09:
@geissonator The inband IPMI update sounds like exactly what we want.  The idea is to be able to update a system that doesn't even have the BMC on the network.

pstrinkle, 2019-08-09:
@madscientist159 See github.com/openbmc/phosphor-ipmi-flash

closed 2020-03-09 by stale[bot]