openbmc/openbmc #3594: SOL can not work in ibm-romules recipe

dzg0528, 2019-08-12:
can not open /dev/ttyS*..can not set baud..

dzg0528, 2019-08-12:
root@romulus:/tmp# cat /dev/ttyS2 
cat: read error: Input/output error

geissonator, 2019-08-12:
@mine260309 any known issues here?

mine260309, 2019-08-13:
@dzg0528 Please provide enough information about the issue, and what you are trying to do.

On Romulus, `ttyS2` is not enabled at all. You could refer to it's kernel dts:

https://github.com/openbmc/linux/blob/615c30d03b52e27b3eee2345a58ecdece6faa2dc/arch/arm/boot/dts/aspeed-bmc-opp-romulus.dts#L139-L152

Be noted that `uart1` becomes `ttyS0`:
```
1e787000.serial: ttyS5 at MMIO 0x1e787000 (irq = 33, base_baud = 1546875) is a 16550A
1e783000.serial: ttyS0 at MMIO 0x1e783000 (irq = 31, base_baud = 1500000) is a 16550A
1e784000.serial: ttyS4 at MMIO 0x1e784000 (irq = 32, base_baud = 1500000) is a 16550A
```

So there is no way for you to use `/dev/ttyS2`, unless you enable it in dts.

dzg0528, 2019-08-13:
@mine260309 
Hi
thanks for your help first.
I enable /dev/ttyS2 in dts,but is can not make SOL work.
Because the SOL feature can not work on my intel board,so I ask the question.
and I have some question bellow.
1,the SOL is OK or not in your board?
2,I modify ttyS0 to ttyS2 in the meta-romulus/recipes-phosphor/console/obmc-console$ cat obmc-console.conf,and use CMD "busctl call xyz.openbmc_project.console /xyz/openbmc_project/console xyz.openbmc_project.console setBaudRate u 115200" set baud rate.but the SOL alse can not work..
3,I want any other file need to modify?like config AST2500 register?
4,I find the configuration Volatile Bit Rate is not apply in SOL_cmds.cpp.

mine260309, 2019-08-13:
> 1,the SOL is OK or not in your board?

Yes.

> 2,I modify ttyS0 to ttyS2 in the meta-romulus/recipes-phosphor/console/obmc-console$ cat obmc-console.conf,and use CMD "busctl call xyz.openbmc_project.console /xyz/openbmc_project/console xyz.openbmc_project.console setBaudRate u 115200" set baud rate.but the SOL alse can not work..

Be noted that SOL feature has nothting to do with ttyS0 or ttyS2, it is using `ttyVUART0`. The `obmc-console.conf` only tells obmc-console-server to mirror the SOL to a physical UART, in Romulus' case, it's ttyS0.

> 3,I want any other file need to modify?like config AST2500 register?
> 4,I find the configuration Volatile Bit Rate is not apply in SOL_cmds.cpp.

The suggestion is to check if SOL works by `obmc-console-client` on BMC or `ssh -p 2200 $bmc` from your terminal.
And be noted that SOL requires the host to send console data to VUART, Romulus is a OpenPOWER system instead of Intel's x86 system, the host is different.
It's better to ask such questions in mailing list.

dzg0528, 2019-08-27:
because the 0001-aspeed-Disable-unnecessary-features.patch not work in project.so SOL can not work normal.

closed 2019-08-27 by dzg0528