openbmc/openbmc #3595: Feature Request: Extend OpenBMC to allow NVRAM writes with tunable parameters for configuration of minimum fan PWM

ToxicDragon, 2019-08-13:
As is currently the problem with Raptor Computing Systems BLACKBIRD motherboard (c.f. https://www.raptorcs.com/content/BK1MB1/intro.html), certain fan parameters need to be specified before compilation of the BMC. See the following file for zones definitions: https://git.raptorcs.com/git/talos-openbmc/tree/meta-rcs/meta-talos/recipes-phosphor/fans/phosphor-fan-control-zone-config-native/zones.yaml

However, this situation is kind of unsatisfactory, because these settings may only apply to specific fans. As a test, I used different fans from Noctua which apparently responded differently to the specified default configuration of the BMC.

Therefore, I may request a feature to have some kind of "Controls" / tunable parameters that may be set while the BMC is running (so no need for recompilation).

I consider this request in general for the start of a discussion. Would be nice, if this feature were to be implemented in the not so far future :)

Best regards,
Christian

ToxicDragon, 2019-08-13:
Even better would be some kind of auto-calibration that depending on the utilized fan (just dreaming ;) )...

madscientist159, 2019-08-13:
I see this in several phases:
1.) Adding some kind of runtime-configurable persistent parameter store to OpenBMC (this may already be in place, but it's hard to tell in the tangle of DBUS services)
2.) Extending the IPMI interface to allow parameter discovery and authenticated parameter read/write (restrict write to IPMI superuser) along with a separate parameter reload IPMI command
3.) Adding a configuration menu to Petitboot to expose these controls in a user-friendly manner

bradbishop, 2019-08-14:
> I see this in several phases:
> 1.) Adding some kind of runtime-configurable persistent parameter store  
> to OpenBMC

phosphor-settings-manager probably works for this.  That gets you the  
parameter on dbus, persisted, but you'd still need to wire up a user  
interface (OEM IPMI or redfish).

> (this may already be in place, but it's hard to tell in the tangle of  
> DBUS services)

We definitely need some kind of map to help navigate all the code out there  
:-)

msbarth, 2019-08-14:
There are a few ways that I've considered to enable tuning/overriding parameters within fan control on a live BMC such as converting the yaml over to json and having that as input to the executable, implementing fan control to host even parameters on a set of dbus interfaces for each zone, etc.. 

I'd be interested to see your design proposal on how to handle this.

An initial issue was opened awhile ago for tracking something similar to what you describe with a few general requirements I noted.  https://github.com/ibm-openbmc/dev/issues/115

ToxicDragon, 2020-02-13:
Still needed :)

ToxicDragon, 2020-08-15:
Still needed :)

msbarth, 2020-08-17:
FWIW, phosphor-fan-control is in the works of being converted to use runtime JSON configuration files. This would allow them to be changed/overridden directly on the BMC without recompiling. https://github.com/ibm-openbmc/dev/issues/1363

em-winterschon, 2021-04-01:
Looking forward to this being implemented. Is there an approximate ETA?

msbarth, 2021-04-01:
I've been working on this as time permits, but it should all be in review by June

edtanous, 2024-07-25:
This was heavily discussed on the mailing list at the time, and runtime support of fan control configuration parameters has been supported by phosphor-pid-control for quite a while.  They are writable through bmcwebs Redfish OpenBMC manager schema.

Closing as this is already supported.

closed 2024-07-25 by edtanous

williamspatrick, 2024-07-25:
> runtime support of fan control configuration parameters has been supported by phosphor-pid-control for quite a while

Does phosphor-pid-control persist those?

(Not that I think we should reopen this in either case.)

edtanous, 2024-07-25:
Entity-manager persists the settings.