openbmc/openbmc #3596: about ip is invalid

linuxchien, 2019-08-14:
We first boot to linux but the ip is invalid, if we re-try udhcpc is can get ip
debug console error message
root@evb-ast2500:/usr/sbin# udhcpc
udhcpc: started, v1.30.1
Failed to revert interface configuration: Link eth0 is managed.
udhcpc: sending discover
udhcpc: sending select for 192.168.36.150
udhcpc: lease of 192.168.36.150 obtained, lease time 7200
/etc/udhcpc.d/50default: Adding DNS 8.8.8.8
/etc/udhcpc.d/50default: Adding DNS 8.8.4.4
The specified interface eth0 is managed by systemd-networkd. Operation refused.
Please configure DNS settings for systemd-networkd managed interfaces directly in their .network files.
root@evb-ast2500:/usr/sbin# ifconfig
eth0      Link encap:Ethernet  HWaddr D0:99:87:52:A1:54
          inet addr:192.168.36.150  Bcast:192.168.36.255  Mask:255.255.255.0
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:1614580 errors:0 dropped:39760 overruns:0 frame:4
          TX packets:3476 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:183211374 (174.7 MiB)  TX bytes:448716 (438.1 KiB)
          Interrupt:19

eth1      Link encap:Ethernet  HWaddr D0:99:87:52:A1:55
          inet addr:169.254.149.138  Bcast:169.254.255.255  Mask:255.255.0.0
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:9906 errors:0 dropped:1 overruns:0 frame:0
          TX packets:89 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:981114 (958.1 KiB)  TX bytes:17108 (16.7 KiB)
          Interrupt:20

lo        Link encap:Local Loopback
          inet addr:127.0.0.1  Mask:255.0.0.0
          UP LOOPBACK RUNNING  MTU:65536  Metric:1
          RX packets:132 errors:0 dropped:0 overruns:0 frame:0
          TX packets:132 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:8680 (8.4 KiB)  TX bytes:8680 (8.4 KiB)

sit0      Link encap:UNSPEC  HWaddr 00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00
          UP RUNNING NOARP  MTU:1480  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)

closed 2020-03-14 by stale[bot]

neuberfran, 2020-08-17:
Hi,
I stay trying this : https://community.nxp.com/thread/506880 and solve this https://hub.mender.io/t/issues-with-systemd-networkd/2302

I have this issue (of topic): 

[image]

mdmillerii, 2020-08-19:
Hi @neuberfran .  
You seem to have stumbled across this issue that mentions `udhcpcd` was used in the context of someone exploring a problem space, but we do not use this daemon in our reference software stack.  I don't know an obvoius answer to my 10 second scan of the issue on mendor to which you linked or the errors from `udhcpcd` you posted.

This repository hosts a yocto rollup for creating a software stack for BMC (Baseboard Management Controller) systems.  We have externally defined APIs (IPMI and Redfish) that specify how to configure the networking for these embeded systems, and our community has written some software to configue `systemd-networkd` to try to meet our requirements.  (Our requirements do not include network roaming beyond basic DHCP support).

You are welcome to try our software (subject to its licensing) to see if it meets your requirements.  Much of the network configuration is in [phosphor-networkd](https://github.com/openbmc/phosphor-networkd).  You can read about our software architecture (where we communicate to this software over D-Bus) and how to contribute should you encounter small changes in our [Documentation Repository](https://github.com/openbmc/docs).  (Hint: you can use `busctl` for a command line interface to D-bus).