openbmc/openbmc #3600: CLI11 not present in SDK

geissonator, 2019-08-15:
Been a bit of IRC discussion on this. We need to get CLI11 in the SDK to fix this issue when compiling CLI11 based projects in the SDK.

```
configure: error: Could not find CLI11 CLI/CLI.hpp
```

bjwyman, 2019-08-15:
I noted this problem trying to setup a Meson build for a repository.
I also noticed it affects phosphor-hwmon (autoconf/automake) and phosphor-watchdog (autoconf/automake when introduced, now Meson).

bjwyman, 2019-08-15:
I also did comparison with json.hpp, which is in the SDK.
Tracked down:
https://github.com/openbmc/openbmc/tree/master/meta-openembedded/meta-oe/recipes-support/cli11
https://github.com/openbmc/openbmc/tree/master/meta-openembedded/meta-oe/recipes-devtools/nlohmann-json
and
https://github.com/openbmc/openbmc/blob/master/meta-phosphor/recipes-core/packagegroups/nativesdk-packagegroup-sdk-host.bbappend

I am just not certain what changes would be needed to get this in the SDK.

bjwyman, 2019-08-15:
```
(wspoonsdk)[bjwyman]$ git clone ssh://openbmc.gerrit/openbmc/phosphor-watchdog
Cloning into 'phosphor-watchdog'...
remote: Counting objects: 8, done
remote: Finding sources: 100% (8/8)
remote: Total 335 (delta 0), reused 327 (delta 0)
Receiving objects: 100% (335/335), 119.71 KiB | 833.00 KiB/s, done.
Resolving deltas: 100% (183/183), done.
(wspoonsdk)[bjwyman]$ cd phosphor-watchdog/
(wspoonsdk)[phosphor-watchdog]$ ls
LICENSE  mainapp.cpp  MAINTAINERS  meson.build  meson_options.txt  test  watchdog.cpp  watchdog.hpp
(wspoonsdk)[phosphor-watchdog]$ meson build
The Meson build system
Version: 0.50.1
Source dir: /tmp/bjwyman/phosphor-watchdog
Build dir: /tmp/bjwyman/phosphor-watchdog/build
Build type: cross build
Project name: phosphor-watchdog
Project version: 0.1
Appending CXXFLAGS from environment: ' -O2 -pipe -g -feliminate-unused-debug-types '
Appending LDFLAGS from environment: '-Wl,-O1 -Wl,--hash-style=gnu -Wl,--as-needed'
Native C++ compiler: c++ (gcc 4.8.5 "c++ (GCC) 4.8.5 20150623 (Red Hat 4.8.5-36)")
Cross C++ compiler: arm-openbmc-linux-gnueabi-g++ -march=armv6 -marm -mtune=arm1176jz-s --sysroot=/esw/san2/openbmc/sdk/witherspoon/2019-08-13/sysroots/armv6-openbmc-linux-gnueabi (gcc 9.1.0)
Host machine cpu family: x86_64
Host machine cpu: x86_64
Target machine cpu family: x86_64
Target machine cpu: x86_64
Build machine cpu family: x86_64
Build machine cpu: x86_64
Found pkg-config: /esw/san2/openbmc/sdk/witherspoon/2019-08-13/sysroots/x86_64-oesdk-linux/usr/bin/pkg-config (0.29.2)
Cross dependency phosphor-dbus-interfaces found: YES 1.0
Cross dependency phosphor-logging found: YES 1.0
Cross dependency sdbusplus found: YES 1.0
Cross dependency sdeventplus found: YES 0.1
Cross dependency GTest found: YES 1.9.0
Cross dependency GMock found: YES 1.9.0
Build targets in project: 3
Found ninja-1.9.0 at /esw/san2/openbmc/sdk/witherspoon/2019-08-13/sysroots/x86_64-oesdk-linux/usr/bin/ninja
(wspoonsdk)[phosphor-watchdog]$ cd build
(wspoonsdk)[build]$ ninja --verbose
[1/6] arm-openbmc-linux-gnueabi-g++ -march=armv6 -marm -mtune=arm1176jz-s --sysroot=/esw/san2/openbmc/sdk/witherspoon/2019-08-13/sysroots/armv6-openbmc-linux-gnueabi -Iphosphor-watchdog@exe -fdiagnostics-color=always -pipe -D_FILE_OFFSET_BITS=64 -Wall -Winvalid-pch -Wnon-virtual-dtor -Wextra -Wpedantic -Werror -std=c++17 -g -MD -MQ 'phosphor-watchdog@exe/mainapp.cpp.o' -MF 'phosphor-watchdog@exe/mainapp.cpp.o.d' -o 'phosphor-watchdog@exe/mainapp.cpp.o' -c ../mainapp.cpp
FAILED: phosphor-watchdog@exe/mainapp.cpp.o
arm-openbmc-linux-gnueabi-g++ -march=armv6 -marm -mtune=arm1176jz-s --sysroot=/esw/san2/openbmc/sdk/witherspoon/2019-08-13/sysroots/armv6-openbmc-linux-gnueabi -Iphosphor-watchdog@exe -fdiagnostics-color=always -pipe -D_FILE_OFFSET_BITS=64 -Wall -Winvalid-pch -Wnon-virtual-dtor -Wextra -Wpedantic -Werror -std=c++17 -g -MD -MQ 'phosphor-watchdog@exe/mainapp.cpp.o' -MF 'phosphor-watchdog@exe/mainapp.cpp.o.d' -o 'phosphor-watchdog@exe/mainapp.cpp.o' -c ../mainapp.cpp
../mainapp.cpp:19:10: fatal error: CLI/CLI.hpp: No such file or directory
   19 | #include <CLI/CLI.hpp>
      |          ^~~~~~~~~~~~~
compilation terminated.
[2/6] arm-openbmc-linux-gnueabi-g++ -march=armv6 -marm -mtune=arm1176jz-s --sysroot=/esw/san2/openbmc/sdk/witherspoon/2019-08-13/sysroots/armv6-openbmc-linux-gnueabi -Iwatchdog@sta -fdiagnostics-color=always -pipe -D_FILE_OFFSET_BITS=64 -Wall -Winvalid-pch -Wnon-virtual-dtor -Wextra -Wpedantic -Werror -std=c++17 -g -fPIC -MD -MQ 'watchdog@sta/watchdog.cpp.o' -MF 'watchdog@sta/watchdog.cpp.o.d' -o 'watchdog@sta/watchdog.cpp.o' -c ../watchdog.cpp
[3/6] arm-openbmc-linux-gnueabi-g++ -march=armv6 -marm -mtune=arm1176jz-s --sysroot=/esw/san2/openbmc/sdk/witherspoon/2019-08-13/sysroots/armv6-openbmc-linux-gnueabi -Itest/9f86d08@@watchdog@exe -fdiagnostics-color=always -pipe -D_FILE_OFFSET_BITS=64 -Wall -Winvalid-pch -Wnon-virtual-dtor -Wextra -Wpedantic -Werror -std=c++17 -g -pthread -DGTEST_HAS_PTHREAD=1 -MD -MQ 'test/9f86d08@@watchdog@exe/watchdog.cpp.o' -MF 'test/9f86d08@@watchdog@exe/watchdog.cpp.o.d' -o 'test/9f86d08@@watchdog@exe/watchdog.cpp.o' -c ../test/watchdog.cpp
ninja: build stopped: subcommand failed.
(wspoonsdk)[build]$
```

bjwyman, 2019-08-16:
```
...
Fri [2019/08/16 09:19:54] <radsquirrel> bjwyman I figured out the cli11/sdk thing
Fri [2019/08/16 09:20:00] <radsquirrel> bjwyman I'll get you a patch in a bit.
Fri [2019/08/16 09:20:22] <ShawnMcCarney[m]> radsquirrel :thumbsup:
...
```

bjwyman, 2019-08-16:
@bradbishop pointed me to some changes that will fix this problem.

bradbishop, 2019-08-16:
https://patchwork.openembedded.org/patch/164065/

bjwyman, 2019-08-21:
https://github.com/openbmc/openbmc/commit/90ca747aa04f375af7ff86462a52028245b174b9

closed 2019-09-13 by bradbishop