openbmc/openbmc #3601: How to add another users for login terminal

chineweff, 2019-08-19:
I find two files:
1. conf/local.conf.sample
There is a parameter named EXTRA_USERS_PARAMS
I modified this parameter(add my useradd command behind)
but it doesn't work.

2. meta-phosphor/conf/distro/include/phosphor-defaults.inc
In this file there is a parameter named EXTRA_USERS_PARAMS_pn-obmc-phosphor-image

How can I add users without modifying the common code(phosphor-defaults.inc)?

geissonator, 2019-08-19:
@ratagupt or @rthomaiy any directions we can point people to?

gtmills, 2019-11-16:
@ratagupt or @rthomaiy Any suggestion here?

rthomaiy, 2019-11-19:
EXTRA_USERS_PARAMS in meta-phosphor/conf/distro/include/phosphor-defaults.inc is to set the root password in common way instead of doing it in each & every repo of local.conf.sample file. 

But if you want to add totally new user / or override, then you can do the same through local.conf.sample.

Note: To remove security issue, password setting was done on obmc-phosphor-image, instead of the base one (basically we don't want the password to be copied in the read-only file system if any, instead it must be in the overlay, in case if it is used like that).

rthomaiy, 2019-11-19:
Also, let us know the command you have used, can try to see why it was not working for you.

closed 2020-06-19 by stale[bot]