openbmc/openbmc #3605: Webui User manager can't show any users when new user is created bye ipmitool on openbmc 2.8.0-dev

xiuzhi123, 2019-08-27:
1,Create user test on webui user manager page.  the user list can show every users.
2, Create new user test3 by:
 ipmitool user set name 3 test3                                                  
  ipmitool user list 1                                                            
  ipmitool user enabled 3                                                         
  ipmitool user set password 3 "xxxxxx" 
3, reopen the webui user manager page ,the user list show nothing.
The web debug error:
Failed to load resource: the server responded with a status of 500 (Internal Server Error)
app.bundle.js:53 {"data":{"@odata.context":"/redfish/v1/$metadata#ManagerAccount.ManagerAccount","@odata.type":"#ManagerAccount.v1_0_3.ManagerAccount","Description":"User Account","Enabled":true,"Locked":false,"Locked@Redfish.AllowableValues":["false"],"Name":"User Account","Password":null,"RoleId":"Administrator","error":{"@Message.ExtendedInfo":[{"@odata.type":"/redfish/v1/$metadata#Message.v1_0_0.Message","Message":"The request failed due to an internal service error.  The service is still operational.","MessageArgs":[],"MessageId":"Base.1.4.0.InternalError","Resolution":"Resubmit the request.  If the problem persists, consider resetting the service.","Severity":"Critical"}],"code":"Base.1.4.0.InternalError","message":"The request failed due to an internal service error.  The service is still operational."}},"status":500,"config":{"method":"GET","transformRequest":[null],"transformResponse":[null],"jsonpCallbackParam":"callback","url":"https://192.168.200.132/redfish/v1/AccountService/Accounts/test3","withCredentials":true,"headers":{"Accept":"application/json","X-XSRF-TOKEN":"WMVdmigFQuja5dFq5YpV"},"timeout":20000},"statusText":"Internal Server Error","xhrStatus":"complete"}
﻿
￼ curl  -k https://root:0penBmc@${bmc}/redfish/v1/AccountService/Accounts
{
  "@odata.context": "/redfish/v1/$metadata#ManagerAccountCollection.ManagerAccountCollection",
  "@odata.id": "/redfish/v1/AccountService/Accounts",
  "@odata.type": "#ManagerAccountCollection.ManagerAccountCollection",
  "Description": "BMC User Accounts",
  "Members": [
    {
      "@odata.id": "/redfish/v1/AccountService/Accounts/test3"
    },
    {
      "@odata.id": "/redfish/v1/AccountService/Accounts/test"
    },
    {
      "@odata.id": "/redfish/v1/AccountService/Accounts/root"
    }
  ],
  "Members@odata.count": 3,
  "Name": "Accounts Collection"
}

ratagupt, 2019-08-27:
@xiuzhi123 : Can you please share the output of GET request of individual accounts?
eg: curl -k https://root:0penBmc@${bmc}/redfish/v1/AccountService/Accounts/test3

rthomaiy, 2019-08-28:
bmcweb doesn't handle users with no privilege levels, and ipmi by default creates users with NO-ACCESS privilege. 
There is already a WIP review https://gerrit.openbmc-project.xyz/#/c/openbmc/bmcweb/+/23962/  which will address the same 

@rthomaiy

xiuzhi123, 2019-08-28:
curl -k https://root:0penBmc@${bmc}/redfish/v1/AccountService/Accounts/test3
{
  "@odata.context": "/redfish/v1/$metadata#ManagerAccount.ManagerAccount",
  "@odata.type": "#ManagerAccount.v1_0_3.ManagerAccount",
  "Description": "User Account",
  "Enabled": false,
  "Locked": false,
  "Locked@Redfish.AllowableValues": [
    "false"
  ],
  "Name": "User Account",
  "Password": null,
  "RoleId": "Administrator",
  "error": {
    "@Message.ExtendedInfo": [
      {
        "@odata.type": "/redfish/v1/$metadata#Message.v1_0_0.Message",
        "Message": "The request failed due to an internal service error.  The service is still operational.",
        "MessageArgs": [],
        "MessageId": "Base.1.4.0.InternalError",
        "Resolution": "Resubmit the request.  If the problem persists, consider resetting the service.",
        "Severity": "Critical"
      }
    ],
    "code": "Base.1.4.0.InternalError",
    "message": "The request failed due to an internal service error.  The service is still operational."
  }
}

xiuzhi123, 2019-08-28:
The webui user manager will show the user list after I execute the command :
 ipmitool user priv 3 4 1

The root cause is the webui user manager doesn't have the  Exception Handling for users with incomplete information

xiuzhi123, 2019-08-29:
I tested the patch of 23962 ,it works.

closed 2019-08-29 by xiuzhi123