openbmc/openbmc #3606: Build

onjustgo, 2019-09-04:


onjustgo, 2019-09-04:
I compile meta-evb-enclustra with following steps:
[with branch "Master"]
Step1:
    export TEMPLATECONF=meta-evb/meta-evb-enclustra/meta-evb-zx3-pm3/conf
    . openbmc-env
    bitbake obmc-phosphor-image
Step2:
    Terminal shows file "/meta-evb/meta-evb-enclustra/meta-evb-zx3-pm3/conf/machine/evb-zx3-pm3.conf"   could't require the line "conf/machine/include/machine-xilinx-board.inc", so i comment this line, then it could compile again
Step3:
    I found error again, Terminal shows:
    `ERROR: phosphor-mapper-1.0+gitAUTOINC+5eddf44006-r1 do_compile: oe_runmake failed
ERROR: phosphor-mapper-1.0+gitAUTOINC+5eddf44006-r1 do_compile: Execution of '/home/ubu/openbmc-master/build/tmp/work/cortexa9t2hf-neon-openbmc-linux-gnueabi/phosphor-mapper/1.0+gitAUTOINC+5eddf44006-r1/temp/run.do_compile.70359' failed with exit code 1:
Makefile:1186: target 'check-valgrind-recursive' given more than once in the same rule
Makefile:1186: target 'check-valgrind-memcheck-recursive' given more than once in the same rule
Makefile:1186: target 'check-valgrind-helgrind-recursive' given more than once in the same rule
Makefile:1186: target 'check-valgrind-drd-recursive' given more than once in the same rule
Makefile:1186: target 'check-valgrind-sgcheck-recursive' given more than once in the same rule
make  all-recursive
make[1]: Entering directory '/home/ubu/openbmc-master/build/tmp/work/cortexa9t2hf-neon-openbmc-linux-gnueabi/phosphor-mapper/1.0+gitAUTOINC+5eddf44006-r1/build'
Makefile:1186: target 'check-valgrind-recursive' given more than once in the same rule
Makefile:1186: target 'check-valgrind-memcheck-recursive' given more than once in the same rule
Makefile:1186: target 'check-valgrind-helgrind-recursive' given more than once in the same rule
Makefile:1186: target 'check-valgrind-drd-recursive' given more than once in the same rule
Makefile:1186: target 'check-valgrind-sgcheck-recursive' given more than once in the same rule
Making all in fail-monitor
make[2]: Entering directory '/home/ubu/openbmc-master/build/tmp/work/cortexa9t2hf-neon-openbmc-linux-gnueabi/phosphor-mapper/1.0+gitAUTOINC+5eddf44006-r1/build/fail-monitor'
make[2]: Nothing to be done for 'all'.
make[2]: Leaving directory '/home/ubu/openbmc-master/build/tmp/work/cortexa9t2hf-neon-openbmc-linux-gnueabi/phosphor-mapper/1.0+gitAUTOINC+5eddf44006-r1/build/fail-monitor'
make[2]: Entering directory '/home/ubu/openbmc-master/build/tmp/work/cortexa9t2hf-neon-openbmc-linux-gnueabi/phosphor-mapper/1.0+gitAUTOINC+5eddf44006-r1/build'
Makefile:1186: target 'check-valgrind-recursive' given more than once in the same rule
Makefile:1186: target 'check-valgrind-memcheck-recursive' given more than once in the same rule
Makefile:1186: target 'check-valgrind-helgrind-recursive' given more than once in the same rule
Makefile:1186: target 'check-valgrind-drd-recursive' given more than once in the same rule
Makefile:1186: target 'check-valgrind-sgcheck-recursive' given more than once in the same rule
./arm-openbmc-linux-gnueabi-libtool  --tag=CXX   --mode=link arm-openbmc-linux-gnueabi-g++  -mthumb -mfpu=neon -mfloat-abi=hard -mcpu=cortex-a9 --sysroot=/home/ubu/openbmc-master/build/tmp/work/cortexa9t2hf-neon-openbmc-linux-gnueabi/phosphor-mapper/1.0+gitAUTOINC+5eddf44006-r1/recipe-sysroot -std=c++17  -DBOOST_SYSTEM_NO_DEPRECATED -DBOOST_ERROR_CODE_HEADER_ONLY -DBOOST_ALL_NO_LIB -O2 -pipe -g -feliminate-unused-debug-types -fmacro-prefix-map=/home/ubu/openbmc-master/build/tmp/work/cortexa9t2hf-neon-openbmc-linux-gnueabi/phosphor-mapper/1.0+gitAUTOINC+5eddf44006-r1=/usr/src/debug/phosphor-mapper/1.0+gitAUTOINC+5eddf44006-r1                      -fdebug-prefix-map=/home/ubu/openbmc-master/build/tmp/work/cortexa9t2hf-neon-openbmc-linux-gnueabi/phosphor-mapper/1.0+gitAUTOINC+5eddf44006-r1=/usr/src/debug/phosphor-mapper/1.0+gitAUTOINC+5eddf44006-r1                      -fdebug-prefix-map=/home/ubu/openbmc-master/build/tmp/work/cortexa9t2hf-neon-openbmc-linux-gnueabi/phosphor-mapper/1.0+gitAUTOINC+5eddf44006-r1/recipe-sysroot=                      -fdebug-prefix-map=/home/ubu/openbmc-master/build/tmp/work/cortexa9t2hf-neon-openbmc-linux-gnueabi/phosphor-mapper/1.0+gitAUTOINC+5eddf44006-r1/recipe-sysroot-native=  -fvisibility-inlines-hidden -lsdbusplus -lsystemd -pthread -ltinyxml2 -Wl,-O1 -Wl,--hash-style=gnu -Wl,--as-needed -o mapperx src/mapperx-main.o src/mapperx-argument.o src/mapperx-processing.o src/mapperx-associations.o  
arm-openbmc-linux-gnueabi-libtool: link: arm-openbmc-linux-gnueabi-g++ -mthumb -mfpu=neon -mfloat-abi=hard -mcpu=cortex-a9 --sysroot=/home/ubu/openbmc-master/build/tmp/work/cortexa9t2hf-neon-openbmc-linux-gnueabi/phosphor-mapper/1.0+gitAUTOINC+5eddf44006-r1/recipe-sysroot -std=c++17 -DBOOST_SYSTEM_NO_DEPRECATED -DBOOST_ERROR_CODE_HEADER_ONLY -DBOOST_ALL_NO_LIB -O2 -pipe -g -feliminate-unused-debug-types -fmacro-prefix-map=/home/ubu/openbmc-master/build/tmp/work/cortexa9t2hf-neon-openbmc-linux-gnueabi/phosphor-mapper/1.0+gitAUTOINC+5eddf44006-r1=/usr/src/debug/phosphor-mapper/1.0+gitAUTOINC+5eddf44006-r1 -fdebug-prefix-map=/home/ubu/openbmc-master/build/tmp/work/cortexa9t2hf-neon-openbmc-linux-gnueabi/phosphor-mapper/1.0+gitAUTOINC+5eddf44006-r1=/usr/src/debug/phosphor-mapper/1.0+gitAUTOINC+5eddf44006-r1 -fdebug-prefix-map=/home/ubu/openbmc-master/build/tmp/work/cortexa9t2hf-neon-openbmc-linux-gnueabi/phosphor-mapper/1.0+gitAUTOINC+5eddf44006-r1/recipe-sysroot= -fdebug-prefix-map=/home/ubu/openbmc-master/build/tmp/work/cortexa9t2hf-neon-openbmc-linux-gn [... cut ...]
make[1]: Leaving directory '/home/ubu/openbmc-master/build/tmp/work/cortexa9t2hf-neon-openbmc-linux-gnueabi/phosphor-mapper/1.0+gitAUTOINC+5eddf44006-r1/build'
Makefile:771: recipe for target 'all' failed
make: *** [all] Error 2
WARNING: exit code 1 from a shell command.

ERROR: Logfile of failure stored in: /home/ubu/openbmc-master/build/tmp/work/cortexa9t2hf-neon-openbmc-linux-gnueabi/phosphor-mapper/1.0+gitAUTOINC+5eddf44006-r1/temp/log.do_compile.70359`

    how to fix it, hope anyone can teach me!!!

geissonator, 2019-09-04:
@pstrinkle this board still supported?

bradbishop, 2019-09-09:
Support for this board was added by @alehed about a year ago.  @alehed are you interested in maintaining the support for this board?

alehed, 2019-09-09:
I'm not working with the specific hardware any more so I can't guarantee I will always be able to debug stuff or have the time to keep things up to date.

In this case it seems that this file has been removed upstream and the functionality is replaced by adding `virtual/bootloader` to `EXTRA_IMAGEDEPENDS ` (see Xilinx/meta-xilinx@f24275598687bcec0252186cb1d9c54b426fef9f). So try deleting the include of `machine-xilinx-board.inc` and replacing

```
EXTRA_IMAGEDEPENDS += " \
		u-boot-zynq-uenv \
		virtual/boot-bin \
		"
```

with

```
EXTRA_IMAGEDEPENDS += " \
		u-boot-zynq-uenv \
		virtual/boot-bin \
                virtual/bootloader \
		"
```

closed 2020-04-09 by stale[bot]

dilraj-n, 2020-06-03:
I tried this but received the error "mkimage" command not found - U-Boot images will not be built.
Any solutions

mdmillerii, 2020-06-03:
`mkimage` should come from Das Uboot not sure what's missing, maybe a -native package to create/install the tool?