openbmc/openbmc #3607: Intel S2600WF BMC in 'NotReady' state

lqu, 2019-09-06:
I'm not sure if this is a bug or wrong config. I'm using 2.8.0-dev (Aug 2019), and the Intel S2600WF board with NO modifications. After loading the flash image to the board, the main CPU cannot start and the front panel buttons don't work. The BMC console shows that the BMC state is 'Not Ready'.

```
root@s2600wf:~# cat /etc/os-release
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="2.8.0-dev"
VERSION_ID="2.8.0-dev-307-gadb0b75"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) 2.8.0-dev"
BUILD_ID="2.8.0-dev"
OPENBMC_TARGET_MACHINE="s2600wf"

```

```
root@s2600wf:~# obmcutil state
CurrentBMCState     : xyz.openbmc_project.State.BMC.BMCState.NotReady
CurrentPowerState   : xyz.openbmc_project.State.Chassis.PowerState.Off
Error finding '/xyz/openbmc_project/state/host0' service: No such file or directory
```

```
root@s2600wf:~# systemctl list-jobs --no-pager
JOB UNIT                                                            TYPE  STATE
114 xyz.openbmc_project.Time.Manager.service                        start waiting
165 op-reset-chassis-running@0.service                              start waiting
160 phosphor-reset-host-check@0.service                             start waiting
157 obmc-host-reset@0.target                                        start waiting
  1 multi-user.target                                               start waiting
158 phosphor-reset-sensor-states@0.service                          start waiting
168 obmc-power-reset-on@0.target                                    start waiting
111 mapper-wait@-xyz-openbmc_project-state-host0.service            start running
109 phosphor-discover-system-state@0.service                        start waiting
 82 obmc-led-group-start@bmc_booted.service                         start waiting
164 op-reset-chassis-on@0.service                                   start waiting
167 mapper-wait@-xyz-openbmc_project-Chassis-Control-Power0.service start running
162 obmc-host-reset-running@0.target                                start waiting
100 xyz.openbmc_project.State.Host.service                          start waiting
171 phosphor-reset-host-running@0.service                           start waiting
```

geissonator, 2019-09-06:
Please do a "journalctl > /tmp/journal.txt" and attach that to this bug. A service is probably failing which is hanging up the rest of its dependent services.

lqu, 2019-09-06:
@geissonator Thank you so much! Here is the journal.
[journal.txt](https://github.com/openbmc/openbmc/files/3585132/journal.txt)

```
Aug 29 20:35:17 s2600wf kernel: aspeed_gfx 1e6e6000.display: failed to initialize reserved mem: -19
Aug 29 20:35:17 s2600wf kernel: random: fast init done
Aug 29 20:35:17 s2600wf kernel: random: crng init done
Aug 29 20:35:17 s2600wf kernel: loop: module loaded
Aug 29 20:35:17 s2600wf kernel: aspeed-smc 1e620000.spi: Using 50 MHz SPI frequency
Aug 29 20:35:17 s2600wf kernel: aspeed-smc 1e620000.spi: mx66l51235f (65536 Kbytes)
Aug 29 20:35:17 s2600wf kernel: aspeed-smc 1e620000.spi: CE0 window [ 0x20000000 - 0x24000000 ] 64MB
Aug 29 20:35:17 s2600wf kernel: aspeed-smc 1e620000.spi: CE1 window [ 0x24000000 - 0x2a000000 ] 96MB
Aug 29 20:35:17 s2600wf kernel: aspeed-smc 1e620000.spi: read control register: 203c0641
Aug 29 20:35:17 s2600wf kernel: 5 fixed-partitions partitions found on MTD device bmc
Aug 29 20:35:17 s2600wf kernel: Creating 5 MTD partitions on "bmc":
Aug 29 20:35:17 s2600wf kernel: 0x000000000000-0x000000060000 : "u-boot"
Aug 29 20:35:17 s2600wf kernel: 0x000000060000-0x000000080000 : "u-boot-env"
Aug 29 20:35:17 s2600wf kernel: 0x000000080000-0x0000004c0000 : "kernel"
Aug 29 20:35:17 s2600wf kernel: 0x0000004c0000-0x000001c00000 : "rofs"
Aug 29 20:35:17 s2600wf kernel: 0x000001c00000-0x000002000000 : "rwfs"
Aug 29 20:35:17 s2600wf kernel: aspeed-smc 1e630000.spi: Error applying setting, reverse things back
Aug 29 20:35:17 s2600wf kernel: aspeed-smc 1e630000.spi: Using 50 MHz SPI frequency
Aug 29 20:35:17 s2600wf kernel: aspeed-smc 1e630000.spi: unrecognized JEDEC id bytes: 0a 0a 0a 0a 0a 0a
Aug 29 20:35:17 s2600wf kernel: aspeed-smc 1e630000.spi: Aspeed SMC probe failed -2
Aug 29 20:35:17 s2600wf kernel: aspeed-smc: probe of 1e630000.spi failed with error -2
......
Aug 29 20:35:30 s2600wf systemd[1]: Started Fan Sensor.
Aug 29 20:35:30 s2600wf fw_setenv[163]: Warning: Bad CRC, using default environment
Aug 29 20:35:30 s2600wf systemd[1]: Starting SSH Key Generation...
Aug 29 20:35:30 s2600wf create_usbhid.sh[162]: /usr/bin/create_usbhid.sh: line 133: echo: write error: No such device
Aug 29 20:35:30 s2600wf systemd[1]: Started Phosphor certificate manager for nslcd.
Aug 29 20:35:31 s2600wf systemd[1]: Started Hwmon Temp Sensor.
Aug 29 20:35:31 s2600wf systemd[1]: Starting Phosphor Inventory Manager...
Aug 29 20:35:31 s2600wf systemd[1]: Starting Avahi mDNS/DNS-SD Stack...
Aug 29 20:35:32 s2600wf systemd[1]: Started Phosphor certificate manager for authority.
Aug 29 20:35:32 s2600wf systemd[1]: Starting Phosphor Download Manager...
Aug 29 20:35:32 s2600wf systemd[1]: Started Adc Sensor.
Aug 29 20:35:32 s2600wf systemd[1]: Started PSU Sensor.
Aug 29 20:35:33 s2600wf sh[173]: Generating 2048 bit rsa key, this may take a while...
Aug 29 20:35:33 s2600wf systemd[1]: Started Phosphor certificate manager for bmcweb.
Aug 29 20:35:34 s2600wf fansensor[167]: Error communicating to entity manager
Aug 29 20:35:34 s2600wf fansensor[167]: error communicating to entity manager
Aug 29 20:35:34 s2600wf fansensor[167]: Error calling entity manager
Aug 29 20:35:34 s2600wf systemd[1]: Starting Phosphor Download Manager...
Aug 29 20:35:34 s2600wf mcutempsensor[161]: Error contacting entity manager
Aug 29 20:35:34 s2600wf systemd[1]: Starting Phosphor Settings Daemon...
Aug 29 20:35:34 s2600wf systemd[1]: Started Phosphor-Pid-Control Margin-based Fan Control Daemon.
Aug 29 20:35:34 s2600wf hwmontempsensor[178]: Error communicating to entity manager
Aug 29 20:35:34 s2600wf hwmontempsensor[178]: error communicating to entity manager

```

geissonator, 2019-09-06:
I'm not very familiar with the S2600WF but I see 2 service fails, kvm and kcs. My guess is the kcs failure is causing your issues.

```
Aug 29 20:36:09 s2600wf kcsbridged[331]: Couldn't open SMS channel O_RDWR
Aug 29 20:36:09 s2600wf systemd[1]: phosphor-ipmi-kcs@ipmi-kcs3.service: Main process exited, code=exited, status=1/FAILURE
Aug 29 20:36:09 s2600wf systemd[1]: phosphor-ipmi-kcs@ipmi-kcs3.service: Failed with result 'exit-code'.
```

Probably need some input from @yoojae or @edtanous on what could cause that failure.

lqu, 2019-09-10:
I'd like to know the current development status of OpenBMC on S2600wf. Is there a previous version that gives a good demo? Is there a qemu machine to try the image? (I tried ast2500-evb, but saw the same errors.)

yoojae, 2019-09-10:
Hi Liang,

> I'd like to know the current development status of OpenBMC on S2600wf. 
> Is there a previous version that gives a good demo? Is there a qemu 
> machine to try the image? (I tried ast2500-evb, but saw the same errors.)

We aren't taking care of S2600WF in OpenBMC tree. Please check
https://github.com/intel-bmc if you need a support for the platform.

Thanks,
Jae

lqu, 2019-09-11:
Thank you Jae! I checked out intel-bmc/openbmc, but there was a build error. Is master/intel the right branch to use?
```
ERROR: boost-1.69.0-r0 do_package_write_rpm: Function failed: BUILDSPEC```

lqu, 2019-09-12:
I fixed the build issue, but intel-bmc/openbmc has the same problem of this thread (BMC not ready).

lqu, 2019-09-13:
More discoveries:
obmc-ikvm complains that /dev/hidg0 is missing. kcsbridge compalins /dev/ipmi-kcs3 is missing.

My image doesn't have any /dev/ipmi*, but qemu romulus VM has /dev/ipmi-bt-host.

lqu, 2019-10-29:
Use github.com/intel-bmc/openbmc/meta-openbmc-mods/meta-wolfpass.

closed 2019-10-29 by lqu