openbmc/openbmc #3609: add system total power sensor

FrankRenn, 2019-09-12:
I need to add a sensor called sys_total_power, which is the sum of the output power of the two PSUs. 
By configuring the device tree, corresponding files can be generated in /sys/class/hwmon.
But how to add the power values of the two files to the same sensor(sys_total_power), as shown below

[image]

closed 2019-09-23 by FrankRenn

NKV-obmc, 2021-11-25:
Hi renweihang,

Did u got the solution.?
How you implemented this?

spinler, 2021-11-29:
One can now use a virtual sensor, like https://github.com/openbmc/openbmc/blob/master/meta-bytedance/meta-g220a/recipes-phosphor/sensors/phosphor-virtual-sensor/virtual_sensor_config.json#L5.

NKV-obmc, 2021-11-29:
> One can now use a virtual sensor, like https://github.com/openbmc/openbmc/blob/master/meta-bytedance/meta-g220a/recipes-phosphor/sensors/phosphor-virtual-sensor/virtual_sensor_config.json#L5.

Thanks spinler, this  will meet my requirements...

NKV-obmc, 2021-12-06:
Is it possible to link virtual sensors with sensors inside phosphor-hwmon?..

spinler, 2021-12-06:
@NKV-obmc Sure, I don't see why not.

NKV-obmc, 2021-12-07:
I tried to map one of the actual sensor in phosphor-hwmon with the virtual sensor.
so i modified the virtual_sensor_config.json file like this 

[
    {
        "Desc":
        {
            "Name": "Ambient_Virtual_Temp",
            "SensorType": "temperature"
        },
        "Threshold" :
        {
            "CriticalHigh": 100,
            "CriticalLow": 0,
            "WarningHigh": 100,
            "WarningLow": 0
        },
        "Params":
        {
            "DbusParam":
            [
                {
                    "ParamName": "T0",
                    "Desc":
                    {
                        "Name": "VR CPU0 VCCIN",
                        "SensorType": "temperature"
                    }
                }
            ]
        },
        "Expression": "T0"
    }
]

After this executing the command d-bus introspect

 busctl introspect  xyz.openbmc_project.VirtualSensor
/xyz/openbmc_project/sensors/temperature/Ambient_Virtual_Temp

xyz.openbmc_project.Sensor.Threshold.Critical interface -         -                                        -
.CriticalAlarmHigh                            property  b         false                                    emits-change writable
.CriticalAlarmHighEnd                         property  b         false                                    emits-change writable
.CriticalAlarmLow                             property  b         false                                    emits-change writable
.CriticalAlarmLowEnd                          property  b         false                                    emits-change writable
.CriticalHigh                                 property  d         100                                      emits-change writable
.CriticalLow                                  property  d         0                                        emits-change writable
xyz.openbmc_project.Sensor.Threshold.Warning  interface -         -                                        -
.WarningAlarmHigh                             property  b         false                                    emits-change writable
.WarningAlarmHighEnd                          property  b         false                                    emits-change writable
.WarningAlarmLow                              property  b         false                                    emits-change writable
.WarningAlarmLowEnd                           property  b         false                                    emits-change writable
.WarningHigh                                  property  d         100                                      emits-change writable
.WarningLow                                   property  d         0                                        emits-change writable
xyz.openbmc_project.Sensor.Value              interface -         -                                        -
.MaxValue                                     property  d         0                                        emits-change writable
.MinValue                                     property  d         0                                        emits-change writable
.Unit                                         property  s         "xyz.openbmc_project.Sensor.Value.Unit.… emits-change writable
.Value                                        property  d         nan                                      emits-change writable

Its showing like this.
But actually mapped sensor have some values.
Its not showing.

Also, Is entity-manager is required for working phophur-virtual sensor?

NKV-obmc, 2021-12-09:
> @NKV-obmc Sure, I don't see why not.

Hi Spinler,
Its working now, able to list in dbus.
Previously faced some sensor related issues..

NKV-obmc, 2021-12-13:
How can we map the virtual sensor with redfish?
I tried modifying inside virtual_sensor_config.json

		"Associations":
		[
			[
				"chassis",
				"all_sensors",
				"/xyz/openbmc_project/inventory/system/chassis/motherboard"
			]
		],
But its not worked for me.
Its also not listing under xyz.openbmc_project.ObjectMapper.
But all other sensors are listing under xyz.openbmc_project.ObjectMapper..

My Object mapper is 
Service xyz.openbmc_project.ObjectMapper:
  └─/xyz/openbmc_project
    ├─/xyz/openbmc_project/inventory
    │ └─/xyz/openbmc_project/inventory/system
    │   └─/xyz/openbmc_project/inventory/system/chassis
    │     └─/xyz/openbmc_project/inventory/system/chassis/motherboard
    │       └─/xyz/openbmc_project/inventory/system/chassis/motherboard/all_sensors

Is there any other configuration need to do?

spinler, 2021-12-13:
I wonder if redfish is looking for associations to /xyz/openbmc_project/inventory/system/chassis and not /xyz/openbmc_project/inventory/system/chassis/motherboard?

NKV-obmc, 2021-12-15:
Hi Spinler,

Not listing in redfish yet.
Tried /xyz/openbmc_project/inventory/system/chassis also..

spinler, 2021-12-15:
Does  /xyz/openbmc_project/inventory/system/chassis show up in your chassis list in Redfish?

NKV-obmc, 2021-12-17:
Hi Spinler,

Thanks very much for your support.
Issue was my code base for virtual sensor package is older one.
So i updated with latest revision .
Now its able to list in web-ui.

Regards
Nikhil