openbmc/openbmc #3611: ipkvm error reporting needs improvement

shenki, 2019-09-16:
When booting openbmc master (2.8.0-dev-391-g283f4a7bb) on palmetto with a development kernel I see this:

```
Sep 16 07:40:35 palm8-bmc systemd[1]: Started OpenBMC ipKVM daemon.
Sep 16 07:40:36 palm8-bmc obmc-ikvm[270]: Failed to open input device
Sep 16 07:40:36 palm8-bmc obmc-ikvm[270]: Failed to open a file
Sep 16 07:40:36 palm8-bmc obmc-ikvm[270]: terminate called after throwing an instance of 'sdbusplus::xyz::openbmc_project::Common::File::Error::Open'
Sep 16 07:40:36 palm8-bmc obmc-ikvm[270]:   what():  xyz.openbmc_project.Common.File.Error.Open: Failed to open a file
Sep 16 07:40:36 palm8-bmc systemd[1]: start-ipkvm.service: Main process exited, code=killed, status=6/ABRT
Sep 16 07:40:36 palm8-bmc systemd[1]: start-ipkvm.service: Failed with result 'signal'.
```

I am sure that the error is due to my custom kernel and I do not need help fixing that.

I would like to see the tool report which input device it could not open (a path?).

mine260309, 2019-09-16:
Possibly the info you need is in journal, try `journalctl -o json-pretty` and see if there is information you need.

geissonator, 2019-09-16:
Yeah, I saw this recently elsewhere, it's in the metadata of the journal message:

```
journalctl -o json-pretty MESSAGE="Failed to open input device"
{
        "MESSAGE" : "Failed to open input device",
        "_TRANSPORT" : "journal",
        "ERROR" : "No such file or directory",
        ...
        **"PATH" : "/dev/hidg0",**
        ...
}
```

closed 2020-04-16 by stale[bot]

hariharanrangasamy, 2025-12-05:
FYI. I got the same error. Adding the patch mentioned [here ](https://github.com/openbmc/obmc-ikvm/issues/15) will fix the issue.