openbmc/openbmc #3613: bmcweb_git.bb:do_compile failed with exit code 1

LiveFreeAndRoam, 2019-09-23:
This is my first attempt to build OpenBMC.  The template I'm using is:

<pre><font color="#8AE234"><b>matt@marathon</b></font>:<font color="#729FCF"><b>~</b></font>$ echo $TEMPLATECONF 
meta-evb/meta-evb-aspeed/meta-evb-ast2500/conf/
</pre>

The build stopped, and seemingly bombed out when trying to compile webserver_main.cpp.  However, I was able to copy-n-paste the compilation command and run it from the shell, where it compiled successfully.  I then reran bitbake, and it completed successfully.  It's not clear what caused the compilation to stop.  I'm pushing forward, but now unsure if there will be some issue with the final output or whether I can trust the rebuild.

I will be glad for any observations from the error log below:

<pre><font color="#8AE234"><b>matt@marathon</b></font>:<font color="#729FCF"><b>~/openbmc</b></font>$ bitbake obmc-phosphor-image
<font color="#FCE94F"><b>WARNING</b></font>: <font color="#C4A000">Layer evb should set LAYERSERIES_COMPAT_evb in its conf/layer.conf file to list the core layer names it is compatible with.</font>
<font color="#FCE94F"><b>WARNING</b></font>: <font color="#C4A000">Layer evb-aspeed should set LAYERSERIES_COMPAT_evb-aspeed in its conf/layer.conf file to list the core layer names it is compatible with.</font>
<font color="#FCE94F"><b>WARNING</b></font>: <font color="#C4A000">Layer evb-ast2500 should set LAYERSERIES_COMPAT_evb-ast2500 in its conf/layer.conf file to list the core layer names it is compatible with.</font>
<font color="#FCE94F"><b>WARNING</b></font>: <font color="#C4A000">Layer evb should set LAYERSERIES_COMPAT_evb in its conf/layer.conf file to list the core layer names it is compatible with.</font>
<font color="#FCE94F"><b>WARNING</b></font>: <font color="#C4A000">Layer evb-aspeed should set LAYERSERIES_COMPAT_evb-aspeed in its conf/layer.conf file to list the core layer names it is compatible with.</font>
<font color="#FCE94F"><b>WARNING</b></font>: <font color="#C4A000">Layer evb-ast2500 should set LAYERSERIES_COMPAT_evb-ast2500 in its conf/layer.conf file to list the core layer names it is compatible with.</font>
Parsing recipes: 100% |##########################################| Time: 0:02:12
Parsing of 2360 .bb files complete (0 cached, 2360 parsed). 3480 targets, 337 skipped, 0 masked, 0 errors.
<font color="#FCE94F"><b>WARNING</b></font>: <font color="#C4A000">No bb files matched BBFILE_PATTERN_evb &apos;&apos;</font>
<font color="#FCE94F"><b>WARNING</b></font>: <font color="#C4A000">No bb files matched BBFILE_PATTERN_evb-aspeed &apos;&apos;</font>
<font color="#FCE94F"><b>WARNING</b></font>: <font color="#C4A000">No bb files matched BBFILE_PATTERN_evb-ast2500 &apos;&apos;</font>
<b>NOTE</b>: Resolving any missing task queue dependencies

Build Configuration:
BB_VERSION           = &quot;1.43.1&quot;
BUILD_SYS            = &quot;x86_64-linux&quot;
NATIVELSBSTRING      = &quot;ubuntu-18.04&quot;
TARGET_SYS           = &quot;arm-openbmc-linux-gnueabi&quot;
MACHINE              = &quot;evb-ast2500&quot;
DISTRO               = &quot;openbmc-phosphor&quot;
DISTRO_VERSION       = &quot;0.1.0&quot;
TUNE_FEATURES        = &quot;arm thumb arm1176jzs&quot;
TARGET_FPU           = &quot;soft&quot;
meta                 
meta-poky            
meta-oe              
meta-networking      
meta-python          
meta-webserver       
meta-phosphor        
meta-aspeed          
meta-evb             
meta-evb-aspeed      
meta-evb-ast2500     = &quot;master:44d8109d65a066a3a4931ee189e95121e506c545&quot;

Initialising tasks: 100% |#######################################| Time: 0:00:04
Sstate summary: Wanted 1381 Found 0 Missed 1381 Current 0 (0% match, 0% complete)
<b>NOTE</b>: Executing Tasks
<b>NOTE</b>: Setscene tasks completed
<font color="#FCE94F"><b>WARNING</b></font>: <font color="#C4A000">xz-native-5.2.4-r0 do_fetch: Failed to fetch URL http://tukaani.org/xz/xz-5.2.4.tar.gz, attempting MIRRORS if available</font>
<font color="#FCE94F"><b>WARNING</b></font>: <font color="#C4A000">flex-native-2.6.0-r0 do_fetch: Failed to fetch URL https://downloads.sourceforge.net/flex/flex-2.6.0.tar.bz2, attempting MIRRORS if available</font>
<font color="#FCE94F"><b>WARNING</b></font>: <font color="#C4A000">sdbusplus-1.0+gitAUTOINC+4212292bcf-r1 do_fetch: Failed to fetch URL git://github.com/openbmc/sdbusplus, attempting MIRRORS if available</font>
<font color="#FCE94F"><b>WARNING</b></font>: <font color="#C4A000">python-pygobject-2.28.7-r0 do_patch: Fuzz detected:</font>

<font color="#C4A000">Applying patch obsolete_automake_macros.patch</font>
<font color="#C4A000">patching file configure.ac</font>
<font color="#C4A000">Hunk #2 succeeded at 82 with fuzz 1.</font>

<font color="#C4A000">The context lines in the patches can be updated with devtool:</font>

<font color="#C4A000">    devtool modify python-pygobject</font>
<font color="#C4A000">    devtool finish --force-patch-refresh python-pygobject &lt;layer_path&gt;</font>

<font color="#C4A000">Don&apos;t forget to review changes done by devtool!</font>

<font color="#FCE94F"><b>WARNING</b></font>: <font color="#C4A000">python-pygobject-2.28.7-r0 do_patch: QA Issue: Patch log indicates that patches do not apply cleanly. [patch-fuzz]</font>
<font color="#EF2929"><b>ERROR</b></font>: <font color="#CC0000">bmcweb-1.0+gitAUTOINC+f8c3e6f0c2-r0 do_compile: Execution of &apos;/home/matt/openbmc/build/tmp/work/arm1176jzs-openbmc-linux-gnueabi/bmcweb/1.0+gitAUTOINC+f8c3e6f0c2-r0/temp/run.do_compile.18806&apos; failed with exit code 1:</font>
<font color="#CC0000">[1/4] /home/matt/openbmc/build/tmp/work/arm1176jzs-openbmc-linux-gnueabi/bmcweb/1.0+gitAUTOINC+f8c3e6f0c2-r0/recipe-sysroot-native/usr/bin/arm-openbmc-linux-gnueabi/arm-openbmc-linux-gnueabi-g++  -DBMCWEB_ENABLE_DBUS_REST -DBMCWEB_ENABLE_HOST_SERIAL_WEBSOCKET -DBMCWEB_ENABLE_KVM -DBMCWEB_ENABLE_REDFISH -DBMCWEB_ENABLE_SSL -DBMCWEB_ENABLE_STATIC_HOSTING -DBMCWEB_ENABLE_VM_WEBSOCKET -DBOOST_ALL_NO_LIB -DBOOST_ASIO_DISABLE_THREADS -DBOOST_BEAST_USE_STD_STRING_VIEW -DBOOST_COROUTINES_NO_DEPRECATION_WARNING -DBOOST_ERROR_CODE_HEADER_ONLY -DBOOST_NO_RTTI -DBOOST_NO_TYPEID -DBOOST_SYSTEM_NO_DEPRECATED -I/home/matt/openbmc/build/tmp/work/arm1176jzs-openbmc-linux-gnueabi/bmcweb/1.0+gitAUTOINC+f8c3e6f0c2-r0/git/crow/include -I/home/matt/openbmc/build/tmp/work/arm1176jzs-openbmc-linux-gnueabi/bmcweb/1.0+gitAUTOINC+f8c3e6f0c2-r0/git/include -I/home/matt/openbmc/build/tmp/work/arm1176jzs-openbmc-linux-gnueabi/bmcweb/1.0+gitAUTOINC+f8c3e6f0c2-r0/git/redfish-core/include -Iinclude -marm -mcpu=arm1176jz-s  --sysroot=/home/matt/openbmc/build/tmp/work/arm1176jzs-openbmc-linux-gnueabi/bmcweb/1.0+gitAUTOINC+f8c3e6f0c2-r0/recipe-sysroot  -Os -pipe  -fvisibility-inlines-hidden  -marm -mcpu=arm1176jz-s  --sysroot=/home/matt/openbmc/build/tmp/work/arm1176jzs-openbmc-linux-gnueabi/bmcweb/1.0+gitAUTOINC+f8c3e6f0c2-r0/recipe-sysroot -fno-rtti -flto -fno-fat-lto-objects         -fvisibility=hidden         -fvisibility-inlines-hidden         -Wl,--exclude-libs,ALL   -Wno-attributes -std=gnu++17 -MD -MT CMakeFiles/bmcweb.dir/redfish-core/src/utils/json_utils.cpp.o -MF CMakeFiles/bmcweb.dir/redfish-core/src/utils/json_utils.cpp.o.d -o CMakeFiles/bmcweb.dir/redfish-core/src/utils/json_utils.cpp.o -c /home/matt/openbmc/build/tmp/work/arm1176jzs-openbmc-linux-gnueabi/bmcweb/1.0+gitAUTOINC+f8c3e6f0c2-r0/git/redfish-core/src/utils/json_utils.cpp</font>
<font color="#CC0000">[2/4] /home/matt/openbmc/build/tmp/work/arm1176jzs-openbmc-linux-gnueabi/bmcweb/1.0+gitAUTOINC+f8c3e6f0c2-r0/recipe-sysroot-native/usr/bin/arm-openbmc-linux-gnueabi/arm-openbmc-linux-gnueabi-g++  -DBMCWEB_ENABLE_DBUS_REST -DBMCWEB_ENABLE_HOST_SERIAL_WEBSOCKET -DBMCWEB_ENABLE_KVM -DBMCWEB_ENABLE_REDFISH -DBMCWEB_ENABLE_SSL -DBMCWEB_ENABLE_STATIC_HOSTING -DBMCWEB_ENABLE_VM_WEBSOCKET -DBOOST_ALL_NO_LIB -DBOOST_ASIO_DISABLE_THREADS -DBOOST_BEAST_USE_STD_STRING_VIEW -DBOOST_COROUTINES_NO_DEPRECATION_WARNING -DBOOST_ERROR_CODE_HEADER_ONLY -DBOOST_NO_RTTI -DBOOST_NO_TYPEID -DBOOST_SYSTEM_NO_DEPRECATED -I/home/matt/openbmc/build/tmp/work/arm1176jzs-openbmc-linux-gnueabi/bmcweb/1.0+gitAUTOINC+f8c3e6f0c2-r0/git/crow/include -I/home/matt/openbmc/build/tmp/work/arm1176jzs-openbmc-linux-gnueabi/bmcweb/1.0+gitAUTOINC+f8c3e6f0c2-r0/git/include -I/home/matt/openbmc/build/tmp/work/arm1176jzs-openbmc-linux-gnueabi/bmcweb/1.0+gitAUTOINC+f8c3e6f0c2-r0/git/redfish-core/include -Iinclude -marm -mcpu=arm1176jz-s  --sysroot=/home/matt/openbmc/build/tmp/work/arm1176jzs-openbmc-linux [... cut ...]
<b>NOTE</b>: Executing Tasks
<b>NOTE</b>: Setscene tasks completed
<font color="#FCE94F"><b>WARNING</b></font>: <font color="#C4A000">obmc-phosphor-sysd-1.0-r1 do_package_qa: QA Issue: obmc-phosphor-sysd: SRC_URI uses PN not BPN [src-uri-bad]</font>
<b>NOTE</b>: Tasks Summary: Attempted 4006 tasks of which 3279 didn&apos;t need to be rerun and all succeeded.

Summary: There were 10 WARNING messages shown.
<font color="#8AE234"><b>matt@marathon</b></font>:<font color="#729FCF"><b>~/openbmc</b></font>$ 

</pre>

geissonator, 2019-09-23:
That type of error:
```
/home/matt/openbmc/build/tmp/work/arm1176jzs-openbmc-linux-gnueabi/bmcweb/1.0+gitAUTOINC+f8c3e6f0c2-r0/recipe-sysroot-native/usr/bin/arm-openbmc-linux-gnueabi/arm-openbmc-linux-gnueabi-g++  -DBMCWEB_ENABLE_DBUS_REST -DBMCWEB_ENABLE_HOST_SERIAL_WEBSOCKET -DBMCWEB_ENABLE_KVM -DBMCWEB_ENABLE_REDFISH -DBMCWEB_ENABLE_SSL -DBMCWEB_ENABLE_STATIC_HOSTING -DBMCWEB_ENABLE_VM_WEBSOCKET -DBOOST_ALL_NO_LIB -DBOOST_ASIO_DISABLE_THREADS -DBOOST_BEAST_USE_STD_STRING_VIEW -DBOOST_COROUTINES_NO_DEPRECATION_WARNING -DBOOST_ERROR_CODE_HEADER_ONLY -DBOOST_NO_RTTI -DBOOST_NO_TYPEID -DBOOST_SYSTEM_NO_DEPRECATED -I/home/matt/openbmc/build/tmp/work/arm1176jzs-openbmc-linux-gnueabi/bmcweb/1.0+gitAUTOINC+f8c3e6f0c2-r0/git/crow/include -I/home/matt/openbmc/build/tmp/work/arm1176jzs-openbmc-linux-gnueabi/bmcweb/1.0+gitAUTOINC+f8c3e6f0c2-r0/git/include -I/home/matt/openbmc/build/tmp/work/arm1176jzs-openbmc-linux-gnueabi/bmcweb/1.0+gitAUTOINC+f8c3e6f0c2-r0/git/redfish-core/include -Iinclude -marm -mcpu=arm1176jz-s  --sysroot=/home/matt/openbmc/build/tmp/work/arm1176jzs-openbmc-linux-gnueabi/bmcweb/1.0+gitAUTOINC+f8c3e6f0c2-r0/recipe-sysroot  -Os -pipe  -fvisibility-inlines-hidden  -marm -mcpu=arm1176jz-s  --sysroot=/home/matt/openbmc/build/tmp/work/arm1176jzs-openbmc-linux-gnueabi/bmcweb/1.0+gitAUTOINC+f8c3e6f0c2-r0/recipe-sysroot -fno-rtti -flto -fno-fat-lto-objects         -fvisibility=hidden         -fvisibility-inlines-hidden         -Wl,--exclude-libs,ALL   -Wno-attributes -std=gnu++17 -MD -MT CMakeFiles/bmcweb.dir/src/webserver_main.cpp.o -MF CMakeFiles/bmcweb.dir/src/webserver_main.cpp.o.d -o CMakeFiles/bmcweb.dir/src/webserver_main.cpp.o -c /home/matt/openbmc/build/tmp/work/arm1176jzs-openbmc-linux-gnueabi/bmcweb/1.0+gitAUTOINC+f8c3e6f0c2-r0/git/src/webserver_main.cpp
arm-openbmc-linux-gnueabi-g++: fatal error: Killed signal terminated program cc1plus
compilation terminated.
```
Is usually associated with your system not having enough memory. What are the details of the computer you're building this on?

LiveFreeAndRoam, 2019-09-23:
Thanks Andrew.  I allocated a total of 4GB RAM and 4 threads to my VirtualBox VM.  This amounts to half of my system's resources.  I read the ideal minimum resources was 16GB, and 8-threads, but I'm working with what I have for now.

gtmills, 2019-11-16:
@LiveFreeAndRoam Still a problem or can this be closed? 
Thanks, 
Gunnar

LiveFreeAndRoam, 2019-11-17:
Hi Gunner,

The documentation is clear about minimum requirements, and I was operating
under those.  After allocating more memory (16GB) the build reliably
completes successfully.

It can be closed.

Cheers,
Matt.

closed 2019-11-17 by gtmills