openbmc/openbmc #3615: Consider removing OpenBMC.priv signing key from repo

bluecmd, 2019-10-09:
Hi,

Right now there appears to exist a publicly accessible private key in openbmc/meta-phosphor/recipes-phosphor/flash/files/OpenBMC.priv. This key I assume is used as a default signing key.

While it can be convenient to have a private key available, they make it easy for a vendor to accidentally ship OpenBMC with the default keys.

I recommend adding a step in the build documentation to generate a build key, or fetch the one that is intended for use. If no key is available, fail the build.

joseph-reynolds, 2020-04-09:
I agree.  The risk is having the certificate in the source tree.  I've added this issue to the OpenBMC Security Working group agenda - 
https://docs.google.com/document/d/1b7x9BaxsfcukQDqbvZsU2ehMq4xoJRQvLxxsDUWmAOI

As I see it, image signing is an important input to the build process.  What is the right way to supply the image signing certificate?
1. Supply the cert content in an environment variable?
2. Supply the cert filename in an environment variable?
3. Rename the default signing cert so builders have to handle the cert (for example, rename it back)?

Or can we change the project default to disable image signing by default?  Then we will need builder instructions to enable signing **and** supply the cert.

References:
- Proposed digital signature docs here: https://gerrit.openbmc-project.xyz/c/openbmc/phosphor-dbus-interfaces/+/9025
- Build steps are described here: https://github.com/openbmc/openbmc/blob/master/README.md
- Cheat sheet here: https://github.com/openbmc/docs/blob/master/cheatsheet.md

bradbishop, 2020-04-10:
at 12:31 PM, Joseph Reynolds <notifications@github.com> wrote:

> Or can we change the project default to disable image signing by default?

Is this any better than having a public signing key?

I would like to avoid needing to maintain an image signing setup while I am  
developing the bitbake metadata.  When/if that becomes the “out of box"  
experience can we also add an unsigned-image IMAGE_FEATURE and add it to  
debug-tweaks?

I’m willing to do that.  Just let me know when the work is occurring.

https://www.yoctoproject.org/docs/latest/mega-manual/mega-manual.html#ref-features-image

joseph-reynolds, 2020-04-14:
Yeah, it's not really better.

The proposal is:
- Add a new `IMAGE_FEATURE debug-tweaks += unsigned-image`, with default ON (meaning the image is unsigned)

bradbishop, 2020-04-15:
> Add a new IMAGE_FEATURE debug-tweaks += digitally-signed-image, with default OFF. <-- Edited from the original proposal to remove negative logic.

FWIW I don't have a preference on the default state of the flag.  Agree with the proposal with a minor tweak:

If the default is for unsigned images, we should add a new IMAGE_FEATURE called "signed-image" or something like that rather than adding it to debug-tweaks IMO.

If the default is is for signed images, we can add existing behavior to the existing debug-tweaks IMAGE_FEATURE.

joseph-reynolds, 2020-04-15:
We discussed this in the OpenBMC security working group 2020-04-15 https://docs.google.com/document/d/1b7x9BaxsfcukQDqbvZsU2ehMq4xoJRQvLxxsDUWmAOI
There was general agreement to disable this recipe by default, resulting in unsigned openbmc images.

A difficulty was identified when the image is not signed by the bitbake build process, and signed by a later process, for example a signed by a hardware security module (HSM).  Whoever signs the image will need to put the public key into the BMC's root file system.

The revised proposal:
1. Add a new IMAGE_FEATURE, signed-image, default OFF.  The docs for this feature would point to the example in the source tree, and give summary steps for using your own signing keys.
2. Keep the private key in the source tree as an example.  Note this key would be used if someone used the signed-image IMAGE_FEATURE without replacing the key.
3. Add docs for the BMC system integrator to optionally the signed-image IMAGE_FEATURE.  (Planned per https://github.com/ibm-openbmc/dev/issues/1531).

Any volunteers? :-)

closed 2020-11-15 by stale[bot]