openbmc/openbmc #3616: Document security vulnerabilities disclosure process

bluecmd, 2019-10-09:
Hi,

Consider adding a closed mailinglist as a point of contact at https://github.com/openbmc/openbmc#bug-reporting to help people to report security vulns without making them public by doing so.

E.g. right now if I were to figure out a remote code execution bug in OpenBMC, I would have a hard time to figure out where to send the report in a safe and responsible manner.

geissonator, 2019-10-09:
Please take a look at https://github.com/openbmc/docs/blob/master/security/how-to-report-a-security-vulnerability.md  @joseph-reynolds runs the openbmc security workgroup.

bluecmd, 2019-10-09:
Thanks, could that be added to the main README.md?

geissonator, 2019-10-10:
> Thanks, could that be added to the main README.md?

Sure, patches welcome. https://github.com/openbmc/docs/blob/master/CONTRIBUTING.md#submitting-changes-via-gerrit-server

joseph-reynolds, 2019-10-11:
Security vulnerability reports are welcome via the link above.  Let me know (or the email list openbmc@lists.ozlabs.org) if you have questions about the process. 

Adding a line for this to https://github.com/openbmc/openbmc/blob/master/README.md#bug-reporting would be a good way to get started contributing.  Let me know if you intend to do that, or else I will do it.

See also https://github.com/openbmc/openbmc/wiki/Security-working-group to discuss non-confidential OpenBMC security topics.

agnesamreetha, 2019-10-11:
I am beginner to openbmc & started working on openbmc codes. I would like to start contributing. Can I do this?

agnesamreetha, 2019-10-16:
Thank you Joseph for sharing me the details. I am an embedded firmware engineer working in different domains like networking, automotive, IoT & digital audio home domains. I am interested in learning and contributing through opensource communities. I have got a chance to know about OpenBMC from one of our customers and started working on it. 

Thanks again! I will share the patch for review soon.

gtmills, 2019-11-16:
@agnesamreetha Still have plans to submit something in this area?
Thanaks,
Gunnar

agnesamreetha, 2019-11-18:
@gtmills  Yes, I will commit this. I was working with CLA and CI authorization part to commit this patch. Now everything is done. I will commit this soon

gtmills, 2020-04-09:
Any luck here @agnesamreetha  ?

closed 2020-11-09 by stale[bot]