openbmc/openbmc #3617: IPMI add sel issue

Selvaganapathi-M, 2019-10-10:
Hi,

https://github.com/openbmc/openbmc/issues/3283#issuecomment-415782252

I have quistion from the above link.How to map association in code level.through yaml file file are any other way?

geissonator, 2019-10-15:
This seems more like a IRC or mailing list question then github issue. Please see https://github.com/openbmc/openbmc#contact for contact info.

closed 2020-05-15 by stale[bot]

sthanga, 2024-10-17:
root@ast2600-default:~# ipmitool sel add sel.txt                                                         
ipmi_sel_main: Inside Main SEL

ipmi_sel_print_std_entry: line no: 1776, evt->record_type: 2 

   0 |  Pre-Init  |0000000000| Voltage #0x0c | Upper Critical going high | Asserted
ipmi_sel_print_std_entry: line no: 1776, evt->record_type: 2 

   0 |  Pre-Init  |0000000000| Fan #0x06 | Lower Critical going low  | Asserted
ipmi_sel_print_std_entry: line no: 1776, evt->record_type: 2 

   0 |  Pre-Init  |0000000000| Fan #0x06 | Lower Non-critical going low  | Asserted
ipmi_sel_print_std_entry: line no: 1776, evt->record_type: 2 

   0 |  Pre-Init  |0000000000| Fan #0x05 | Lower Critical going low  | Asserted
ipmi_sel_print_std_entry: line no: 1776, evt->record_type: 2 

   0 |  Pre-Init  |0000000000| Fan #0x05 | Lower Non-critical going low  | Asserted
ipmi_sel_print_std_entry: line no: 1776, evt->record_type: 2 

   0 |  Pre-Init  |0000000000| Fan #0x07 | Lower Critical going low  | Asserted
ipmi_sel_print_std_entry: line no: 1776, evt->record_type: 2 

   0 |  Pre-Init  |0000000000| Fan #0x07 | Lower Non-critical going low  | Asserted
ipmi_sel_print_std_entry: line no: 1776, evt->record_type: 2 

   0 |  Pre-Init  |0000000000| Fan #0x04 | Lower Critical going low  | Asserted
ipmi_sel_print_std_entry: line no: 1776, evt->record_type: 2 

   0 |  Pre-Init  |0000000000| Fan #0x04 | Lower Non-critical going low  | Asserted
ipmi_sel_print_std_entry: line no: 1776, evt->record_type: 2 

   0 |  Pre-Init  |0000000000| Voltage #0x08 | Lower Non-critical going low  | Asserted
ipmi_sel_print_std_entry: line no: 1776, evt->record_type: 2 

   0 |  Pre-Init  |0000000000| Voltage #0x08 | Lower Critical going low  | Asserted
ipmi_sel_print_std_entry: line no: 1776, evt->record_type: 2 

   0 |  Pre-Init  |0000000000| Voltage #0x0a | Lower Non-critical going low  | Asserted
ipmi_sel_print_std_entry: line no: 1776, evt->record_type: 2 

   0 |  Pre-Init  |0000000000| Voltage #0x0a | Lower Critical going low  | Asserted
ipmi_sel_print_std_entry: line no: 1776, evt->record_type: 2 

   0 |  Pre-Init  |0000000000| Voltage #0x09 | Lower Non-critical going low  | Asserted
ipmi_sel_print_std_entry: line no: 1776, evt->record_type: 2 

   0 |  Pre-Init  |0000000000| Voltage #0x09 | Lower Critical going low  | Asserted
it is not adding into /var/log/ipmi_sel file. Going with 0 |  Pre-Init  |0000000000| , why i am getting like this all ?