openbmc/openbmc #3621: Auto-logout on serial console

bluecmd, 2019-10-15:
Hi,

I suggest setting the idle-terminal logout option by default in OpenBMC. This is for bash TMOUT.

It works like this:

```
palmetto login: root
Password: 
Last login: Tue Oct 15 08:41:14 UTC 2019 on ttyS4
root@palmetto:~# TMOUT=10
root@palmetto:~# timed out waiting for input: auto-logout

Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) 0.1.0 palmetto ttyS4

palmetto login: 
```

E.g. adding this to /etc/profile would be a nice way to enforce that forgotten serial consoles do not give an attacker automatic root shell.

I tested the above by doing:

```
root@palmetto:~# echo 'TMOUT=10' >> /etc/profile
```

Results:

```
palmetto login: root
Password: 
Last login: Tue Oct 15 08:42:09 UTC 2019 on ttyS4
root@palmetto:~# timed out waiting for input: auto-logout

Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) 0.1.0 palmetto ttyS4

palmetto login: 
```

Thanks for your consideration

edtanous, 2019-10-15:
Sounds reasonable to me.  Do you have a patchset put together for this to try out?

gtmills, 2019-11-16:
Sounds reasonable to me as well. @bluecmd Any plans to submit a patchset for this?

bluecmd, 2019-11-16:
Sadly no, the development process is too much hassle for me - I'll have to try to contribute through outer means like bug reports I guess.

joseph-reynolds, 2020-02-19:
Thanks for your report!  I've added this to my big bucket of work https://github.com/ibm-openbmc/dev/issues/242 and will try to pick this up.

joseph-reynolds, 2020-08-26:
This still seems worthwhile. But the solution is for bash and the project default shell is  busybox.   Here is what I found for busy box: https://busybox.busybox.narkive.com/wdLOmbCn/idle-or-inactivity-timeout-in-ash

joseph-reynolds, 2020-08-26:
Maybe we can use: ASH_IDLE_TIMEOUT with TMOUT
https://github.com/brgl/busybox/blob/abbf17abccbf832365d9acf1c280369ba7d5f8b2/shell/ash.c#L85

gtmills, 2021-06-03:
Still hasn't been done.

matthewfischer, 2024-12-18:
I pushed a fix for this here. https://gerrit.openbmc.org/c/openbmc/openbmc/+/76855

After looking at the options, I think the PAM option is the cleanest since it limits it to tty only cleanly.

matthewfischer, 2024-12-18:
Results:

```root@romulus:/etc/pam.d#
root@romulus:/etc/pam.d# env
...
TMOUT=300
root@romulus:/etc/pam.d# timed out waiting for input: auto-logout

Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) nodistro.0 romulus ttyS4

romulus login:

closed 2026-09-18 by geissonator