openbmc/openbmc #3623: Unable to start bitbake server with "meta-evb-zx3-pm3"

onjustgo, 2019-10-23:
**Note on Bugs**
I get an error when I set the 'TEMPLATECONF' to 'meta-evb/meta-evb-enclustra/meta-evb-zx3-pm3/conf' in branch master.

**Unexpected behavior you saw**
ERROR: Unable to start bitbake server (None)
ERROR: Server log for this session (/home/ubu/openbmc-master/build/bitbake-cookerdaemon.log):
--- Starting bitbake server pid 6244 at 2019-10-23 11:18:49.590029 ---
ERROR: Layer xilinx is not compatible with the core layer which only supports these series: zeus (layer is compatible with warrior)

geissonator, 2019-10-23:
@bradbishop looks like both meta-xilinx and meta-evb probably need an update to zeus?

bradbishop, 2019-10-24:
Yes.  I would accept patches that do that.

> On Oct 23, 2019, at 8:34 AM, Andrew Geissler <notifications@github.com> wrote:
> 
> @bradbishop looks like both meta-xilinx and meta-evb probably need an update to zeus?

closed 2020-05-25 by stale[bot]

dilraj-n, 2020-06-03:
When I tried to build. I got the following error
ERROR: Unable to start bitbake server (None)
ERROR: Server log for this session openbmc/xilinx-build/bitbake-cookerdaemon.log):
--- Starting bitbake server pid 15393 at 2020-06-03 14:40:39.829943 ---
ERROR: Layer xilinx is not compatible with the core layer which only supports these series: dunfell gatesgarth (layer is compatible with warrior)