openbmc/openbmc #3624: OpenBMC 443 port renegotiation attack (CVE-2011-1473)

Journey-x, 2019-10-25:
I used the following command to confirm that there is renegotiation attack.

openssl s_client -connect 127.0.0.1:2443
CONNECTED(00000003)
depth=0 C = US, O = OpenBMC, CN = testhost
verify error:num=18:self signed certificate
verify return:1
depth=0 C = US, O = OpenBMC, CN = testhost
verify return:1
---
Certificate chain
 0 s:/C=US/O=OpenBMC/CN=testhost
   i:/C=US/O=OpenBMC/CN=testhost
---
Server certificate
-----BEGIN CERTIFICATE-----
MIIBTjCB9QIENt8/3DAKBggqhkjOPQQDAjAyMQswCQYDVQQGEwJVUzEQMA4GA1UE
CgwHT3BlbkJNQzERMA8GA1UEAwwIdGVzdGhvc3QwHhcNMTkxMDEwMDkyOTU2WhcN
MjkxMDA3MDkyOTU2WjAyMQswCQYDVQQGEwJVUzEQMA4GA1UECgwHT3BlbkJNQzER
MA8GA1UEAwwIdGVzdGhvc3QwWTATBgcqhkjOPQIBBggqhkjOPQMBBwNCAARlgvcq
1nSbmcG3xsDITv9ySJtTjH3kO40DWRANPzNT2RIjfzIo2XE6ZRHezqrCfBxL5rFr
pAko3Q4ScVb1e1T2MAoGCCqGSM49BAMCA0gAMEUCICke0szn05duDb6n9J21+yYi
Emeud7ED//KYWUrL9VAJAiEA1rvc3f/BeBZe733X7BZlS5EYpG++3kGXcxAcgQEY
db8=
-----END CERTIFICATE-----
subject=/C=US/O=OpenBMC/CN=testhost
issuer=/C=US/O=OpenBMC/CN=testhost
---
No client certificate CA names sent
Peer signing digest: SHA512
Server Temp Key: ECDH, P-256, 256 bits
---
SSL handshake has read 806 bytes and written 431 bytes
---
New, TLSv1/SSLv3, Cipher is ECDHE-ECDSA-AES256-GCM-SHA384
Server public key is 256 bit
Secure Renegotiation IS supported
Compression: NONE
Expansion: NONE
No ALPN negotiated
SSL-Session:
    Protocol  : TLSv1.2
    Cipher    : ECDHE-ECDSA-AES256-GCM-SHA384
    Session-ID: 5BBC43957F3F38CE337A6CCC5616BC4A90AE2CD9F447BF59208DFFE4989B54D1
    Session-ID-ctx:
    Master-Key: 7BEB2AB542182B9780F74AEE847A74E88CEA1321321C3A57E7EDB0CF8658597B1285AE29C202AEED204D82FAFF129BF4
    Key-Arg   : None
    PSK identity: None
    PSK identity hint: None
    SRP username: None
    TLS session ticket lifetime hint: 7200 (seconds)
    TLS session ticket:
    0000 - ee c2 89 68 86 d9 25 88-3d 01 97 0b c9 04 53 d2   ...h..%.=.....S.
    0010 - bd 0a 2f 3a 6a c8 5c 82-49 c7 6c 76 08 b3 dc 96   ../:j.\.I.lv....
    0020 - 00 bb 51 c5 13 ff dd d3-ae 11 e5 57 3e ab 2f 0c   ..Q........W>./.
    0030 - 4b 9f 7a ba 20 ea 0f 25-ea 34 ac 3b 37 0b 22 07   K.z. ..%.4.;7.".
    0040 - b1 a6 5e 68 f5 cc d3 79-a9 bc f8 2d 6d ce c6 00   ..^h...y...-m...
    0050 - 49 15 5f a1 9c 8b 83 74-73 54 e9 28 0c 0b 2c 53   I._....tsT.(..,S
    0060 - df fb 5f 4b f5 85 8b 85-1d 43 f5 ce 15 91 f2 e1   .._K.....C......
    0070 - 5d 67 ae a9 a8 85 98 90-55 05 51 35 e3 60 81 ac   ]g......U.Q5.`..
    0080 - f0 8b f6 bc 7c b3 f6 4e-43 86 93 c6 4f 70 ea ff   ....|..NC...Op..
    0090 - 3f 67 f3 60 2b 21 f0 88-67 84 3b 82 78 34 28 01   ?g.`+!..g.;.x4(.

    Start Time: 1571971719
    Timeout   : 300 (sec)
    Verify return code: 18 (self signed certificate)
---
R
RENEGOTIATING
depth=0 C = US, O = OpenBMC, CN = testhost
verify error:num=18:self signed certificate
verify return:1
depth=0 C = US, O = OpenBMC, CN = testhost
verify return:1
R
RENEGOTIATING
depth=0 C = US, O = OpenBMC, CN = testhost
verify error:num=18:self signed certificate
verify return:1
depth=0 C = US, O = OpenBMC, CN = testhost
verify return:1
R
RENEGOTIATING
depth=0 C = US, O = OpenBMC, CN = testhost
verify error:num=18:self signed certificate
verify return:1
depth=0 C = US, O = OpenBMC, CN = testhost
verify return:1

geissonator, 2019-10-25:
FYI @joseph-reynolds

joseph-reynolds, 2019-10-29:
I've added an entry to discuss this at the next OpenBMC security working group meeting.  See
https://docs.google.com/document/d/1b7x9BaxsfcukQDqbvZsU2ehMq4xoJRQvLxxsDUWmAOI
and see the accompanying wiki for meeting information.  The most recent notice is: https://lists.ozlabs.org/pipermail/openbmc/2019-October/019192.html

joseph-reynolds, 2019-10-30:
We discussed this in today's OpenBMC security working group meeting.  The notes from that meeting are in the minutes, linked above.  A few thoughts:

This is another of OpenBMC's weaknesses in preventing denial of service (DOS) attacks.  The topic of DOS attacks should be added to the OpenBMC network threat model:
https://github.com/openbmc/docs/blob/master/security/network-security-considerations.md
DOS protection was added as an initiative to the https://github.com/openbmc/openbmc/wiki/Security-working-group.

Denial of service has been discussed in the OpenBMC public email list.  For example, search for the words "DOS" or "denial" within the https://lists.ozlabs.org/pipermail/openbmc site:
- Google search: DOS site:https://lists.ozlabs.org/pipermail/openbmc/
- Google search: denial site:https://lists.ozlabs.org/pipermail/openbmc/

joseph-reynolds, 2019-10-30:
I understand OpenSSL wants to allow for renegotiation, like "it's a feature, not a DoS security risk."  In this view, it is up the OpenSSL client (such as OpenBMC interfaces including SSH, HTTPS, IPMI, and other cert-related activity such as LDAP) to disable the renegotiation feature.

However, I am not a network security expert.  I found a blog that seems to explain the situation pretty well: http://orchilles.com/ssl-renegotiation-dos/ 

I think the proposal for OpenBMC would be to use something like: `SSL_CTX_set_options(..., SSL_OP_NO_RENEGOTIATION)`. Would that solve the problem?  Would it cause connection problems?

feistjj, 2019-11-07:
This got brought up on the mailing list, and I created a patch for it

https://lists.ozlabs.org/pipermail/openbmc/2019-November/019374.html

https://gerrit.openbmc-project.xyz/c/openbmc/bmcweb/+/26992

Journey-x, 2019-11-11:
I tested this attact again, It was fixed. I will close this issue.

**openssl s_client -connect 127.0.0.1:2443**
CONNECTED(00000003)
depth=0 C = US, O = OpenBMC, CN = testhost
verify error:num=18:self signed certificate
verify return:1
depth=0 C = US, O = OpenBMC, CN = testhost
verify return:1
---
Certificate chain
 0 s:/C=US/O=OpenBMC/CN=testhost
   i:/C=US/O=OpenBMC/CN=testhost
---
Server certificate
-----BEGIN CERTIFICATE-----
MIIBTzCB9QIE9b2+DjAKBggqhkjOPQQDAjAyMQswCQYDVQQGEwJVUzEQMA4GA1UE
CgwHT3BlbkJNQzERMA8GA1UEAwwIdGVzdGhvc3QwHhcNMTkwOTIwMDc1MTQ2WhcN
MjkwOTE3MDc1MTQ2WjAyMQswCQYDVQQGEwJVUzEQMA4GA1UECgwHT3BlbkJNQzER
MA8GA1UEAwwIdGVzdGhvc3QwWTATBgcqhkjOPQIBBggqhkjOPQMBBwNCAARtNVN/
90d5TtOAhfLXnbJCSHBZm7qkUGEuvu/OIpqJ+Hh9EbnJam44sgJ0Eo5eKACGBcYH
fl5xQ1eX8ySonF3cMAoGCCqGSM49BAMCA0kAMEYCIQDGn4++lb7w+gdUU/ckmWdf
8QVFZBRT38MSef608qUOvQIhAP2nLIaejuvmyLhrG60Fb/dgU+tRdTEwCVqrCJRm
Bd3u
-----END CERTIFICATE-----
subject=/C=US/O=OpenBMC/CN=testhost
issuer=/C=US/O=OpenBMC/CN=testhost
---
No client certificate CA names sent
---
SSL handshake has read 875 bytes and written 489 bytes
---
New, TLSv1/SSLv3, Cipher is ECDHE-ECDSA-AES256-GCM-SHA384
Server public key is 256 bit
Secure Renegotiation IS supported
Compression: NONE
Expansion: NONE
SSL-Session:
    Protocol  : TLSv1.2
    Cipher    : ECDHE-ECDSA-AES256-GCM-SHA384
    Session-ID: 51BDE044C90F1C5DCAC6479547B5519B19989FF2C64489ED52C7FF8B2BC971F1
    Session-ID-ctx:
    Master-Key: 87B68F78BC46BB7D36177DB1CC03698AE9C96545B15378C05515FAB34B27C249A138FE2D76C03B6F02248BA1E50AA3C3
    Key-Arg   : None
    PSK identity: None
    PSK identity hint: None
    SRP username: None
    TLS session ticket lifetime hint: 7200 (seconds)
    TLS session ticket:
    0000 - 87 4d 87 b3 93 a7 9f 35-c9 25 7b e6 e8 58 e5 5d   .M.....5.%{..X.]
    0010 - 3f 62 6e 9f b7 b3 f3 cb-1f da dd 67 27 cd 64 09   ?bn........g'.d.
    0020 - 7a e7 a9 76 75 c7 dd 0e-b9 63 7d 96 2b 0a 5d 8e   z..vu....c}.+.].
    0030 - 7f a4 f2 49 47 66 60 24-d4 7b a2 fe d6 30 d3 3e   ...IGf`$.{...0.>
    0040 - e9 62 97 76 a7 08 1d 9c-44 0e d4 49 b7 a6 8a 06   .b.v....D..I....
    0050 - 29 d3 82 56 95 91 31 7d-be d4 ff 80 81 fe b6 f8   )..V..1}........
    0060 - 4b 71 41 1b 46 eb 66 d6-97 6a 07 6d 03 2f a8 7e   KqA.F.f..j.m./.~
    0070 - c3 67 66 73 4a 61 ad 4c-3d 9b 8e 01 b6 a4 75 dc   .gfsJa.L=.....u.
    0080 - 48 53 8c ca 2e d4 e6 9b-99 8e 44 79 90 06 3b dc   HS........Dy..;.
    0090 - c4 a9 42 28 fc ce f5 46-7c 07 8e f3 8d 2e 84 ae   ..B(...F|.......

    Start Time: 1573437312
    Timeout   : 300 (sec)
    Verify return code: 18 (self signed certificate)
---
R
RENEGOTIATING
140481282262688:error:14094153:SSL routines:SSL3_READ_BYTES:no renegotiation:s3_pkt.c:1264:

joseph-reynolds, 2019-11-11:
Thanks to everyone who worked on this.   We should keep this issue open until the fix is merged into BMCWeb.   And I understand that’s waiting for some groups to finish testing.

feistjj, 2019-11-11:
> And I understand that’s waiting for some groups to finish testing.

I'm finished with my testing, it's waiting on someone to +1 the patch.

closed 2019-11-12 by bradbishop