openbmc/openbmc #3625: why not i build openbmc,prompt this error?

zw-fxm, 2019-10-29:
root@zw-Default-string:/openbmc/openbmc/build# bitbake obmc-phosphor-image
ERROR: Unable to start bitbake server (None)
ERROR: Server log for this session (/openbmc/openbmc/build/bitbake-cookerdaemon.log):
--- Starting bitbake server pid 16794 at 2019-10-29 10:30:45.453915 ---
ERROR: The following required tools (as specified by HOSTTOOLS) appear to be unavailable in PATH, please install them in order to proceed:
  realpath

zw-fxm, 2019-10-29:
hi veryone i need help! 
thanks

zw-fxm, 2019-10-29:
Now  i apt-get install realpath ,and then i bulid,but has errors:
OE-core's config sanity checker detected a potential misconfiguration.
    Either fix the cause of this error or at your own risk disable the checker (see sanity.conf).
    Following is the list of potential problems / advisories:

    Do not use Bitbake as root
so what can i do ???

zw-fxm, 2019-10-29:
my system is ubuntu 14.04.

zw-fxm, 2019-10-29:
Hi all：
 I can use user mode in my system build the openbmc,
Notes:
   zw@zw-Default-string:~/openbmc$ export TEMPLATECONF=meta-ibm/meta-romulus/conf
zw@zw-Default-string:~/openbmc$ . openbmc-env 
### Initializing OE build env ###
mkdir: cannot create directory ‘/home/zw/openbmc/build’: Permission denied
Error: The builddir (/home/zw/openbmc/build) does not exist!
 this prompt use root mode，but  I use root mode ,setup the bitbake obmc-phosphor-image cmd
,is prompt Do not use Bitbake as root?? 
 so what can i do?
Thanks
David

mine260309, 2019-10-30:
> mkdir: cannot create directory ‘/home/zw/openbmc/build’: Permission denied

This means you have permission issue, fix that and try build again.

closed 2019-11-08 by zw-fxm