openbmc/openbmc #3626: Net issue：Cannot ping to BMC from another computer

ChaoyangZhang, 2019-11-01:
BMC（AST2500） link to remote computer by a netline
1、BMC（ast2500） config：
ifconfig eth0 192.168.0.105 netmask 255.255.255.0

2、remote computer config:
ifconfig eth0 192.168.0.100 netmask 255.255.255.0

In remote computer:
ping 192.168.0.105
From 192.168.0.100 icmp_seq=1 Destination Host Unreadchable
From 192.168.0.100 icmp_seq=2 Destination Host Unreadchable
...
Please help me, thank you

ChaoyangZhang, 2019-11-01:
Add the following log：
[  152.085386] WARNING: CPU: 0 PID: 0 at net/sched/sch_generic.c:461 dev_watchdog+0x230/0x24c
[  152.093758] NETDEV WATCHDOG: eth0 (ftgmac100): transmit queue 0 timed out
[  152.100571] CPU: 0 PID: 0 Comm: swapper Not tainted 4.19.16-bbc22aa2953e0330ec9b43f1b866a59ec53fa884 #1
[  152.110031] Hardware name: Generic DT based system
[  152.114895] Backtrace:
[  152.117412] [<8010808c>] (dump_backtrace) from [<801082b0>] (show_stack+0x20/0x24)
[  152.125062]  r7:8052e2ac r6:00000009 r5:00000000 r4:80a01d54
[  152.130820] [<80108290>] (show_stack) from [<8061754c>] (dump_stack+0x20/0x28)
[  152.138073] [<8061752c>] (dump_stack) from [<80116690>] (__warn.part.3+0xb4/0xdc)
[  152.145653] [<801165dc>] (__warn.part.3) from [<80116724>] (warn_slowpath_fmt+0x6c/0x90)
[  152.153821]  r6:000001cd r5:808209d4 r4:80a03008
[  152.158482] [<801166bc>] (warn_slowpath_fmt) from [<8052e2ac>] (dev_watchdog+0x230/0x24c)
[  152.166737]  r3:9e2e3800 r2:80820998
[  152.170330]  r7:80a12ac0 r6:9e2e3800 r5:00000000 r4:9e2e3a2c
[  152.176101] [<8052e07c>] (dev_watchdog) from [<80157030>] (call_timer_fn+0x3c/0x120)
[  152.183935]  r7:8052e07c r6:00000100 r5:9e2e3a2c r4:9e2e3a2c
[  152.189635] [<80156ff4>] (call_timer_fn) from [<801571c0>] (expire_timers+0xac/0xb8)
[  152.197464]  r7:00000000 r6:80a01e04 r5:80a01e04 r4:9e2e3a2c
[  152.203229] [<80157114>] (expire_timers) from [<80157268>] (run_timer_softirq+0x9c/0x190)
[  152.211481]  r9:80a03008 r8:80a12ac0 r7:80a137c0 r6:80a137c0 r5:80a01e04 r4:80a01e04
[  152.219258] [<801571cc>] (run_timer_softirq) from [<8010224c>] (__do_softirq+0xdc/0x2f8)
[  152.227436]  r9:00000100 r8:9e042600 r7:ffffe000 r6:80a5e890 r5:00000002 r4:00000001
[  152.235280] [<80102170>] (__do_softirq) from [<8011a6d8>] (irq_exit+0xfc/0x110)
[  152.242677]  r10:9eeffa00 r9:80a00000 r8:9e042600 r7:00000000 r6:00000001 r5:80a41500
[  152.250516]  r4:00000000
[  152.253162] [<8011a5dc>] (irq_exit) from [<8014af84>] (__handle_domain_irq+0x60/0xb8)
[  152.261081] [<8014af24>] (__handle_domain_irq) from [<80102164>] (avic_handle_irq+0x68/0x70)
[  152.269532]  r9:80a00000 r8:807c7ae4 r7:80a01f1c r6:ffffffff r5:80a01ee8 r4:9e002620
[  152.277363] [<801020fc>] (avic_handle_irq) from [<801019ec>] (__irq_svc+0x6c/0x90)
[  152.285012] Exception stack(0x80a01ee8 to 0x80a01f30)
[  152.290079] 1ee0:                   00000000 00000000 00000000 00000000 ffffe000 00000000
[  152.298332] 1f00: 80a03068 80a442b7 807c7ae4 00000001 9eeffa00 80a01f44 80a01f48 80a01f38
[  152.306583] 1f20: 801039d0 801039d4 60000013 ffffffff
[  152.311698]  r5:60000013 r4:801039d4
[  152.315323] [<8010399c>] (arch_cpu_idle) from [<80632508>] (default_idle_call+0x38/0x3c)
[  152.323524] [<806324d0>] (default_idle_call) from [<8013ea58>] (do_idle+0xa4/0xec)
[  152.331185] [<8013e9b4>] (do_idle) from [<8013ed44>] (cpu_startup_entry+0x1c/0x20)
[  152.338771]  r9:80936a30 r8:80a03008 r7:80a03000 r6:ffffffff r5:00000001 r4:80a0caf8
[  152.346601] [<8013ed28>] (cpu_startup_entry) from [<8062beb4>] (rest_init+0x7c/0x9c)
[  152.354454] [<8062be38>] (rest_init) from [<80901118>] (start_kernel+0x3fc/0x4a8)
[  152.362047]  r5:00000001 r4:80a5e1ec
[  152.365659] [<80900d1c>] (start_kernel) from [<00000000>] (  (null))
[  152.372092] ---[ end trace e60a2a37043322f1 ]---

manikandan-e, 2019-12-31:
Hi ChaoyangZhang,

Please provide the NIC card vendor details such as Mellanox/Broadcom/Intel or other.
I understand that the NCSI Driver involved here and if so, please share the NCSI command request/response log.

michealylj1, 2020-02-12:
Hi All,

I also meet the same issue in our system. We use the RTL 8211 phy connected with RGMII interface in the AST2520. The log as:

root@switch:~# cat /etc/os-release 
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="0.1.0"
VERSION_ID="0.1.0"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) 0.1.0"
BUILD_ID="20191217092514"
OPENBMC_TARGET_MACHINE="evb-ast2500"
root@switch:~# uname -a
Linux switch 5.0.7-b1b37a25644213428f4bab4c427581dd923e35da #1 Tue Dec 17 10:12:51 UTC 2019 armv6l GNU/Linux
root@switch:~# dmesg | grep -i net             
[    0.031165] NET: Registered protocol family 16
[    0.212955] NET: Registered protocol family 2
[    0.215132] NET: Registered protocol family 1
[    2.091311] NET: Registered protocol family 38
[    3.448717] ftgmac100 1e680000.ethernet: Read MAC address 00:0c:31:10:4b:df from chip
[    3.489347] RTL8211F Gigabit Ethernet 1e680000.ethernet--1:00: attached PHY driver [RTL8211F Gigabit Ethernet] (mii_bus:phy_addr=1e680000.ethernet--1:00, irq=POLL)
[    3.505065] ftgmac100 1e680000.ethernet eth0: irq 19, mapped at (ptrval)
[    3.512448] ftgmac100 1e660000.ethernet: Read MAC address 00:0c:3d:d9:ba:4e from chip
[    3.846751] Generic PHY 1e660000.ethernet--1:00: attached PHY driver [Generic PHY] (mii_bus:phy_addr=1e660000.ethernet--1:00, irq=POLL)
[    3.860029] ftgmac100 1e660000.ethernet eth1: irq 20, mapped at (ptrval)
[    4.208363] Driver for 1-wire Dallas network protocol.
[    4.243174] NET: Registered protocol family 10
[    4.272199] NET: Registered protocol family 17
[    4.292316] printk: console [netcon0] enabled
[    4.296795] netconsole: network logging started
[   10.109892] systemd[1]: Listening on Network Service Netlink Socket.
[   17.451279] ftgmac100 1e660000.ethernet eth1: Link is Up - 10Mbps/Half - flow control off
[   17.566706] IPv6: ADDRCONF(NETDEV_CHANGE): eth1: link becomes ready
[   18.486822] ftgmac100 1e660000.ethernet eth1: Link is Up - 1Gbps/Full - flow control off
[   20.967346] ftgmac100 1e680000.ethernet eth0: Link is Up - 100Mbps/Full - flow control rx/tx
[   20.975854] IPv6: ADDRCONF(NETDEV_CHANGE): eth0: link becomes ready
[   26.887778] ftgmac100 1e660000.ethernet eth1: Link is Down
[   48.367937] ftgmac100 1e660000.ethernet eth1: Link is Up - 10Mbps/Half - flow control off
[   48.376184] IPv6: ADDRCONF(NETDEV_CHANGE): eth1: link becomes ready
[   48.478678] IPv6: ADDRCONF(NETDEV_CHANGE): eth1.3951: link becomes ready
[   48.485860] IPv6: ADDRCONF(NETDEV_CHANGE): eth1.3952: link becomes ready
[   48.493163] IPv6: ADDRCONF(NETDEV_CHANGE): eth1.3953: link becomes ready
[   48.500444] IPv6: ADDRCONF(NETDEV_CHANGE): eth1.3954: link becomes ready
[   50.327235] ftgmac100 1e680000.ethernet eth0: Link is Up - 100Mbps/Full - flow control rx/tx
[   50.335741] IPv6: ADDRCONF(NETDEV_CHANGE): eth0: link becomes ready
[   74.091285] WARNING: CPU: 0 PID: 0 at net/sched/sch_generic.c:461 dev_watchdog+0x230/0x24c
[   74.099547] NETDEV WATCHDOG: eth1 (ftgmac100): transmit queue 0 timed out
[1799258.804141] ftgmac100 1e680000.ethernet eth0: Link is Down
[1799262.964153] ftgmac100 1e680000.ethernet eth0: Link is Up - 100Mbps/Full - flow control rx/tx
root@switch:~# 

the dump stack log as:

[   20.967346] ftgmac100 1e680000.ethernet eth0: Link is Up - 100Mbps/Full - flow control rx/tx
[   20.975854] IPv6: ADDRCONF(NETDEV_CHANGE): eth0: link becomes ready
[   21.125431] cs5643_48p_psu 6-0053: psu msa 'cs5643_48p_psu1'
[   21.131407] i2c i2c-6: new_device: Instantiated device cs5643_48p_psu1 at 0x53
[   21.233323] cs5643_48p_psu 6-005b: new chip found
[   21.238297] cs5643_48p_psu 6-005b: hwmon_device_register() is deprecated. Please convert the driver to use hwmon_device_register_with_info().
[   21.251510] i2c i2c-6: new_device: Instantiated device cs5643_48p_psu1 at 0x5b
[   21.681949] aspeed-g5-pinctrl 1e6e2000.syscon:pinctrl: request() failed for pin 192
[   21.689742] aspeed-g5-pinctrl 1e6e2000.syscon:pinctrl: pin-192 (1e780000.gpio:472) status -1
[   21.898970] i2c i2c-0: new_device: Instantiated device pca9534 at 0x27
[   21.997062] at24 0-0050: 512 byte 24c04 EEPROM, writable, 1 bytes/write
[   22.003863] i2c i2c-0: new_device: Instantiated device 24c04 at 0x50
[   22.117903] lm75 0-0049: hwmon3: sensor 'tmp75'
[   22.122626] i2c i2c-0: new_device: Instantiated device tmp75 at 0x49
[   22.221278] aspeed-g5-pinctrl 1e6e2000.syscon:pinctrl: request() failed for pin 193
[   22.229070] aspeed-g5-pinctrl 1e6e2000.syscon:pinctrl: pin-193 (1e780000.gpio:473) status -1
[   22.398416] i2c i2c-1: new_device: Instantiated device pca9534 at 0x27
[... 55 lines cut ...]
[   74.273023] [<8014cca8>] (__handle_domain_irq) from [<80102164>] (avic_handle_irq+0x68/0x70)
[   74.281455]  r9:80a00000 r8:807dcb64 r7:80a01f0c r6:ffffffff r5:80a01ed8 r4:8e402480
[   74.289263] [<801020fc>] (avic_handle_irq) from [<801019ec>] (__irq_svc+0x6c/0x90)
[   74.296836] Exception stack(0x80a01ed8 to 0x80a01f20)
[   74.301888] 1ec0:                                                       00000000 00000000
[   74.310063] 1ee0: 8f808000 00000000 ffffe000 00000000 80a07088 80a51db8 807dcb64 00000001
[   74.318236] 1f00: 8efffd20 80a01f34 80a01f38 80a01f28 80103de0 80103de4 60000013 ffffffff
[   74.326402]  r5:60000013 r4:80103de4
[   74.329999] [<80103dac>] (arch_cpu_idle) from [<806cdb90>] (default_idle_call+0x38/0x3c)
[... cut ...]
[   74.368758] [<806c77e0>] (rest_init) from [<80900b48>] (arch_call_rest_init+0x18/0x1c)
[   74.376670]  r5:00000001 r4:80a6760c
[   74.380252] [<80900b30>] (arch_call_rest_init) from [<80900fa8>] (start_kernel+0x3fc/0x4ac)
[   74.388600] [<80900bac>] (start_kernel) from [<00000000>] (  (null))
[   74.394948] ---[ end trace a45a47185877900b ]---
[   92.405624] random: crng init done
[   92.409164] random: 7 urandom warning(s) missed due to ratelimiting

Please tell me how to fix the issue.

Thanks,
Michael

closed 2020-09-12 by stale[bot]

swimBirds, 2020-11-12:
I also meet the same issue in our system.