openbmc/openbmc #3627: Systemd unit variables error in installing from repo

mine260309, 2019-11-04:
Issues are found in systemd units that are installed from repo, that the variables in the unit are not replaced by value, but the raw text is kept.

Examples are:
* phosphor-gpio-monitor
   ```
   # phosphor-gpio-monitor@.service
   ExecStart=/usr/bin/phosphor-gpio-monitor --path=${{DEVPATH}} --key=${{KEY}} --polarity=${{POLARITY}} --target=${{TARGET}} ${{EXTRA_ARGS}}
   ```
* phosphor-power:
   ```
   #power-supply-monitor@.service
   EnvironmentFile={envfiledir}/obmc/power-supply-monitor/power-supply-monitor-%i.conf
   ExecStart=/usr/bin/env psu-monitor --path=${{DEVPATH}} --instance=${{INSTANCE}} --inventory=${{INVENTORY}} --num-history-records=${{NUM_HISTORY_RECORDS}} --sync-gpio-path=${{SYNC_GPIO_PATH}} --sync-gpio-num=${{SYNC_GPIO_NUM}}
   ```

Be noted that the services contain variables like `${{DEVPATH}}`, where the expected variables should be `${DEVPATH}`
The services started on BMC will get incorrect variables and fail to start.

Before moving to repos, the services were installed in openbmc tree by `obmc-phosphor-systemd.bbclass`, which replaces such variables and it worked fine.

mine260309, 2019-11-04:
This issue impacts meta-witherspoon for below services:
* phosphor-gpio-monitor@.service
* phosphor-gpio-presence@.service
* power-supply-monitor@.service

Not sure if there are other services are impacted by not built for witherspoon.

The fix could either be modifying the services in repo, or trying to making them correct during installation.

@bradbishop any ideas?

devenrao, 2019-11-04:
brad has done some thing similar here https://github.com/openbmc/openbmc/blob/master/meta-phosphor/recipes-phosphor/certificate/phosphor-bmcweb-cert-config.bb

mine260309, 2019-11-04:
> brad has done some thing similar here https://github.com/openbmc/openbmc/blob/master/meta-phosphor/recipes-phosphor/certificate/phosphor-bmcweb-cert-config.bb

It's different, the above adds the systemd links, but not resolves the variables issue.

devenrao, 2019-11-04:
the variables are defined in an env file which is referred in the service
https://github.com/openbmc/openbmc/blob/master/meta-phosphor/recipes-phosphor/certificate/phosphor-certificate-manager/phosphor-certificate-manager%40.service

mine260309, 2019-11-04:
Please refer to the issue description:

> Before moving to repos, the services were installed in openbmc tree by obmc-phosphor-systemd.bbclass, which replaces such variables and it worked fine.

The `phosphor-certificate-manager@.service` is in openbmc tree, so that is OK.
This issue is talking about the services that are in repo instead of in openbmc tree.

geissonator, 2019-11-04:
The gpio fixes are up at https://gerrit.openbmc-project.xyz/c/openbmc/phosphor-gpio-monitor/+/26798

geissonator, 2019-11-04:
Looks like we haven't switched over to the power-supply-monitor@.service in phosphor-power yet? I fixed it up with https://gerrit.openbmc-project.xyz/c/openbmc/phosphor-power/+/26838

bradbishop, 2019-11-04:
> @bradbishop any ideas?

@mine260309 do you have a good handle on the problem and resolution for this one?

mine260309, 2019-11-05:
Thanks for @geissonator to fix this.

closed 2019-11-05 by mine260309