openbmc/openbmc #3630: phosphor-hwmon creates fans failure

z1908144712, 2019-11-07:
When I add fan sensors, the photospor-hwmon doesn't work properly. My BMC chip is AST2400, here is my configuration.

My device tree configuration:
```
&pwm_tacho {
	status = "okay";

	pinctrl-names = "default";
	pinctrl-0 = <&pinctrl_pwm0_default
		&pinctrl_pwm1_default
		&pinctrl_pwm2_default
		&pinctrl_pwm3_default
		&pinctrl_pwm4_default
		&pinctrl_pwm5_default
		&pinctrl_pwm6_default
		&pinctrl_pwm7_default>;

	fan@0 {
		reg = <0x00>;
		aspeed,fan-tach-ch = /bits/ 8 <0x00>;
	};

	fan@1 {
		reg = <0x01>;
		aspeed,fan-tach-ch = /bits/ 8 <0x01>;
	};

	fan@2 {
		reg = <0x04>;
		aspeed,fan-tach-ch = /bits/ 8 <0x02>;
	};

	fan@3 {
		reg = <0x05>;
		aspeed,fan-tach-ch = /bits/ 8 <0x03>;
	};
	
	fan@4 {
		reg = <0x02>;
		aspeed,fan-tach-ch = /bits/ 8 <0x04>;
	};
	
	fan@5 {
		reg = <0x03>;
		aspeed,fan-tach-ch = /bits/ 8 <0x05>;
	};
	
	fan@6 {
		reg = <0x06>;
		aspeed,fan-tach-ch = /bits/ 8 <0x06>;
	};
	
	fan@7 {
		reg = <0x07>;
		aspeed,fan-tach-ch = /bits/ 8 <0x07>;
	};
};
```
My phosphor-hwmon_%.bbappend file:
```
FILESEXTRAPATHS_prepend := "${THISDIR}/${PN}:"

EXTRA_OECONF_append_xx = " --enable-negative-errno-on-fail"

ITEMS = "ahb/apb/i2c@1e78a000/i2c-bus@40/lm75b@4f.conf"
ITEMS += " ahb/apb/pwm-tacho-controller@1e786000.conf"
ITEMS += " iio-hwmon.conf"

ENVS = "obmc/hwmon/{0}"
SYSTEMD_ENVIRONMENT_FILE_${PN} += "${@compose_list(d, 'ENVS', 'ITEMS')}"
```
My pwm-tacho-controller@1e786000.conf file:
```
LABEL_fan1=fan1
PWM_TARGET_fan1=1
LABEL_fan2=fan2
PWM_TARGET_fan2=2
LABEL_fan3=fan3
PWM_TARGET_fan3=3
LABEL_fan4=fan4
PWM_TARGET_fan4=4
LABEL_fan5=fan5
PWM_TARGET_fan5=5
LABEL_fan6=fan6
PWM_TARGET_fan6=6
LABEL_fan7=fan7
PWM_TARGET_fan7=7
LABEL_fan8=fan8
PWM_TARGET_fan8=8
```
My OpenBMC version:
```
root@xxxx:~# cat /etc/os-release 
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC"
VERSION="2.6.0-1"
VERSION_ID="2.6.0-1-g27aa253ce-dirty"
PRETTY_NAME="Phosphor OpenBMC 2.6.0-1"
BUILD_ID="2.6.0"
```
When I start up my bmc, I can find the fan hwmon had been created. I can also read fans' speed and control the fans.
```
root@xxxx:/sys/class/hwmon/hwmon0# ls -l
lrwxrwxrwx    1 root     root             0 Mar  7 09:45 device -> ../../../1e786000.pwm-tacho-controller
-r--r--r--    1 root     root          4096 Mar  7 09:45 fan1_input
-r--r--r--    1 root     root          4096 Mar  7 09:45 fan2_input
-r--r--r--    1 root     root          4096 Mar  7 09:45 fan3_input
-r--r--r--    1 root     root          4096 Mar  7 09:45 fan4_input
-r--r--r--    1 root     root          4096 Mar  7 09:45 fan5_input
-r--r--r--    1 root     root          4096 Mar  7 09:45 fan6_input
-r--r--r--    1 root     root          4096 Mar  7 09:45 fan7_input
-r--r--r--    1 root     root          4096 Mar  7 09:45 fan8_input
-r--r--r--    1 root     root          4096 Mar  7 09:45 name
lrwxrwxrwx    1 root     root             0 Mar  7 09:45 of_node -> ../../../../../../../firmware/devicetree/base/ahb/apb/pwm-tacho-controller@1e786000
drwxr-xr-x    2 root     root             0 Mar  7 09:45 power
-rw-r--r--    1 root     root          4096 Mar  7 09:45 pwm1
-rw-r--r--    1 root     root          4096 Mar  7 09:45 pwm2
-rw-r--r--    1 root     root          4096 Mar  7 09:45 pwm3
-rw-r--r--    1 root     root          4096 Mar  7 09:45 pwm4
-rw-r--r--    1 root     root          4096 Mar  7 09:45 pwm5
-rw-r--r--    1 root     root          4096 Mar  7 09:45 pwm6
-rw-r--r--    1 root     root          4096 Mar  7 09:45 pwm7
-rw-r--r--    1 root     root          4096 Mar  7 09:45 pwm8
lrwxrwxrwx    1 root     root             0 Mar  7 09:45 subsystem -> ../../../../../../../class/hwmon
-rw-r--r--    1 root     root          4096 Mar  7 09:45 uevent
```
But when I login webui, there are several high errors:
```
#1 HIGH ERROR Mar 7, 2019 GMT+8 下午5:02:14
xyz.openbmc_project.Sensor.Device.Error.ReadFailure

CALLOUT_DEVICE_PATH=/sys/devices/platform/ahb/ahb:apb/1e786000.pwm-tacho-controller CALLOUT_ERRNO=110 _PID=1035
```
More log about phosphor-hwmon:
```
root@xxxx:~# journalctl | grep phosphor-hwmon-readd
Mar 07 09:02:28 sw6a phosphor-hwmon-readd[926]: Failed to read from device.
Mar 07 09:02:30 sw6a phosphor-hwmon-readd[926]: Logging failing sysfs file
Mar 07 09:02:38 sw6a phosphor-hwmon-readd[968]: Failed to read from device.
Mar 07 09:02:40 sw6a phosphor-hwmon-readd[968]: Logging failing sysfs file
```
Can someone help me?

geissonator, 2019-11-07:
@spinler or @pstrinkle any thoughts?

mine260309, 2019-11-07:
`-110` is ETIMEDOUT, which means the driver is getting ETIMEDOUT error on read some fans, probably the fan is not spinning at all.

You probably need to add `enable-negative-errno-on-fail` option for hwmon to handle such errorno.
See example in Romulus: https://github.com/openbmc/openbmc/blob/893928eee68168deb91e8cb764ac475ebab2882a/meta-ibm/meta-romulus/recipes-phosphor/sensors/phosphor-hwmon_%25.bbappend#L3

z1908144712, 2019-11-08:
@geissonator @mine260309 Thank you very much. I have solved this problem according to your tips. Thank you again.

mine260309, 2019-11-11:
@z1908144712 Please close the issue when it's resolved.

closed 2019-11-11 by z1908144712