openbmc/openbmc #3631: Serial over LAN (SOL) has 15 character offset to echo the input and take effect

kmkiran99, 2019-11-07:
To access the server host console over LAN using IPMITOOL commands, it was seen that there is a 15 character offset to echo / take effect on the console.

**Unexpected behavior you saw**
Flashed the openBMC image (Warrior) on the BMC. After the system boots, use the following IPMI command to access the Host shell.
$ipmitool -U root -P 0penBmc -I lanplus -H <BMC IP Address> sol activate

Once the SOL shell appears, start typing the command / characters of the command. You will see the first character being echoed after the 16th character is typed in. 

To test, i started typing in characters and "Enter" key combination. The characters appear after the 16th character is typed in and then onwards, the input keys / characters are echoed in the order that was input.

**Expected behavior**
A clear and concise description of what you expected to happen.

**To Reproduce**
Steps to reproduce the behavior:
1. Flash openBMC image on a Intel reference Wolfpass server platform.
2. Boot the BMC and Host.
3. use the above mention ipmitool command to start an SOL session and type in characters / keys of any Host shell command.
4. See the input characters / keys of the command taking effect with an offset of 15 characters

**OpenBMC Information:**
 - /etc/os-release:
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="2.7.0-5"
VERSION_ID="2.7.0-5-ge3da7a4d6"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) 2.7.0-5"
BUILD_ID="2.7.0"
OPENBMC_TARGET_MACHINE="s2600wf"

 - uname -a: 
Linux s2600wf 5.3.5-9b52011 #1 Thu Nov 7 16:46:22 UTC 2019 armv6l GNU/Linux
 - Any additional patches or changes:

**Additional context**
Add any other context about the problem here.

geissonator, 2019-11-11:
@gkeishin can you see if this is something we can recreate on our witherspoon system? @tomjoseph83 any known issues in this area for ipmi?

kmkiran99, 2019-11-22:
Tried to validate with the IBM Romulus platform using the build from IBM openBMC repository. SOL doesn't work with that build on Romulus platform. The session opens, but nothing is seen on console. Host was rebooted multiple times and checked.

kmkiran99, 2019-12-02:
The same behavior is observed with the app "obmc-console" on the BMC as well. Once the app is invoked, the first 15 characters does not echo on the console. No action corresponding to those first 15 key inputs is observed on Host either.

mdmillerii, 2019-12-03:
a 16550 has a 16 character fifo, which is suspicous.  The problemm could be with the host waiting for the interrupt and not checking for partial fifo characters, or something waiting for a full batch of 16 characters to send to the fifio.

closed 2020-07-03 by stale[bot]