openbmc/openbmc #3633: Some problems in IPMI adding fan sensor

z1908144712, 2019-11-22:
My OpenBMC version is 2.7
```
root@xxxx:~# cat /etc/os-release 
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="2.7.0-0"
VERSION_ID="2.7.0-0-gcb91a7773-dirty"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) 2.7.0-0"
BUILD_ID="2.7.0"
OPENBMC_TARGET_MACHINE="xxxx"
```
My YAML file:
```
1:
    bExp: 0
    interfaces:
        xyz.openbmc_project.Sensor.Value:
            Value:
                Offsets:
                    255:
                        type: int64_t
    multiplierM: 1
    offsetB: 0
    mutability: Mutability::Write|Mutability::Read
    path: /xyz/openbmc_project/sensors/temperature/cpu
    rExp: 0
    readingType: readingData
    scale: -3
    sensorNamePattern: nameLeaf
    sensorReadingType: 1
    sensorType: 1
    serviceInterface: org.freedesktop.DBus.Properties
    unit: xyz.openbmc_project.Sensor.Value.Unit.DegreesC
2:
    bExp: 0
    interfaces:
        xyz.openbmc_project.Sensor.Value:
            Value:
                Offsets:
                    255:
                        type: int64_t
    multiplierM: 100
    offsetB: 0
    mutability: Mutability::Write|Mutability::Read
    path: /xyz/openbmc_project/sensors/fan_tach/fan1
    rExp: 0
    readingType: readingData
    scale: 0
    sensorNamePattern: nameLeaf
    sensorReadingType: 1
    sensorType: 4
    serviceInterface: org.freedesktop.DBus.Properties
    unit: xyz.openbmc_project.Sensor.Value.Unit.RPMS
3:
    bExp: 0
    interfaces:
        xyz.openbmc_project.Sensor.Value:
            Value:
                Offsets:
                    255:
                        type: int64_t
    multiplierM: 100
    offsetB: 0
    mutability: Mutability::Write|Mutability::Read
    path: /xyz/openbmc_project/sensors/fan_tach/fan2
    rExp: 0
    readingType: readingData
    scale: 0
    sensorNamePattern: nameLeaf
    sensorReadingType: 1
[... 308 lines cut ...]
Get SDR 0002 command failed
fan2             | 1500.000   | RPM        | ok    | na        | 0.000     | 0.000     | 10100.000 | 10100.000 | na        
Get SDR 0003 command failed
fan3             | 1500.000   | RPM        | ok    | na        | 0.000     | 0.000     | 10100.000 | 10100.000 | na        
Get SDR 0004 command failed
fan4             | 0.000      | RPM        | ok    | na        | 0.000     | 0.000     | 10100.000 | 10100.000 | na        
Get SDR 0005 command failed
fan5             | 0.000      | RPM        | ok    | na        | 0.000     | 0.000     | 10100.000 | 10100.000 | na        
Get SDR 0006 command failed
fan6             | 0.000      | RPM        | ok    | na        | 0.000     | 0.000     | 10100.000 | 10100.000 | na        
Get SDR 0007 command failed
fan7             | 0.000      | RPM        | ok    | na        | 0.000     | 0.000     | 10100.000 | 10100.000 | na        
Get SDR 0008 command failed
fan8             | 3700.000   | RPM        | ok    | na        | 0.000     | 0.000     | 10100.000 | 10100.000 | na        
Get SDR 0009 command failed
CPUVCC09         | 1.065      | Volts      | ok    | na        | 0.810     | 0.810     | 1.125     | 1.125     | na        
VDD25            | 2.552      | Volts      | ok    | na        | 2.288     | 2.288     | 2.750     | 2.750     | na        
VSB33            | 3.420      | Volts      | ok    | na        | 3.030     | 3.030     | 3.645     | 3.645     | na        
8748_VDD09       | 0.960      | Volts      | ok    | na        | 0.810     | 0.810     | 1.125     | 1.125     | na        
ICH_VDD09        | 0.985      | Volts      | ok    | na        | 0.810     | 0.810     | 1.125     | 1.125     | na        
VDD5             | 5.148      | Volts      | ok    | na        | 4.620     | 4.620     | 5.566     | 5.566     | na        
VDD18            | 1.800      | Volts      | ok    | na        | 1.664     | 1.664     | 1.992     | 1.992     | na        
12V_HCP0805_R    | 12.116     | Volts      | ok    | na        | 11.024    | 11.024    | 13.260    | 13.260    | na 
```
Actually fan1-3 and fan8 are normal, others are not used. Can someone give me some advice ?

geissonator, 2019-11-22:
You'll want to be sure something is actually filling those sensors in for you. For us, it's phosphor-hwmon services that read and populate the dbus sensors (which is what you are pointing IPMI to above).

```
# mapper get-service /xyz/openbmc_project/sensors/fan_tach
xyz.openbmc_project.Hwmon-4110222412.Hwmon1

busctl tree xyz.openbmc_project.Hwmon-4110222412.Hwmon1 | cat
...

busctl introspect xyz.openbmc_project.Hwmon-4110222412.Hwmon1 /xyz/openbmc_project/sensors/fan_tach/fan0_0 | grep Value
xyz.openbmc_project.Sensor.Value                      interface -         -                                        -
.MaxValue                                             property  x         0                                        emits-change writable
.MinValue                                             property  x         0                                        emits-change writable
.Unit                                                 property  s         "xyz.openbmc_project.Sensor.Value.Unit.\u2026 emits-change writable
.Value                                                property  x         10162                                    emits-change writable
```

z1908144712, 2019-11-25:
@geissonator Yes, I also use the photospor-hwmon services, and they are running normally.
```
root@xxxx:~# mapper get-service /xyz/openbmc_project/sensors/fan_tach
xyz.openbmc_project.Hwmon-1644477290.Hwmon1
root@xxxx:~# busctl tree xyz.openbmc_project.Hwmon-1644477290.Hwmon1 | cat
└─/xyz
  └─/xyz/openbmc_project
    └─/xyz/openbmc_project/sensors
      └─/xyz/openbmc_project/sensors/fan_tach
        ├─/xyz/openbmc_project/sensors/fan_tach/fan1
        ├─/xyz/openbmc_project/sensors/fan_tach/fan2
        ├─/xyz/openbmc_project/sensors/fan_tach/fan3
        ├─/xyz/openbmc_project/sensors/fan_tach/fan4
        ├─/xyz/openbmc_project/sensors/fan_tach/fan5
        ├─/xyz/openbmc_project/sensors/fan_tach/fan6
        ├─/xyz/openbmc_project/sensors/fan_tach/fan7
        └─/xyz/openbmc_project/sensors/fan_tach/fan8
root@xxxx:~# busctl introspect xyz.openbmc_project.Hwmon-1644477290.Hwmon1 /xyz/openbmc_project/sensors/fan_tach/fan1 | grep Value
xyz.openbmc_project.Sensor.Value                      interface -         -                                        -
.MaxValue                                             property  x         0                                        emits-change writable
.MinValue                                             property  x         0                                        emits-change writable
.Unit                                                 property  s         "xyz.openbmc_project.Sensor.Value.Unit.… emits-change writable
.Value                                                property  x         3102                                     emits-change writable
root@xxxx:~# busctl introspect xyz.openbmc_project.Hwmon-1644477290.Hwmon1 /xyz/openbmc_project/sensors/fan_tach/fan2 | grep Value
xyz.openbmc_project.Sensor.Value                      interface -         -                                        -
.MaxValue                                             property  x         0                                        emits-change writable
.MinValue                                             property  x         0                                        emits-change writable
.Unit                                                 property  s         "xyz.openbmc_project.Sensor.Value.Unit.… emits-change writable
.Value                                                property  x         3032                                     emits-change writable
root@xxxx:~# busctl introspect xyz.openbmc_project.Hwmon-1644477290.Hwmon1 /xyz/openbmc_project/sensors/fan_tach/fan3 | grep Value
xyz.openbmc_project.Sensor.Value                      interface -         -                                        -
.MaxValue                                             property  x         0                                        emits-change writable
.MinValue                                             property  x         0                                        emits-change writable
.Unit                                                 property  s         "xyz.openbmc_project.Sensor.Value.Unit.… emits-change writable
.Value                                                property  x         3063                                     emits-change writable
root@xxxx:~# busctl introspect xyz.openbmc_project.Hwmon-1644477290.Hwmon1 /xyz/openbmc_project/sensors/fan_tach/fan4 | grep Value
xyz.openbmc_project.Sensor.Value                      interface -         -                                        -
.MaxValue                                             property  x         0                                        emits-change writable
.MinValue                                             property  x         0                                        emits-change writable
.Unit                                                 property  s         "xyz.openbmc_project.Sensor.Value.Unit.… emits-change writable
.Value                                                property  x         -110                                     emits-change writable
root@xxxx:~# busctl introspect xyz.openbmc_project.Hwmon-1644477290.Hwmon1 /xyz/openbmc_project/sensors/fan_tach/fan5 | grep Value
xyz.openbmc_project.Sensor.Value                      interface -         -                                        -
.MaxValue                                             property  x         0                                        emits-change writable
.MinValue                                             property  x         0                                        emits-change writable
.Unit                                                 property  s         "xyz.openbmc_project.Sensor.Value.Unit.… emits-change writable
.Value                                                property  x         -110                                     emits-change writable
root@xxxx:~# busctl introspect xyz.openbmc_project.Hwmon-1644477290.Hwmon1 /xyz/openbmc_project/sensors/fan_tach/fan6 | grep Value
xyz.openbmc_project.Sensor.Value                      interface -         -                                        -
.MaxValue                                             property  x         0                                        emits-change writable
.MinValue                                             property  x         0                                        emits-change writable
.Unit                                                 property  s         "xyz.openbmc_project.Sensor.Value.Unit.… emits-change writable
[... cut ...]
.MaxValue                                             property  x         0                                        emits-change writable
.MinValue                                             property  x         0                                        emits-change writable
.Unit                                                 property  s         "xyz.openbmc_project.Sensor.Value.Unit.… emits-change writable
.Value                                                property  x         3151                                     emits-change writable
```

geissonator, 2019-12-02:
ok, well that def shows issue on the hwmon side with those -110 readings. Anything special that indicates @spinler ?

spinler, 2019-12-02:
I think the ones that are getting the -110 are the unused ones?  You may want to remove them from the YAML.

I'm not that familiar with the IPMI code.  Do you see anything in the journal from the IPMI daemon when you make those IPMI calls that are failing?

z1908144712, 2019-12-03:
@spinler Yes, if the value is - 110, there is no fan connected. After I remove them, there are still some problems in the data display, such as the upper and lower thresholds of fan speed cannot be displayed.
```
root@xxxx:~# ipmitool sdr
cpu              | 42 degrees C      | ok
fan1             | 2400 RPM          | ok
fan2             | disabled          | ns
fan3             | 2300 RPM          | ok
fan4             | disabled          | ns
CPUVCC09         | 1 Volts           | ok
VDD25            | 2.46 Volts        | ok
VSB33            | 3.30 Volts        | ok
8748_VDD09       | 0.90 Volts        | ok
ICH_VDD09        | 0.92 Volts        | ok
VDD5             | 4.97 Volts        | ok
VDD18            | 1.74 Volts        | ok
12V_HCP0805_R    | 11.70 Volts       | ok
root@xxxx:~# ipmitool sensor
cpu              | 42.000     | degrees C  | ok    | na        | na        | na        | 70.000    | 70.000    | na        
fan1             | 3700.000   | RPM        | ok    | na        | na        | na        | na        | na        | na        
fan2             | 3600.000   | RPM        | ok    | na        | na        | na        | na        | na        | na        
fan3             | 3600.000   | RPM        | ok    | na        | na        | na        | na        | na        | na        
fan4             | 2400.000   | RPM        | ok    | na        | na        | na        | na        | na        | na        
CPUVCC09         | 1.000      | Volts      | ok    | na        | 0.810     | 0.810     | 1.125     | 1.125     | na        
VDD25            | 2.464      | Volts      | ok    | na        | 2.288     | 2.288     | 2.750     | 2.750     | na        
VSB33            | 3.300      | Volts      | ok    | na        | 3.030     | 3.030     | 3.645     | 3.645     | na        
8748_VDD09       | 0.895      | Volts      | ok    | na        | 0.810     | 0.810     | 1.125     | 1.125     | na        
ICH_VDD09        | 0.920      | Volts      | ok    | na        | 0.810     | 0.810     | 1.125     | 1.125     | na        
VDD5             | 4.972      | Volts      | ok    | na        | 4.620     | 4.620     | 5.566     | 5.566     | na        
VDD18            | 1.736      | Volts      | ok    | na        | 1.664     | 1.664     | 1.992     | 1.992     | na        
12V_HCP0805_R    | 11.700     | Volts      | ok    | na        | 11.024    | 11.024    | 13.260    | 13.260    | na        
```
```
root@xxxx:~# journalctl | grep ipmi
Dec 03 10:37:47  kernel: ipmi-bt-host 1e789140.ibt: Found bt bmc device
Dec 03 10:37:47  kernel: ipmi-bt-host 1e789140.ibt: Using IRQ 30
Dec 03 10:37:47  systemd[1]: /lib/systemd/system/phosphor-ipmi-net@.socket:3: Invalid interface name, ignoring: sys-subsystem-net-devices-%i.device
Dec 03 10:37:47  systemd[1]: Created slice system-phosphor\x2dipmi\x2dnet.slice.
Dec 03 10:38:07  systemd[1]: Listening on phosphor-ipmi-net@eth0.socket.
Dec 03 10:38:59  ipmid[267]: JSON file not found
Dec 03 10:39:00  ipmid[267]: JSON file not found
Dec 03 10:39:00  ipmid[267]: Registering OEM:[0X00C2CF], Cmd:[0X04] for Manual Zone Control
Dec 03 10:39:05  ipmid[267]: Error in reading IPMI user data file
Dec 03 10:39:05  netipmid[276]: Failed to get bus name
Dec 03 10:39:06  ipmid[267]: Loading whitelist filter
Dec 03 10:39:06  ipmid[267]: Set restrictedMode = false
Dec 03 10:39:06  ipmid[267]: Failed interface mapping, no such name
Dec 03 10:39:06  ipmid[267]: New interface mapping
```
There are some error logs, but the cause of the problem is not obvious.

spinler, 2019-12-03:
@z1908144712 Do you have the thresholds set in the hwmon config file such that the Thresholds interface is on D-Bus?

z1908144712, 2019-12-03:
@spinler Yes, I set the upper and lower thresholds, and there is no problem on the web.But now my problem is that, for a while, only the value is displayed, the threshold is not displayed, and for a while, the threshold is displayed, and the value is not displayed.
[image]

qingshuifuqu, 2019-12-20:
@geissonator  
HI：
I want to kwon that where is the /xyz/openbmc_project/sensors/fan_tach/fan1?

ging450, 2020-05-12:
@all  how to add sensors?

kingzmm, 2020-10-30:
@z1908144712 
I want to know how you realized the ipmitool sensor display data can be accurate to three decimal places, thank you very much

closed 2021-07-08 by stale[bot]

zpdsg123, 2022-09-02:
> @spinler Yes, if the value is - 110, there is no fan connected. After I remove them, there are still some problems in the data display, such as the upper and lower thresholds of fan speed cannot be [...]

why the fan is not connected？ I have finished the devicetree configuration and yaml.config