openbmc/openbmc #3634: dbus-sensors: Attempt to use kernel headers from user space

shenki, 2019-11-29:
When updating to the 5.4 kernel the dbus-sensors application fails to build.

```
00:35:01 /tmp/openbmc/work-shared/tiogapass/kernel-source/include/uapi/linux/types.h:10:2: warning: #warning "Attempt to use kernel headers from user space, see http://kernelnewbies.org/KernelHeaders" [-Wcpp]
00:35:01    10 | #warning "Attempt to use kernel headers from user space, see http://kernelnewbies.org/KernelHeaders"
```

```
00:35:01 ERROR: dbus-sensors-0.1+gitAUTOINC+14f0ad806c-r0 do_compile: Execution of '/tmp/openbmc/work/arm1176jzs-openbmc-linux-gnueabi/dbus-sensors/0.1+gitAUTOINC+14f0ad806c-r0/temp/run.do_compile.70683' failed with exit code 1:
00:35:01 [1/44] /tmp/openbmc/work/arm1176jzs-openbmc-linux-gnueabi/dbus-sensors/0.1+gitAUTOINC+14f0ad806c-r0/recipe-sysroot-native/usr/bin/arm-openbmc-linux-gnueabi/arm-openbmc-linux-gnueabi-g++  -DBOOST_ALL_NO_LIB -DBOOST_ASIO_DISABLE_THREADS -DBOOST_ERROR_CODE_HEADER_ONLY -DBOOST_NO_RTTI -DBOOST_NO_TYPEID -DBOOST_SYSTEM_NO_DEPRECATED -I/tmp/openbmc/work/arm1176jzs-openbmc-linux-gnueabi/dbus-sensors/0.1+gitAUTOINC+14f0ad806c-r0/git/include -marm -mcpu=arm1176jz-s  --sysroot=/tmp/openbmc/work/arm1176jzs-openbmc-linux-gnueabi/dbus-sensors/0.1+gitAUTOINC+14f0ad806c-r0/recipe-sysroot  -O2 -pipe -g -feliminate-unused-debug-types -fmacro-prefix-map=/tmp/openbmc/work/arm1176jzs-openbmc-linux-gnueabi/dbus-sensors/0.1+gitAUTOINC+14f0ad806c-r0=/usr/src/debug/dbus-sensors/0.1+gitAUTOINC+14f0ad806c-r0                      -fdebug-prefix-map=/tmp/openbmc/work/arm1176jzs-openbmc-linux-gnueabi/dbus-sensors/0.1+gitAUTOINC+14f0ad806c-r0=/usr/src/debug/dbus-sensors/0.1+gitAUTOINC+14f0ad806c-r0                      -fdebug-prefix-map=/tmp/openbmc/work/arm1176jzs-openbmc-linux-gnueabi/dbus-sensors/0.1+gitAUTOINC+14f0ad806c-r0/recipe-sysroot=                      -fdebug-prefix-map=/tmp/openbmc/work/arm1176jzs-openbmc-linux-gnueabi/dbus-sensors/0.1+gitAUTOINC+14f0ad806c-r0/recipe-sysroot-native=  -fvisibility-inlines-hidden  -marm -mcpu=arm1176jz-s  --sysroot=/tmp/openbmc/work/arm1176jzs-openbmc-linux-gnueabi/dbus-sensors/0.1+gitAUTOINC+14f0ad806c-r0/recipe-sysroot -I /tmp/openbmc/work-shared/tiogapass/kernel-source/include/uapi -I /tmp/openbmc/work-shared/tiogapass/kernel-source/include   -std=gnu++17 -MD -MT CMakeFiles/intrusionsensor.dir/src/ChassisIntrusionSensor.cpp.o -MF CMakeFiles/intrusionsensor.dir/src/ChassisIntrusionSensor.cpp.o.d -o CMakeFiles/intrusionsensor.dir/src/ChassisIntrusionSensor.cpp.o -c /tmp/openbmc/work/arm1176jzs-openbmc-linux-gnueabi/dbus-sensors/0.1+gitAUTOINC+14f0ad806c-r0/git/src/ChassisIntrusionSensor.cpp
00:35:01 In file included from /tmp/openbmc/work-shared/tiogapass/kernel-source/include/uapi/linux/stat.h:5,
00:35:01                  from /tmp/openbmc/work/arm1176jzs-openbmc-linux-gnueabi/dbus-sensors/0.1+gitAUTOINC+14f0ad806c-r0/recipe-sysroot/usr/include/bits/statx.h:30,
00:35:01                  from /tmp/openbmc/work/arm1176jzs-openbmc-linux-gnueabi/dbus-sensors/0.1+gitAUTOINC+14f0ad806c-r0/recipe-sysroot/usr/include/sys/stat.h:446,
00:35:01                  from /tmp/openbmc/work/arm1176jzs-openbmc-linux-gnueabi/dbus-sensors/0.1+gitAUTOINC+14f0ad806c-r0/recipe-sysroot/usr/include/boost/coroutine/posix/protected_stack_allocator.hpp:13,
00:35:01                  from /tmp/openbmc/work/arm1176jzs-openbmc-linux-gnueabi/dbus-sensors/0.1+gitAUTOINC+14f0ad806c-r0/recipe-sysroot/usr/include/boost/coroutine/protected_stack_allocator.hpp:12,
00:35:01                  from /tmp/openbmc/work/arm1176jzs-openbmc-linux-gnueabi/dbus-sensors/0.1+gitAUTOINC+14f0ad806c-r0/recipe-sysroot/usr/include/boost/coroutine/all.hpp:14,
00:35:01                  from /tmp/openbmc/work/arm1176jzs-openbmc-linux-gnueabi/dbus-sensors/0.1+gitAUTOINC+14f0ad806c-r0/recipe-sysroot/usr/include/boost/asio/spawn.hpp:19,
00:35:01                  from /tmp/openbmc/work/arm1176jzs-openbmc-linux-gnueabi/dbus-sensors/0.1+gitAUTOINC+14f0ad806c-r0/recipe-sysroot/usr/include/sdbusplus/asio/object_server.hpp:10,
00:35:01                  from /tmp/openbmc/work/arm1176jzs-openbmc-linux-gnueabi/dbus-sensors/0.1+gitAUTOINC+14f0ad806c-r0/git/include/ChassisIntrusionSensor.hpp:4,
00:35:01                  from /tmp/openbmc/work/arm1176jzs-openbmc-linux-gnueabi/dbus-sensors/0.1+gitAUTOINC+14f0ad806c-r0/git/src/ChassisIntrusionSensor.cpp:17:
00:35:01 /tmp/openbmc/work-shared/tiogapass/kernel-source/include/uapi/linux/types.h:10:2: warning: #warning "Attempt to use kernel headers from user space, see http://kernelnewbies.org/KernelHeaders" [-Wcpp]
00:35:01    10 | #warning "Attempt to use kernel headers from user space, see http://kernelnewbies.org/KernelHeaders"
00:35:01       |  ^~~~~~~
```

shenki, 2019-12-02:
https://gerrit.openbmc-project.xyz/c/openbmc/meta-phosphor/+/27633

The build process needs to install the headers to the filesystem. OpenBMC doesn't do this, it uses a different version of the kernel to build userspace against. This needs to be resolved.

The second issue is the PECI drivers have not been merged upstream. As there has been no progress in merging them since January, they have been dropped from the OpenBMC kernel tree, so this header will not exist even if the first issue is resolved. This needs to be resolved by posting the PECI patches upstream.

feistjj, 2019-12-02:
@yoojae can take a look?

shenki, 2019-12-09:
Jae has said on the mailing list that he would post a new set of PECI patches to the list for inclusion by Thursday 12th.

https://lists.ozlabs.org/pipermail/openbmc/2019-December/019728.html

Once these are merged the second issue will be resolved. The first will remain; this must be fixed by changing openbmc to build it's userspace headers using the openbmc kernel tree.

shenki, 2019-12-12:
Jae has now posted kernel patches. Thanks Jae, I will merge them soon.

I am attempting to workaround the first issue that I identified, with this patch. Please review: https://gerrit.openbmc-project.xyz/c/openbmc/dbus-sensors/+/27920

closed 2019-12-17 by geissonator