openbmc/openbmc #3636: The “/lib/moudles” directory does not exist, I cannot load modules manually?

shrd20190131, 2019-12-09:
> root@jbof:/lib# lsmod
Error: could not get list of modules: No such file or directory

> root@jbof:/lib# cat /etc/os-release
ID="openbmc-phosphor"
NAME="USI BMC (OpenBMC     Project Reference Distro)"
VERSION="2.7.0-0"
VERSION_ID="2.7.0-0-1.0-20191119"
PRETTY_NAME="USI BMC (OpenBMC     Project Reference Distro) 2.7.0-0"
BUILD_ID="2.7.0"

mine260309, 2019-12-09:
The kernel module is not enabled by default.
Suggest to close this issue.

shrd20190131, 2019-12-09:
> The kernel module is not enabled by default.
> Suggest to close this issue.

Can you tell me how to enable the kernel moudle? Thanks!

> MACHINE_EXTRA_RDEPENDS = "kernel-modules"

mine260309, 2019-12-09:
I did not use `MACHINE_EXTRA_RDEPENDS = "kernel-modules"` before and not sure if it works or not.

You could enable `CONFIG_MODULES` in your kernel config.

closed 2019-12-09 by shrd20190131