openbmc/openbmc #3637: How to enable the net-snmp service feature in the project construction?

ddfb2004, 2019-12-09:
Hi，I want to use the snmpd service to receive the snmp package from the client. After the construction,  although the net-snmp package has been  builded, the snmpd service and the related files (e.g. snmpd.conf, snmpd) have not been included in the image file.
How to enable the net-snmp service feature in the project construction? Could you give me a configuration method to deal with this?

**To Reproduce**
Steps to reproduce the behavior:
1. Follow the READM.md instructions
2. export TEMPLATECONF=meta-ibm/meta-palmetto/conf
    . openbmc-env
    bitbake obmc-phosphor-image
3. there is not any relating files about snmpd service, in the directory of rootfs
myproject name\obmc-phosphor-image\1.0-r0\rootfs\

**Screenshots**
1. the net-snmp directory
[image]
2. the builded net-snmp directory
[image]
3. there is not any relating files about snmpd service, in the directory of rootfs
[image]

**OpenBMC Information:**
 - /etc/os-release:
NAME="Ubuntu"
VERSION="18.04.3 LTS (Bionic Beaver)"
ID=ubuntu
ID_LIKE=debian
PRETTY_NAME="Ubuntu 18.04.3 LTS"
VERSION_ID="18.04"
HOME_URL="https://www.ubuntu.com/"
SUPPORT_URL="https://help.ubuntu.com/"
BUG_REPORT_URL="https://bugs.launchpad.net/ubuntu/"
PRIVACY_POLICY_URL="https://www.ubuntu.com/legal/terms-and-policies/privacy-policy"
VERSION_CODENAME=bionic
UBUNTU_CODENAME=bionic

 - uname -a:
Linux zd-tech-xa 4.15.0-70-generic #79-Ubuntu SMP Tue Nov 12 10:36:11 UTC 2019 x86_64 x86_64 x86_64 GNU/Linux

 - Any additional patches or changes:
using the latest version of openbmc

geissonator, 2019-12-09:
@ratagupt any advice? I doubt we packaged snmp into palmetto? Is it in Romulus?

nest1ing, 2019-12-18:
You could add `net-snmp-server-snmpd` package to your firmware image.
E.g. for palmetto it might be done by editing the `meta-ibm/meta-palmetto/reciepes-phosphor/images/obmc-phosphor-image.bbappend`:
```
OBMC_IMAGE_EXTRA_INSTALL_append_palmetto = " mbox net-snmp-server-snmpd"
```

ddfb2004, 2019-12-18:
Thanks for your suggestions, it worked well, I found the snmpd in the rootfs. Now I will close the issue.

closed 2019-12-18 by ddfb2004