openbmc/openbmc #3640: Firmware not booting when failover flash is empty.

akashgj, 2019-12-31:
Added the following configuration dts file for BMC failover flash. 
+	flash@1 {
+		status = "okay";
+		label = "alt";
+		m25p,fast-read;
+		spi-max-frequency = <50000000>;
+	};

Firmware is booting only if primary and failover flash chips are present.

When I removed failover flash chip, booting stuck with following error.

[    2.906962] Run /init as init process
grep: /sys/class/mtd/*/name: No such file or directory
grep: /sys/class/mtd/*/name: No such file or directory
grep: /sys/class/mtd/*/name: No such file or directory
cat: read error: Is a directory
grep: /sys/class/mtd/*/name: No such file or directory
cat: read error: Is a directory
rofs = squashfs rwfs = jffs2
mount: mounting /dev/mtdblock on run/initramfs/ro failed: No such file or directory
mount: mounting /dev/mtdblock on run/initramfs/rw failed: No such file or directory

Mounting read-write /dev/mtdblock filesystem failed.  Please fix and run
	mount /dev/mtdblock run/initramfs/rw -t jffs2 -o rw
to to continue, or do change nothing to run from RAM for this boot.
Enter password to try to manually fix.

mdmillerii, 2019-12-31:
From the error messages, on your failed boot the kernel did not have any flash devices registered.  You did not post the kernel message log which may have clude, and would indicate which chip you have and which driver you are trying to use.

However, I did look at the driver for the Aspeed Technology 2400, 2500, and 2600 flash controller (`/drivers/mtd/spi-nor/aspeed-smc.c`), and it looks like any device not being found will cause it to unregister all other chips.  @legoater @shenki We should consider handling -ENODEV for some chip selects (I'm assuming that is the error with the chip not present).  We would likely need to assign a window to the missing chip to use later chip selects.

akashgj, 2020-01-01:
The following are the kernel messages

Starting kernel ...

[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Linux version 5.2.9-c008a29-dirty-eec0145 (oe-user@oe-host) (gcc version 9.2.0 (GCC)) #1 Thu Dec 19 06:02:33 UTC 2019
[    0.000000] CPU: ARMv6-compatible processor [410fb767] revision 7 (ARMv7), cr=00c5387d
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, VIPT nonaliasing instruction cache
[    0.000000] OF: fdt: Machine model: S2600WF BMC
[    0.000000] Memory policy: Data cache writeback
[    0.000000] cma: Reserved 16 MiB at 0x9d800000
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 125984
[    0.000000] Kernel command line: console=ttyS4,115200n8 root=/dev/ram rw
[    0.000000] Dentry cache hash table entries: 65536 (order: 6, 262144 bytes)
[    0.000000] Inode-cache hash table entries: 32768 (order: 5, 131072 bytes)
[    0.000000] Memory: 475688K/507904K available (6144K kernel code, 387K rwdata, 1416K rodata, 1024K init, 145K bss, 15832K reserved, 16384K cma-reserved)
[    0.000000] random: get_random_u32 called from cache_alloc_refill+0x3d0/0x8fc with crng_init=0
[    0.000000] ftrace: allocating 20743 entries in 41 pages
[    0.000000] NR_IRQS: 16, nr_irqs: 16, preallocated irqs: 16
[    0.000000] i2c controller registered, irq 16
[    0.000000] clocksource: FTTMR010-TIMER2: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 77222644334 ns
[    0.000018] sched_clock: 32 bits at 24MHz, resolution 40ns, wraps every 86767015915ns
[    0.000091] Switching to timer-based delay loop, resolution 40ns
[    0.001258] Calibrating delay loop (skipped), value calculated using timer frequency.. 49.50 BogoMIPS (lpj=247500)
[    0.001311] pid_max: default: 32768 minimum: 301
[    0.002050] Mount-cache hash table entries: 1024 (order: 0, 4096 bytes)
[    0.002091] Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes)
[    0.003669] CPU: Testing write buffer coherency: ok
[    0.005458] Setting up static identity map for 0x80100000 - 0x80100038
[    0.009497] devtmpfs: initialized
[    0.023203] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 19112604462750000 ns
[    0.023260] futex hash table entries: 256 (order: -1, 3072 bytes)
[    0.026572] pinctrl core: initialized pinctrl subsystem
[    0.027629] NET: Registered protocol family 16
[    0.029555] DMA: preallocated 256 KiB pool for atomic coherent allocations
[    0.031738] hw-breakpoint: found 6 breakpoint and 1 watchpoint registers.
[    0.031769] hw-breakpoint: maximum watchpoint size is 4 bytes.
[    0.069528] videodev: Linux video capture interface: v2.00
[    0.072735] clocksource: Switched to clocksource FTTMR010-TIMER2
[    0.124062] NET: Registered protocol family 2
[    0.125294] tcp_listen_portaddr_hash hash table entries: 512 (order: 0, 4096 bytes)
[    0.125382] TCP established hash table entries: 4096 (order: 2, 16384 bytes)
[    0.125465] TCP bind hash table entries: 4096 (order: 2, 16384 bytes)
[    0.125558] TCP: Hash tables configured (established 4096 bind 4096)
[    0.125784] UDP hash table entries: 256 (order: 0, 4096 bytes)
[    0.125846] UDP-Lite hash table entries: 256 (order: 0, 4096 bytes)
[    0.126381] NET: Registered protocol family 1
[    0.127249] Unpacking initramfs...
[    1.816768] Freeing initrd memory: 1496K
[    1.819484] workingset: timestamp_bits=30 max_order=17 bucket_order=0
[    1.820009] squashfs: version 4.0 (2009/01/31) Phillip Lougher
[    1.820045] jffs2: version 2.2. (SUMMARY)  © 2001-2006 Red Hat, Inc.
[    1.820784] NET: Registered protocol family 38
[    1.823784] aspeed-kcs-bmc: channel=3 addr=0xca2 idr=0x2c odr=0x38 str=0x44
[    1.824801] aspeed-kcs-bmc: channel=4 addr=0xca4 idr=0x94 odr=0x98 str=0x9c
[    1.829636] Serial: 8250/16550 driver, 6 ports, IRQ sharing enabled
[    1.833774] 1e787000.serial: ttyS5 at MMIO 0x1e787000 (irq = 31, base_baud = 1546875) is a 16550A
[    1.835792] 1e783000.serial: ttyS0 at MMIO 0x1e783000 (irq = 29, base_baud = 1500000) is a 16550A
[    1.837135] 1e784000.serial: ttyS4 at MMIO 0x1e784000 (irq = 30, base_baud = 1500000) is a 16550A
[    2.198163] printk: console [ttyS4] enabled
[    2.204042] 1e78e000.serial: ttyS2 at MMIO 0x1e78e000 (irq = 33, base_baud = 1500000) is a 16550A
[    2.214839] timeriomem_rng 1e6e2078.hwrng: 32bits from 0x(ptrval) @ 1us
[    2.222954] random: fast init done
[    2.234507] random: crng init done
[    2.246032] loop: module loaded
[    2.298054] aspeed-smc 1e620000.spi: Using 50 MHz SPI frequency
[    2.304243] aspeed-smc 1e620000.spi: w25m512jv (32768 Kbytes)
[    2.310025] aspeed-smc 1e620000.spi: CE0 window [ 0x20000000 - 0x22000000 ] 32MB
[    2.317534] aspeed-smc 1e620000.spi: CE1 window [ 0x22000000 - 0x2a000000 ] 128MB
[    2.325111] aspeed-smc 1e620000.spi: read control register: 203b0641
[... 63 lines cut ...]
to to continue, or do change nothing to run from RAM for this boot.
Enter password to try to manually fix.
After fixing run exit to continue this script, or reboot -f to retry, or
touch /takeover and exit to become PID 1 allowing editing of this script.
Give root password for system maintenance
(or type Control-D for normal startup):
sulogin: normal startup
[   16.981988] overlayfs: upper fs does not support xattr, falling back to index=off and metacopy=off.
chroot: can't execute '/bin/sh': No such file or directory

Unable to confirm /sbin/init is an executable non-empty file
in merged file system mounted at /root.

[... cut ...] g emergency shell.
Enter password to try to manually fix.
After fixing run exit to continue this script, or reboot -f to retry, or
touch /takeover and exit to become PID 1 allowing editing of this script.
Give root password for system maintenance
(or type Control-D for normal startup):
sulogin: Login incorrect
Give root password for system maintenance
(or type Control-D for normal startup):
sulogin: Login incorrect
Give root password for system maintenance
(or type Control-D for normal startup):

closed 2020-08-01 by stale[bot]

chnguyen-ampere, 2020-11-25:
Hi @akashgj , anything else update ! I'm facing the issue like that. Thank you !

mdmillerii, 2020-11-25:
It's a known limitation of the aspeed driver that it requires a jedec response to the device probe if the flash is listed in the device tree.