openbmc/openbmc #3641: Adding local kernel source to compile

prakeerthi, 2020-01-23:
HI All, I want to compile a Local kernel source .  For this I found that I need to add the below Lines in local.conf file to avoid the  external network fetching git access.

SOURCE_MIRROR_URL ?= "file://source_mirror/sources/"
INHERIT += "own-mirrors"
BB_GENERATE_MIRROR_TARBALLS = "1"
BB_NO_NETWORK = "1"

When I add the  the above , My yocto build is not fetching the source from external git, but it is failing to fetch from internal source_mirro/kernel PATH..  I have copied the kernel source already . I know that we need to make some changes in .bb files to fetch the source from Local source. Kindly help me , if Any of you have idea on this.

geissonator, 2020-01-23:
When I did this in the past I just updated KSRC in https://github.com/openbmc/meta-aspeed/blob/master/recipes-kernel/linux/linux-aspeed.inc#L9
```
KSRC ?= "git:///home/andrewg/obmc_dev/linux;protocol=file;branch=${KBRANCH}"
```
You may need to update KBRANCH as well in https://github.com/openbmc/meta-aspeed/blob/master/recipes-kernel/linux/linux-aspeed_git.bb#L1 to use your branch from that git repo.

You could also use "devtool modify linux-aspeed" as well I believe to get a copy you could update in your yocto tree.

prakeerthi, 2020-01-24:
Hello geissonator,
I tried  the above mentioned steps   and I can able to fetch my local kernel with the steps mentioned.
But still I am facing  the issue with boot loader. 
Here I am providing the  u-boot-common-aspeed_2016.07.inc file changes along with my directory structure under downloads directory.KIndly help to resolve this.

### **u-boot-common-aspeed_2016.07.inc**

HOMEPAGE = "https://github.com/openbmc/u-boot"
SECTION = "bootloaders"
DEPENDS += "flex-native bison-native"

LICENSE = "GPLv2+"
LIC_FILES_CHKSUM = "file://Licenses/README;md5=a2c678cfd4a4d97135585cad908541c6"
PE = "1"

# We use the revision in order to avoid having to fetch it from the
# repo during parse
SRCREV = "fc8646ea8a4f24ff8267fa2eee0ee33327f1f36c"
UBRANCH = "v2016.07-aspeed-openbmc"

//SRC_URI = "git://git@github.com/openbmc/u-boot.git;branch=${UBRANCH};protocol=https"
USRC ?= "git:///home/prakeerthiy/customer_repo/wsyu_nwsa/openbmc-2.6.0/build/downloads/source_mirror/git;protocol=file;branch=${UBRANCH}"
SRC_URI = "${USRC}"

Please find the below directory structure for boot loader in downloads.
prakeerthiy@yocot-VirtualBox:~/customer_repo/wsyu_nwsa/openbmc-2.6.0/build/downloads/source_mirror$ ls
kernel-source  sources  u-boot-aspeed

prakeerthiy@yocot-VirtualBox:~/customer_repo/wsyu_nwsa/openbmc-2.6.0/build/downloads/source_mirror$ cd u-boot-aspeed/

prakeerthiy@yocot-VirtualBox:~/customer_repo/wsyu_nwsa/openbmc-2.6.0/build/downloads
/source_mirror/u-boot-aspeed$ ls
1_v2016.07+gitAUTOINC+fc8646ea8a-r0

prakeerthiy@yocot-VirtualBox:~/customer_repo/wsyu_nwsa/openbmc-2.6.0/build/downloads

/source_mirror/u-boot-aspeed$ cd 1_v2016.07+gitAUTOINC+fc8646ea8a-r0/

prakeerthiy@yocot-VirtualBox:~/customer_repo/wsyu_nwsa/openbmc-2.6.0/build/downloads
/source_mirror/u-boot-aspeed/1_v2016.07+gitAUTOINC+fc8646ea8a-r0$ ls
0001-configs-ast-Add-redundnant-env.patch  deploy-rpms           git              package         pseudo                 sysroot-destdir
build                                      deploy-u-boot-aspeed  image            packages-split  recipe-sysroot         temp
configure.sstate                           disp_uboot.patch      license-destdir  pkgdata         recipe-sysroot-native  u-boot-aspeed.spec

prakeerthiy@yocot-VirtualBox:~/customer_repo/wsyu_nwsa/openbmc-2.6.0/build/downloads/source_mirror/u-boot-aspeed/1_v2016.07+gitAUTOINC+fc8646ea8a-r0$ cd git/

prakeerthiy@yocot-VirtualBox:~/customer_repo/wsyu_nwsa/openbmc-2.6.0/build/downloads/source_mirror/u-boot-aspeed/1_v2016.07+gitAUTOINC+fc8646ea8a-r0/git$ ls

api    cmd        configs  drivers   fs       Kconfig   MAINTAINERS  net      README           test
arch   common     disk     dts       include  lib       MAKEALL      patches  scripts          tools
board  config.mk  doc      examples  Kbuild   Licenses  Makefile     post     snapshot.commit

prakeerthiy@yocot-VirtualBox:~/customer_repo/wsyu_nwsa/openbmc-2.6.0/build/downloads/source_mirror/u-boot-aspeed/1_v2016.07+gitAUTOINC+fc8646ea8a-r0/git$

prakeerthi, 2020-01-24:
The error Log during fetching phase :
WARNING: u-boot-aspeed-1_v2016.07+gitAUTOINC+fc8646ea8a-r0 do_fetch: Failed to fetch URL git:///home/prakeerthiy/customer_repo/wsyu_nwsa/openbmc-2.6.0/build/downloads/source_mirror/u-boot-aspeed;protocol=file;branch=v2016.07-aspeed-openbmc, attempting MIRRORS if available
ERROR: u-boot-aspeed-1_v2016.07+gitAUTOINC+fc8646ea8a-r0 do_fetch: Fetcher failure: Fetch command export PSEUDO_DISABLED=1; export PATH="/home/prakeerthiy/customer_repo/wsyu_nwsa/openbmc-2.6.0/scripts:/home/prakeerthiy/customer_repo/wsyu_nwsa/openbmc-2.6.0/build/tmp/work/winsystems_c446-openbmc-linux-gnueabi/u-boot-aspeed/1_v2016.07+gitAUTOINC+fc8646ea8a-r0/recipe-sysroot-native/usr/bin/arm-openbmc-linux-gnueabi:/home/prakeerthiy/customer_repo/wsyu_nwsa/openbmc-2.6.0/build/tmp/work/winsystems_c446-openbmc-linux-gnueabi/u-boot-aspeed/1_v2016.07+gitAUTOINC+fc8646ea8a-r0/recipe-sysroot/usr/bin/crossscripts:/home/prakeerthiy/customer_repo/wsyu_nwsa/openbmc-2.6.0/build/tmp/work/winsystems_c446-openbmc-linux-gnueabi/u-boot-aspeed/1_v2016.07+gitAUTOINC+fc8646ea8a-r0/recipe-sysroot-native/usr/sbin:/home/prakeerthiy/customer_repo/wsyu_nwsa/openbmc-2.6.0/build/tmp/work/winsystems_c446-openbmc-linux-gnueabi/u-boot-aspeed/1_v2016.07+gitAUTOINC+fc8646ea8a-r0/recipe-sysroot-native/usr/bin:/home/prakeerthiy/customer_repo/wsyu_nwsa/openbmc-2.6.0/build/tmp/work/winsystems_c446-openbmc-linux-gnueabi/u-boot-aspeed/1_v2016.07+gitAUTOINC+fc8646ea8a-r0/recipe-sysroot-native/sbin:/home/prakeerthiy/customer_repo/wsyu_nwsa/openbmc-2.6.0/build/tmp/work/winsystems_c446-openbmc-linux-gnueabi/u-boot-aspeed/1_v2016.07+gitAUTOINC+fc8646ea8a-r0/recipe-sysroot-native/bin:/home/prakeerthiy/customer_repo/wsyu_nwsa/openbmc-2.6.0/poky/bitbake/bin:/home/prakeerthiy/customer_repo/wsyu_nwsa/openbmc-2.6.0/build/tmp/hosttools"; export HOME="/home/prakeerthiy"; LANG=C git -c core.fsyncobjectfiles=0 clone --bare --mirror /home/prakeerthiy/customer_repo/wsyu_nwsa/openbmc-2.6.0/build/downloads/source_mirror/u-boot-aspeed /home/prakeerthiy/customer_repo/wsyu_nwsa/openbmc-2.6.0/build/downloads/git2/home.prakeerthiy.customer_repo.wsyu_nwsa.openbmc-2.6.0.build.downloads.source_mirror.u-boot-aspeed --progress failed with exit code 128, output:
fatal: repository '/home/prakeerthiy/customer_repo/wsyu_nwsa/openbmc-2.6.0/build/downloads/source_mirror/u-boot-aspeed' does not exist

ERROR: u-boot-aspeed-1_v2016.07+gitAUTOINC+fc8646ea8a-r0 do_fetch: Fetcher failure for URL: 'git:///home/prakeerthiy/customer_repo/wsyu_nwsa/openbmc-2.6.0/build/downloads/source_mirror/u-boot-aspeed;protocol=file;branch=v2016.07-aspeed-openbmc'. Unable to fetch URL from any source.
ERROR: u-boot-aspeed-1_v2016.07+gitAUTOINC+fc8646ea8a-r0 do_fetch: Function failed: base_do_fetch
ERROR: Logfile of failure stored in: /home/prakeerthiy/customer_repo/wsyu_nwsa/openbmc-2.6.0/build/tmp/work/winsystems_c446-openbmc-linux-gnueabi/u-boot-aspeed/1_v2016.07+gitAUTOINC+fc8646ea8a-r0/temp/log.do_fetch.22370
ERROR: Task (/home/prakeerthiy/customer_repo/wsyu_nwsa/openbmc-2.6.0/meta-winsystems/meta-c446/winsystems-c446/recipes-bsp/u-boot/u-boot-aspeed_2016.07.bb:do_fetch) failed with exit code '1'

geissonator, 2020-01-24:
That error seems pretty clear, you sure you've got the right path?
```
fatal: repository '/home/prakeerthiy/customer_repo/wsyu_nwsa/openbmc-2.6.0/build/downloads/source_mirror/u-boot-aspeed' does not exist```

prakeerthi, 2020-01-27:
HI  geissonator , 
As I mentioned in my previous post , I have the boot loader source code in the "/home/prakeerthiy/customer_repo/wsyu_nwsa/openbmc-2.6.0/build/downloads/source_mirror/u-boot-aspeed " path.  Kindly check the logs closely once and let me know , what changes we need to do , or else If you have any sample code for u-boot , Plz share . I will also try.

Thank you.

prakeerthi, 2020-01-27:
HI Geissonator ,
To give further clarity on the machine and recipe we used,
We have cloned our machine from meta-evb , kernel and bsp recipes from meta-aspeed.
As you righly pointed out in kernel earlier, We tried the similar in boot loader , and the struck with this issue. It is very clear the source code available in above mentioned PATH.
Please check at your end for the same and let us know the changes ..
Thank you.

mdmillerii, 2020-01-27:
can you do this on your machine:

prakeerthi, 2020-01-28:
HI mdmillerii,
As you suspected , the git repo not getting created , when I ran the above command .
I also tried to fallow the below steps.
1. Created a local repo in file System and try to fetch ... I even tried by initializing the git in the local repo and the download/source-mirrors/u-boot-aspeed PATH.
But  still the same .
I even tried few trail and error , and none of us worked for me. 
Can any one help me on this.

closed 2020-08-29 by stale[bot]