openbmc/openbmc #3642: IP-KVM not working

mine260309, 2020-02-12:
The IP-KVM is not working on master, that it shows a blank page on OpenBMC WebUI.

The known good version is 90a837f71a2a8e4ad3ab8d626605d9204705a5ca
But I do not know which revision starts to fail, and not sure which component breaks it.

geissonator, 2020-02-12:
@eddiejames can you take a look?

eddiejames, 2020-02-12:
obmc-ikvm can't seem to connect to the systemd socket. @gtmills may want to look

gtmills, 2020-02-20:
@mine260309 I think https://gerrit.openbmc-project.xyz/c/openbmc/bmcweb/+/29541 should have fixed this

mdmillerii, 2020-02-20:
That fix changes from IPv6 loopback to IPv4 loopback address.

XiaochaoMa, 2020-02-21:
Test:
Take the above patch. Use the Chrome browser to link the BMC and successfully start the KVM function. The boot process shows normal.

closed 2020-02-24 by mine260309