openbmc/openbmc #3643: SSH access requires priv-admin instead of ssh

joseph-reynolds, 2020-02-19:
Per the [OpenBMC User Management docs](https://github.com/openbmc/docs/blob/master/architecture/user_management.md), access to the BMC via SSH should be controlled by membership in the 'ssh' group.  However, the [implementation](https://github.com/openbmc/openbmc/blob/master/meta-phosphor/recipes-core/dropbear/dropbear/dropbear.default) requires membership in the 'priv-admin' group.

Note that Admin users are added to the "priv-admin" group, and I am not aware of any supported REST API to change this association.  That means membership in the priv-admin group implies access to ssh, so the behavior seems correct.  However, the design clearly separates the group roles, and someone who uses an unsupported technique (such as the shell command `usermod -G ssh ${USER}`) to change the groups should expect SSH access to work as expected.

rthomaiy, 2020-11-17:
@joseph-reynolds  for SSH to work fine, user must be part of priv-admin and must have _command/shell_ as /bin/sh under /etc/passwd file instead of /bin/nologin. Note: There is no direct group called ssh under /etc/group, instead it is just emulated one from phosphor-user-manager to add corresponding shell binary to the user. 
usermod --shell /bin/sh -G priv-admin ${USER} 

If requirement is SSH to be allowed based on group and allowed for all user privileges, then user shell can be updated using usermod --shell /bin/sh  itself, but need to remove EXTRA_ARGS from the [dropbear.default ](https://github.com/openbmc/openbmc/blob/master/meta-phosphor/recipes-core/dropbear/dropbear/dropbear.default)

joseph-reynolds, 2020-11-18:
There was recent discussion on the email list.  In summary, the proposed changes are:

Create new D-Bus property 
```
    - name: DefaultGroups
      type: dict[string, array[string]]
      flags:
          - const
      description: >
          Maps from each privilege in AllPrivileges to the default groups for
          that privilege role (a subset of AllGroups) in a form suitable for
          CreateUser.GroupNames.

```

with implementation in phosphor-user-manager / user_mgr
```
    std::map<std::string,std::vector<std::string>> privToGroupsMgr = 
    {
        {"priv-admin", {"web", "redfish", "ipmi", "ssh"}},
        {"priv-operator", {"web", "redfish", "ipmi"}},
        {"priv-user", {"web", "redfish", "ipmi"}},
        {"priv-noaccess", {}}
    };
```

and then change BMCWeb to use this new property when creating new local accounts (instead of `AllGroups`).

edtanous, 2020-11-18:
> There was recent discussion on the email list. In summary, the proposed changes are:
>
> Create new D-Bus property
>
> ```
>     - name: DefaultGroups
>       type: dict[string, array[string]] [...]

Lets start with a design doc.  I have concerns about how the above interface looks in both bmcweb, and externally on Redfish.  We've essentially overloaded groups to include both channels, and privilege level, and we need to make sure that the external user facing interfaces on IPMI and Redfish make sense when viewed without understanding the internals of OpenBMC.

rthomaiy, 2020-11-19:
(Excerpt from mail discussion).
Privileges & groups are designed to exist separately. i.e. All users created in OpenBMC belong to {"web", "redfish", "ipmi", "ssh"} groups, irrespective of their privilege levels

privileges / roles --> Used to provide authorization level for the OpenBMC application.

groups --> Used to indicate whether user is allowed to be authenticated with the specified application.

Both IPMI & Redfish today allows to configure the privileges/roles for a user, but doesn't allow to configure the groups (hence we had a build time configured group roles - fixed). With the new AccountTypes property discussed, groups can be configured through Redfish and for IPMI we can rely on OpenBMC OEM IPMI commands.

i.e. User can belong to group {"web", "redfish", "ipmi", "ssh"} and with "priv-user". In this case, web/redfish /ipmi application must allow only user privilege related commands to be executed (As authorization based control is not available for SSH, for security reason, we limited ssh support for priv-admin user only). i.e. Through Redfish / ipmi, if the user privilege is increased to "priv-admin" (Administrative) privilege, then user can login to ssh too, and if it is changed to "priv-noaccess", then it means user is disabled for authorization, and not allowed to do any access.

closed 2021-07-08 by stale[bot]