openbmc/openbmc #3646: Who is feeding wdt1 & wdt2

MichaelWang2017, 2020-03-03:
Hi all,
We have the following requirements:
wdt1 is used for system deadlock, and wdt2 is disabled in user space.

However, I found that something is feeding wdt1 and wdt2 (0x1e785000 and 0x1e785020 are constantly changing).
I have not found which upper-level code calls WDIOC_KEEPALIVE ioctl (), will the feeding action come from kernel? 
Please provide some ideas or suggestions, thank you.

mdmillerii, 2020-03-04:
The kernel will attempt to keep all watchdog devices from resetting the system.  If a device has no way out set, then the kernel will create a timer to keep it pinged every half interval if the device is not open.

By default systemd will open /dev/watchdog although they added code to allow the chosen device to be configurable.

Also note that u-boot will ping all known watchdog timers it knows about in its command prompt loop.

MichaelWang2017, 2020-03-05:
Thanks @mdmillerii
I also noticed that kernel has a configuration item called WATCHDOG_HANDLE_BOOT_ENABLED, it can take care of watchdog before userspace using the /dev/watchdogX.

MichaelWang2017, 2020-03-05:
I just watched watchdog_dev.c in kernel 4.19. Is the following understanding correct?
operation 1:
int flags = WDIOS_DISABLECARD;
ioctl (fd, WDIOC_SETOPTIONS, & flags);
close (fd);
=> watchdog disabled. Both kernel and userspace don't care about it anymore.
operation 2:
write (fd, "V", 1);
close (fd);
=> watchdog is still alive, but userspace is no longer concerned, and kernel will be responsible for feeding it in the future.

MichaelWang2017, 2020-03-08:
Both methods can disable watchdog.

closed 2020-03-08 by MichaelWang2017