openbmc/openbmc #3647: OpenBMC build errors with Ubuntu 18.04 and Virtualbox

master882020, 2020-03-09:
Loading cache: 100% |############################################| Time: 0:00:03
Loaded 3896 entries from dependency cache.
Parsing recipes: 100% |##########################################| Time: 0:00:02
Parsing of 2586 .bb files complete (2585 cached, 1 parsed). 3897 targets, 355 skipped, 0 masked, 0 errors.
NOTE: Resolving any missing task queue dependencies

Build Configuration:
BB_VERSION           = "1.44.0"
BUILD_SYS            = "x86_64-linux"
NATIVELSBSTRING      = "ubuntu-18.04"
TARGET_SYS           = "arm-openbmc-linux-gnueabi"
MACHINE              = "romulus"
DISTRO               = "openbmc-phosphor"
DISTRO_VERSION       = "0.1.0"
TUNE_FEATURES        = "arm thumb arm1176jzs"
TARGET_FPU           = "soft"
meta                 
meta-oe              
meta-networking      
meta-perl            
meta-python          
meta-phosphor        
meta-aspeed          
meta-openpower       
meta-romulus         = "master:5457b9264db224ed39215ad425e270d44db37ed2"

Initialising tasks: 100% |#######################################| Time: 0:00:05
Sstate summary: Wanted 914 Found 0 Missed 914 Current 765 (0% match, 45% complete)
NOTE: Executing Tasks
NOTE: Setscene tasks completed
ERROR: phosphor-logging-1.0+gitAUTOINC+51e927cc63-r1 do_compile: oe_runmake failed
ERROR: phosphor-logging-1.0+gitAUTOINC+51e927cc63-r1 do_compile: Execution of '/home/hj/Desktop/openbmc/build/tmp/work/arm1176jzs-openbmc-linux-gnueabi/phosphor-logging/1.0+gitAUTOINC+51e927cc63-r1/temp/run.do_compile.5110' failed with exit code 1:
make: *** No rule to make target '/home/hj/Desktop/openbmc-master/build/tmp/work/arm1176jzs-openbmc-linux-gnueabi/phosphor-logging/1.0+gitAUTOINC+51e927cc63-r1/build/../git/tools/elog-gen.py', needed by 'phosphor-logging/elog-errors.hpp'.  Stop.
WARNING: exit code 1 from a shell command.

ERROR: Logfile of failure stored in: /home/hj/Desktop/openbmc/build/tmp/work/arm1176jzs-openbmc-linux-gnueabi/phosphor-logging/1.0+gitAUTOINC+51e927cc63-r1/temp/log.do_compile.5110
Log data follows:
| DEBUG: Executing python function autotools_aclocals
| DEBUG: SITE files ['endian-little', 'bit-32', 'arm-common', 'arm-32', 'common-linux', 'common-glibc', 'arm-linux', 'arm-linux-gnueabi', 'common']
| DEBUG: Python function autotools_aclocals finished
| DEBUG: Executing shell function do_compile
| NOTE: make -j 4
| make: *** No rule to make target '/home/hj/Desktop/openbmc-master/build/tmp/work/arm1176jzs-openbmc-linux-gnueabi/phosphor-logging/1.0+gitAUTOINC+51e927cc63-r1/build/../git/tools/elog-gen.py', needed by 'phosphor-logging/elog-errors.hpp'.  Stop.
| ERROR: oe_runmake failed
| WARNING: exit code 1 from a shell command.
| ERROR: Execution of '/home/hj/Desktop/openbmc/build/tmp/work/arm1176jzs-openbmc-linux-gnueabi/phosphor-logging/1.0+gitAUTOINC+51e927cc63-r1/temp/run.do_compile.5110' failed with exit code 1:
| make: *** No rule to make target '/home/hj/Desktop/openbmc-master/build/tmp/work/arm1176jzs-openbmc-linux-gnueabi/phosphor-logging/1.0+gitAUTOINC+51e927cc63-r1/build/../git/tools/elog-gen.py', needed by 'phosphor-logging/elog-errors.hpp'.  Stop.
| WARNING: exit code 1 from a shell command.
| 
ERROR: Task (/home/hj/Desktop/openbmc/meta-phosphor/recipes-phosphor/logging/phosphor-logging_git.bb:do_compile) failed with exit code '1'

mdmillerii, 2020-03-09:
> `No rule to make target '/home/hj/Desktop/openbmc-master/build/tmp/work/arm1176jzs-openbmc-linux-gnueabi/phosphor-logging/1.0+gitAUTOINC+51e927cc63-r1/build/../git/tools/elog-gen.py', needed by 'phosphor-logging/elog-errors.hpp'. `

Looks like a missing dependancy @spinler @dkodihal

master882020, 2020-03-10:
> > `No rule to make target '/home/hj/Desktop/openbmc-master/build/tmp/work/arm1176jzs-openbmc-linux-gnueabi/phosphor-logging/1.0+gitAUTOINC+51e927cc63-r1/build/../git/tools/elog-gen.py', needed by 'phosphor-logging/elog-errors.hpp'. `
> 
> Looks like a missing dependancy @spinler @dkodihal

Thank you！

 Because I changed the folder name, modified it to solve.

closed 2020-03-10 by master882020