openbmc/openbmc #3648: OpenBMC build errors with Ubuntu 18.04 and Virtualbox：do_fetch:Failed

master882020, 2020-03-09:
NOTE: Resolving any missing task queue dependencies

Build Configuration:
BB_VERSION           = "1.44.0"
BUILD_SYS            = "x86_64-linux"
NATIVELSBSTRING      = "ubuntu-18.04"
TARGET_SYS           = "arm-openbmc-linux-gnueabi"
MACHINE              = "romulus"
DISTRO               = "openbmc-phosphor"
DISTRO_VERSION       = "0.1.0"
TUNE_FEATURES        = "arm thumb arm1176jzs"
TARGET_FPU           = "soft"
meta                 
meta-oe              
meta-networking      
meta-perl            
meta-python          
meta-phosphor        
meta-aspeed          
meta-openpower       
meta-romulus         = "<unknown>:<unknown>"

Sstate summary: Wanted 1025 Found 0 Missed 1025 Current 654 (0% match, 38% complete)
NOTE: Executing Tasks
NOTE: Setscene tasks completed
NOTE: Running task 655 of 4780 (/home/hj/Desktop/openbmc-master/meta-aspeed/recipes-kernel/linux/linux-aspeed_git.bb:do_fetch)
NOTE: recipe linux-aspeed-5.4.23+gitAUTOINC+86f83a82c7-r0: task do_fetch: Started
WARNING: linux-aspeed-5.4.23+gitAUTOINC+86f83a82c7-r0 do_fetch: Failed to fetch URL git://github.com/openbmc/linux;protocol=git;branch=dev-5.4, attempting MIRRORS if available
ERROR: linux-aspeed-5.4.23+gitAUTOINC+86f83a82c7-r0 do_fetch: Fetcher failure: Fetch command export PSEUDO_DISABLED=1; unset _PYTHON_SYSCONFIGDATA_NAME; export DBUS_SESSION_BUS_ADDRESS="unix:path=/run/user/1000/bus,guid=29dd3389a8958c5a6cb44c315e632528"; export SSH_AGENT_PID="2870"; export SSH_AUTH_SOCK="/run/user/1000/keyring/ssh"; export PATH="/home/hj/Desktop/openbmc-master/scripts:/home/hj/Desktop/openbmc-master/build/tmp/work/romulus-openbmc-linux-gnueabi/linux-aspeed/5.4.23+gitAUTOINC+86f83a82c7-r0/recipe-sysroot-native/usr/bin/arm-openbmc-linux-gnueabi:/home/hj/Desktop/openbmc-master/build/tmp/work/romulus-openbmc-linux-gnueabi/linux-aspeed/5.4.23+gitAUTOINC+86f83a82c7-r0/recipe-sysroot/usr/bin/crossscripts:/home/hj/Desktop/openbmc-master/build/tmp/work/romulus-openbmc-linux-gnueabi/linux-aspeed/5.4.23+gitAUTOINC+86f83a82c7-r0/recipe-sysroot-native/usr/sbin:/home/hj/Desktop/openbmc-master/build/tmp/work/romulus-openbmc-linux-gnueabi/linux-aspeed/5.4.23+gitAUTOINC+86f83a82c7-r0/recipe-sysroot-native/usr/bin:/home/hj/Desktop/openbmc-master/build/tmp/work/romulus-openbmc-linux-gnueabi/linux-aspeed/5.4.23+gitAUTOINC+86f83a82c7-r0/recipe-sysroot-native/sbin:/home/hj/Desktop/openbmc-master/build/tmp/work/romulus-openbmc-linux-gnueabi/linux-aspeed/5.4.23+gitAUTOINC+86f83a82c7-r0/recipe-sysroot-native/bin:/home/hj/Desktop/openbmc-master/poky/bitbake/bin:/home/hj/Desktop/openbmc-master/build/tmp/hosttools"; export HOME="/home/hj"; LANG=C git -c core.fsyncobjectfiles=0 clone --bare --mirror git://github.com/openbmc/linux /home/hj/Desktop/downloads/git2/github.com.openbmc.linux --progress failed with exit code 128, output:
Cloning into bare repository '/home/hj/Desktop/downloads/git2/github.com.openbmc.linux'...
fatal: Unable to look up github.com (port 9418) (Name or service not known)

ERROR: linux-aspeed-5.4.23+gitAUTOINC+86f83a82c7-r0 do_fetch: Fetcher failure for URL: 'git://github.com/openbmc/linux;protocol=git;branch=dev-5.4'. Unable to fetch URL from any source.
ERROR: Logfile of failure stored in: /home/hj/Desktop/openbmc-master/build/tmp/work/romulus-openbmc-linux-gnueabi/linux-aspeed/5.4.23+gitAUTOINC+86f83a82c7-r0/temp/log.do_fetch.79953
NOTE: recipe linux-aspeed-5.4.23+gitAUTOINC+86f83a82c7-r0: task do_fetch: Failed
ERROR: Task (/home/hj/Desktop/openbmc-master/meta-aspeed/recipes-kernel/linux/linux-aspeed_git.bb:do_fetch) failed with exit code '1'
NOTE: Tasks Summary: Attempted 1095 tasks of which 1094 didn't need to be rerun and 1 failed.

Every run is an error!!!!!

mine260309, 2020-03-09:
This is caused by the failure to download the Linux kernel source from Github.
Check your network or proxy settings.

master882020, 2020-03-09:
> This is caused by the failure to download the Linux kernel source from Github.
> Check your network or proxy settings.

@mine260309 Thanks a lot;！
Tthe error log show
[log.do_fetch .5089.txt](https://github.com/openbmc/openbmc/files/4306554/log.do_fetch.5089.txt)

warning: redirecting to https://github.com/openbmc/linux/
remote: Enumerating objects: 1, done.
remote: Counting objects: 100% (1/1), done.
error: RPC failed; curl 18 transfer closed with outstanding read data remaining
fatal: The remote end hung up unexpectedly
fatal: early EOF
fatal: index-pack failed

but the terminal show download speed about 300KiB/s !
like:
linux-aspeed-5.4.23+gitAUTOINC+6ba7086874-r0 do_fetch (pid 51399)  27% | ######                 | 276KiB/s

I tried to modify the git configuration according to the online method, but it was still this error!

Please tell me how can I do！

Thank you very much ！

master882020, 2020-03-10:
I don't konw  why！
but I repeatedly run to build and solve it.
Maybe the network is bad ！

closed 2020-03-10 by master882020