openbmc/openbmc #3649: initramfs doesn't have password set

shenki, 2020-03-12:
When booting openbmc master on a tacoma (SPI NOR layout with initramfs), if the rootfs fails to mount we drop to an emergency console. In the past this had the same root password as the default system password. Now this does not work:

```
mount: mounting /dev/mtdblock5 on run/initramfs/rw failed: Input/output error

Mounting read-write /dev/mtdblock5 filesystem failed.  Please fix and run
	mount /dev/mtdblock5 run/initramfs/rw -t jffs2 -o rw
to to continue, or do change nothing to run from RAM for this boot.
Enter password to try to manually fix.
After fixing run exit to continue this script, or reboot -f to retry, or
touch /takeover and exit to become PID 1 allowing editing of this script.
Give root password for system maintenance
(or type Control-D for normal startup):
sulogin: Login incorrect
Give root password for system maintenance
(or type Control-D for normal startup):
sulogin: Login incorrect
Give root password for system maintenance
(or type Control-D for normal startup):
```

joseph-reynolds, 2020-03-12:
Was the BMC root user password changed away from 0penBmc?  If that is the issue, you can force it back as noted in the expired-password cheatsheet: https://gerrit.openbmc-project.xyz/c/openbmc/docs/+/25706/2/expired-password-cheatsheet.md#120

- If you have changed the password and wish to set it back to 0penBmc,
  access the BMC shell running as the root user, and enter the
  following command:
  ```
  sed -i -e '/^root:/c\root:$1$UGMqyqdG$FZiylVFmRRfl9Z0Ue8G7e/:18148:0:99999:7:::' /etc/shadow
  ```
  Note the `passwd` command cannot be used because 0penBmc does not meet password complexity rules.

shenki, 2020-03-12:
No. This is a brand new image built from CI, flashed on a board for bringup.

joseph-reynolds, 2020-03-12:
Ah, oh.  I am not aware of any other password-related changes.

mdmillerii, 2020-03-31:
https://gerrit.openbmc-project.xyz/c/openbmc/meta-phosphor/+/30260

closed 2020-07-13 by geissonator