openbmc/openbmc #3651: Libpam compilation issue

HCLOpenBMC, 2020-03-23:
I have cloned latest openbmc build, got libpam compilation issue. Please find the below error log. 
Please suggest to resolve this issue.

openbmc-linux-gnueabi/libpam/1.3.1-r0/recipe-sysroot-native/lib
| parse_y.o: In function `yyparse':
| parse_y.c:(.text+0xc23): undefined reference to `__warn_memset_zero_len'
| collect2: error: ld returned 1 exit status
| ERROR: oe_runmake failed
| Makefile:459: recipe for target 'padout' failed
| make[4]: *** [padout] Error 1
| make[4]: Leaving directory '/home/openbmc/kumar/yv2/openbmc/build/tmp/work/arm1176jzs-openbmc-linux-gnueabi/libpam/1.3.1-r0/build/doc/specs'
| Makefile:412: recipe for target 'all' failed
| make[3]: *** [all] Error 2
| make[3]: Leaving directory '/home/openbmc/kumar/yv2/openbmc/build/tmp/work/arm1176jzs-openbmc-linux-gnueabi/libpam/1.3.1-r0/build/doc/specs'
| Makefile:466: recipe for target 'all-recursive' failed
| make[2]: *** [all-recursive] Error 1
| make[2]: Leaving directory '/home/openbmc/kumar/yv2/openbmc/build/tmp/work/arm1176jzs-openbmc-linux-gnueabi/libpam/1.3.1-r0/build/doc'
| Makefile:472: recipe for target 'all-recursive' failed
| make[1]: *** [all-recursive] Error 1
| make[1]: Leaving directory '/home/openbmc/kumar/yv2/openbmc/build/tmp/work/arm1176jzs-openbmc-linux-gnueabi/libpam/1.3.1-r0/build'
| Makefile:404: recipe for target 'all' failed
| make: *** [all] Error 2
| WARNING: exit code 1 from a shell command.
| ERROR: Execution of '/home/openbmc/kumar/yv2/openbmc/build/tmp/work/arm1176jzs-openbmc-linux-gnueabi/libpam/1.3.1-r0/temp/run.do_compile.12283' failed with exit code 1:
| make  all-recursive
| make[1]: Entering directory '/home/openbmc/kumar/yv2/openbmc/build/tmp/work/arm1176jzs-openbmc-linux-gnueabi/libpam/1.3.1-r0/build'
| Making all in libpam
| make[2]: Entering directory '/home/openbmc/kumar/yv2/openbmc/build/tmp/work/arm1176jzs-openbmc-linux-gnueabi/libpam/1.3.1-r0/build/libpam'
| make[2]: Nothing to be done for 'all'.
| make[2]: Leaving directory '/home/openbmc/kumar/yv2/openbmc/build/tmp/work/arm1176jzs-openbmc-linux-gnueabi/libpam/1.3.1-r0/build/libpam'
| Making all in tests
| make[2]: Entering directory '/home/openbmc/kumar/yv2/openbmc/build/tmp/work/arm1176jzs-openbmc-linux-gnueabi/libpam/1.3.1-r0/build/tests'
| make[2]: Nothing to be done for 'all'.
| make[2]: Leaving directory '/home/openbmc/kumar/yv2/openbmc/build/tmp/work/arm1176jzs-openbmc-linux-gnueabi/libpam/1.3.1-r0/build/tests'
| Making all in libpamc
| make[2]: Entering directory '/home/openbmc/kumar/yv2/openbmc/build/tmp/work/arm1176jzs-openbmc-linux-gnueabi/libpam/1.3.1-r0/build/libpamc'
| Making all in test
| make[3]: Entering directory '/home/openbmc/kumar/yv2/openbmc/build/tmp/work/arm1176jzs-openbmc-linux-gnueabi/libpam/1.3.1-r0/build/libpamc/test'
| make[3]: Nothing to be done for 'all'.
| make[3]: Leaving directory '/home/openbmc/kumar/yv2/openbmc/build/tmp/work/arm1176jzs-openbmc-linux-gnueabi/libpam/1.3.1-r0/build/libpamc/test'
| make[3]: Entering directory '/home/openbmc/kumar/yv2/openbmc/build/tmp/work/arm1176jzs-openbmc-linux-gnueabi/libpam/1.3.1-r0/build/libpamc'
| make[3]: Nothing to be done for 'all-am'.
| make[3]: Leaving directory '/home/openbmc/kumar/yv2/openbmc/build/tmp/work/arm1176jzs-openbmc-linux-gnueabi/libpam/1.3.1-r0/build/libpamc'
| make[2]: Leaving directory '/home/openbmc/kumar/yv2/openbmc/build/tmp/work/arm1176jzs-openbmc-linux-gnueabi/libpam/1.3.1-r0/build/libpamc'
| Making all in libpam_misc
| make[2]: Entering directory '/home/openbmc/kumar/yv2/openbmc/build/tmp/work/arm1176jzs-openbmc-linux-gnueabi/libpam/1.3.1-r0/build/libpam_misc'
| make[2]: Nothing to be done for 'all'.
| make[2]: Leaving directory '/home/openbmc/kumar/yv2/openbmc/build/tmp/work/arm1176jzs-openbmc-linux-gnueabi/libpam/1.3.1-r0/build/libpam_misc'
| Making all in modules
| make[2]: Entering directory '/home/openbmc/kumar/yv2/openbmc/build/tmp/work/arm1176jzs-openbmc-linux-gnueabi/libpam/1.3.1-r0/build/modules'
| Making all in pam_access
| make[3]: Entering directory '/home/openbmc/kumar/yv2/openbmc/build/tmp/work/arm1176jzs-openbmc-linux-gnueabi/libpam/1.3.1-r0/build/modules/pam_access'
| make[3]: Nothing to be done for 'all'.
| make[3]: Leaving directory '/home/openbmc/kumar/yv2/openbmc/build/tmp/work/arm1176jzs-openbmc-linux-gnueabi/libpam/1.3.1-r0/build/modules/pam_access'
| Making all in pam_cracklib
| make[3]: Entering directory '/home/openbmc/kumar/yv2/openbmc/build/tmp/work/arm1176jzs-openbmc-linux-gnueabi/libpam/1.3.1-r0/build/modules/pam_cracklib'
| make[3]: Nothing to be done for 'all'.
| make[3]: Leaving directory '/home/openbmc/kumar/yv2/openbmc/build/tmp/work/arm1176jzs-openbmc-linux-gnueabi/libpam/1.3.1-r0/build/modules/pam_cracklib'
| Making all in pam_debug
| make[3]: Entering directory '/home/openbmc/kumar/yv2/openbmc/build/tmp/work/arm1176jzs-openbmc-linux-gnueabi/libpam/1.3.1-r0/build/modules/pam_debug'
| make[3]: Nothing to be done for 'all'.
| make[3]: Leaving directory '/home/openbmc/kumar/yv2/openbmc/build/tmp/work/arm1176jzs-openbmc-linux-gnueabi/libpam/1.3.1-r0/build/modules/pam_debug'
| Making all in pam_deny
| make[3]: Entering directory '/home/openbmc/kumar/yv2/openbmc/build/tmp/work/arm1176jzs-openbmc-linux-gnueabi/libpam/1.3.1-r0/build/modules/pam_deny'
| make[3]: Nothing to be done for 'all'.
| make[3]: Leaving directory '/home/openbmc/kumar/yv2/openbmc/build/tmp/work/arm1176jzs-openbmc-linux-gnueabi/libpam/1.3.1-r0/build/modules/pam_deny'
| Making all in pam_echo
| make[3]: Entering directory '/home/openbmc/kumar/yv2/openbmc/build/tmp/work/arm1176jzs-openbmc-linux-gnueabi/libpam/1.3.1-r0/build/modules/pam_echo'
| make[3]: Nothing to be done for 'all'.
| make[3]: Leaving directory '/home/openbmc/kumar/yv2/openbmc/build/tmp/work/arm1176jzs-openbmc-linux-gnueabi/libpam/1.3.1-r0/build/modules/pam_echo'
[... 200 lines cut ...]
| make[4]: Leaving directory '/home/openbmc/kumar/yv2/openbmc/build/tmp/work/arm1176jzs-openbmc-linux-gnueabi/libpam/1.3.1-r0/build/doc/specs'
| Makefile:412: recipe for target 'all' failed
| make[3]: *** [all] Error 2
| make[3]: Leaving directory '/home/openbmc/kumar/yv2/openbmc/build/tmp/work/arm1176jzs-openbmc-linux-gnueabi/libpam/1.3.1-r0/build/doc/specs'
| Makefile:466: recipe for target 'all-recursive' failed
| make[2]: *** [all-recursive] Error 1
| make[2]: Leaving directory '/home/openbmc/kumar/yv2/openbmc/build/tmp/work/arm1176jzs-openbmc-linux-gnueabi/libpam/1.3.1-r0/build/doc'
| Makefile:472: recipe for target 'all-recursive' failed
| make[1]: *** [all-recursive] Error 1
| make[1]: Leaving directory '/home/openbmc/kumar/yv2/openbmc/build/tmp/work/arm1176jzs-openbmc-linux-gnueabi/libpam/1.3.1-r0/build'
| Makefile:404: recipe for target 'all' failed
| make: *** [all] Error 2
| WARNING: exit code 1 from a shell command.
| 
NOTE: recipe libpam-1.3.1-r0: task do_compile: Failed
NOTE: recipe cracklib-2.9.5-r0: task do_packagedata: Succeeded
NOTE: recipe libtirpc-1.1.4-r0: task do_install: Started
NOTE: recipe bash-completion-2.9-r0: task do_compile: Succeeded
NOTE: recipe libtirpc-1.1.4-r0: task do_install: Succeeded
NOTE: Tasks Summary: Attempted 582 tasks of which 578 didn't need to be rerun and 1 failed.

Summary: 1 task failed:
  /home/openbmc/kumar/yv2/openbmc/meta/recipes-extended/pam/libpam_1.3.1.bb:do_compile
Summary: There were 5 WARNING messages shown.
Summary: There were 2 ERROR messages shown, returning a non-zero exit code.

mdmillerii, 2020-04-15:
> `| parse_y.o: In function yyparse': | parse_y.c:(.text+0xc23): undefined reference to __warn_memset_zero_len'`

looks like your bison is generating for some kind of source foritifcation that doesn't match the generated toolchain.

edineshkumar, 2020-06-16:
parse_y.o: In function yyparse': | parse_y.c:(.text+0xc23): undefined reference to __warn_memset_zero_len'

It looks, there is a variable like **__warn_memset_zero_len** in parse_y.c file. It was **undefined**. May your corresponding make file doesn't link the required header file.

mdmillerii, 2020-06-16:
I still think it's source hardening testing for zero length arg to `memset` is beyond the base standard.  See https://www.man7.org/linux/man-pages/man3/memset.3.html for example.

A quick web search says the check was added to gcc 5, and patches are proposed for some projects.

Figure out if the check is pointing out an error in the code or a compiler vs libc version mismatch or shell incomplete constant propagation by the compiler.

closed 2021-01-23 by stale[bot]

TheNageek, 2022-06-14:
Bump, running into this issue currently.