openbmc/openbmc #3652: phosphor-power recipe not getting correct power-sequencer YAML file

bjwyman, 2020-03-31:
**Unexpected behavior you saw**
It appears as though after 7307bedd3ff50091e35d9c1fffe23fbef84b01d1 and 856ad83d54a62c6cf7e8de6fcddcd93ef011cd7d we end up only getting the example YAML file for the power-sequencer part of the phosphor-power build.

This just so happens to work out okay for Witherspoon builds, or anything that inherits from it, but other builds such as Swift end up getting the wrong `ucd90160_defs.cpp` file generated.

This results in a flurry of GPIO error messages during and after a poweron:
```
pseq-monitor[500]: Unable to find hwmon directory in device base path
pseq-monitor[500]: Could not find GPIO device path
pseq-monitor[515]: Unable to find hwmon directory in device base path
pseq-monitor[515]: Could not find GPIO device path
pseq-monitor[515]: Failed opening GPIO device
```

**Expected behavior**
The build for a specific system should be using the system specific YAML file (`ucd90160.yaml`) in order to generate the `ucd90160_defs.cpp` file that the power-sequencer application needs in order to correctly monitor for power good faults.

**To Reproduce**
Steps to reproduce the behavior:
1. clone/fetch/rebase openbmc/openbmc to latest.
2. export TEMPLATECONF=meta-ibm/meta-witherspoon/conf
3. DISTRO=openbmc-witherspoon MACHINE=swift bitbake obmc-phosphor-image
4. find tmp/work/ -name "ucd*.yaml"
5. Note it is example file and not https://github.com/openbmc/openbmc/blob/master/meta-ibm/meta-witherspoon/recipes-phosphor/power/power-sequencer/swift/ucd90160.yaml
6. find tmp/work/ -name "ucd*defs*.cpp"
7. Note incorrect bus/address due to incorrect ucd90160.yaml input file.

**OpenBMC Information:**

**Additional context**

geissonator, 2020-03-31:
Any thoughts or idead appreciated @mine260309

mine260309, 2020-04-01:
The `power-sequencer` is a dependency of `phosphor-power-sequencer`, which is enabled for `ibm-ac-server`.

I thought all the machines in `meta-witherspoon` except for `mihawk` and `rainier` are belong to `ibm-ac-server`, but it looks like not...

We need to fix `meta-ibm/meta-witherspoon/recipes-phosphor/packagegroups/packagegroup-obmc-apps.bbappend` to install the expected the `phosphor-power-xxx` packages for expected machines.

@geissonator @bjwyman

bjwyman, 2020-04-02:
That does not seem to be the issue. The sequencer package part is picked up via that variable stuff.

mine260309, 2020-04-03:
> That does not seem to be the issue. The sequencer package part is picked up via that variable stuff.

I checked the history of the recipe, when it was `witherspoon-pfault-analysis`, it is only included by `RDEPENDS_${PN}-extras_append_ibm-ac-server`, so at that time I think swift does not include this package.

Now it's `phosphor-power`, and it remains the same that only `RDEPENDS_${PN}-extras_append_ibm-ac-server` includes the package, so swift should not really include this package.

So I am guessing that the `ucd*.yaml` found in `tmp/work/` is not the file for swift, but some file in the previous build.

bjwyman, 2020-04-08:
The EXTRA_OEMESON part for specifying the file somehow does not get down to meson, so the build uses the example file from the phosphor-power repository instead of the YAML file in openbmc/openbmc/meta-ibm/meta-witherspoon/recipes-phosphor/power.

mine260309, 2020-04-09:
@bjwyman Please try to add `power-sequencer` package for Swift and see if the issue is resolved.
Something like:
```
--- a/meta-ibm/meta-witherspoon/recipes-phosphor/packagegroups/packagegroup-obmc-apps.bbappend
+++ b/meta-ibm/meta-witherspoon/recipes-phosphor/packagegroups/packagegroup-obmc-apps.bbappend
@@ -20,6 +20,7 @@ RDEPENDS_${PN}-fan-control_append_ibm-ac-server = " witherspoon-fan-watchdog"
 RDEPENDS_${PN}-extras_append_ibm-ac-server = " ${POWER_SERVICE_PACKAGES_AC_SERVER} witherspoon-power-supply-sync phosphor-webui"
 RDEPENDS_${PN}-extras_append_rainier = " ${POWER_SERVICE_PACKAGES_RAINIER} phosphor-webui"
 RDEPENDS_${PN}-extras_append_mihawk = " phosphor-webui phosphor-image-signing"
+RDEPENDS_${PN}-extras_append_swift = " ${POWER_SERVICE_PACKAGES_AC_SERVER}"
 RDEPENDS_${PN}-extras_remove_rainier = "obmc-ikvm liberation-fonts uart-render-controller"
 RDEPENDS_${PN}-extras_remove_swift = "obmc-ikvm"
 RDEPENDS_${PN}-extras_remove_witherspoon-tacoma = "obmc-ikvm liberation-fonts uart-render-controller"
```

bradbishop, 2020-04-09:
I’ve found the issue and I’m working on a patch.

bjwyman, 2020-04-09:
Resolved-by: https://gerrit.openbmc-project.xyz/c/openbmc/meta-ibm/+/31191

My initial thought was that this could not be the fix, as I had already tried that.

I pulled down that change, re-ran my bitbake, got the wrong `ucd90160_defs.cpp` generated file result. I thought I must be missing something. I started a new shell session, cleared out the build directory, and ran a new bitbake build. The end result of that *was* a correct `ucd90160_defs.cpp` file.

So, the change that @bradbishop pushed up as a fix really does work, you just cannot completely re-use an existing build session. Clearing out `tmp` (or possibly forcing a rebuild of `phosphor-power`) may also work.

mdmillerii, 2020-04-09:
sounds like a missing dependency

bjwyman, 2020-04-09:
Well, I am not quite sure.

It looks like if you want to re-use an existing bitbake session (is that a bad idea?), then the best/fastest way to do that and get the correct result (for this problem) would be:
`bitbake -f -c compile phosphor-power`
`bitbake obmc-phosphor-image`

The package files after the forced compiled of phosphor-power will still be wrong, but the build file (`tmp/work/<system...>/phosphor-power/...gitAUTOINC+.../build/power-equencer/ucd90160_defs.cpp`) will be correct.

During the obmc-phosphor-image run, there will be complaints about `phosphor-power_git.bb:do_compile is tainted from a forced run` and `phosphor-power_git.bb:do_build is tainted from a forced run`, resulting in some do_package stuff for phosphor-power and redo of obmc-phosphor-image. The end result is that all the `ucd90160_defs.cpp` files found are the correct one.

For me, the full start over took about 40 minutes, the forced compile and redo of obmc-phosphor-image totaled up to about 3 minutes.

bradbishop, 2020-04-10:
at 12:07 PM, Milton D. Miller II <notifications@github.com> wrote:

>
> sounds like a missing dependency

here is the build target (meson):

```
ucd90160_defs_cpp = custom_target(
     'ucd90160_defs.cpp',
     command: [
         prog_python, '@INPUT0@',
         '-i', '@INPUT1@', '-o', meson.current_build_dir(),
     ],
     input: [
         'gen-ucd90160-defs.py',
         get_option('ucd90160-yaml'),
         'templates/ucd90160_defs.mako.cpp'
     ],
     output: 'ucd90160_defs.cpp',
)

executable(
     'pseq-monitor',
     'argument.cpp',
     error_hpp,
     'main.cpp',
     'pgood_monitor.cpp',
     'runtime_monitor.cpp',
     'ucd90160.cpp',
     ucd90160_defs_cpp,
     dependencies: [
         phosphor_dbus_interfaces,
         phosphor_logging,
         sdbusplus,
         sdeventplus,
     ],
     include_directories: '..',
     install: true,
     link_with: libpower,
)
```

An alternative to debugging this would be to move this generated code to a  
runtime configuration… ;-)

bradbishop, 2020-04-10:
Here it is again since email replies don't support markdown....

`ucd90160_defs_cpp = custom_target(
     'ucd90160_defs.cpp',
     command: [
         prog_python, '@input0@',
         '-i', '@INPUT1@', '-o', meson.current_build_dir(),
     ],
     input: [
         'gen-ucd90160-defs.py',
         get_option('ucd90160-yaml'),
         'templates/ucd90160_defs.mako.cpp'
     ],
     output: 'ucd90160_defs.cpp',
)

executable(
     'pseq-monitor',
     'argument.cpp',
     error_hpp,
     'main.cpp',
     'pgood_monitor.cpp',
     'runtime_monitor.cpp',
     'ucd90160.cpp',
     ucd90160_defs_cpp,
     dependencies: [
         phosphor_dbus_interfaces,
         phosphor_logging,
         sdbusplus,
         sdeventplus,
     ],
     include_directories: '..',
     install: true,
     link_with: libpower,
)`

bjwyman, 2020-04-10:
There are some plans to move that `pseq-monitor` application away from using that YAML configuration generated file.

bjwyman, 2020-04-14:
https://github.ibm.com/openbmc/openbmc/commit/073614cb41d4fe3a095f9211243ef73425386194

closed 2020-04-14 by bjwyman