openbmc/openbmc #3653: Fan control failure on Tacoma

shenki, 2020-04-07:
Running master (2.8.0-dev-1550-g5dd431aa0) on Tacoma gives this after booting the host:
```
Apr 07 23:47:35 spork-bmc phosphor-hwmon-readd[383]: Failed to read from device.
Apr 07 23:47:35 spork-bmc phosphor-fru-fault-monitor[298]: /xyz/openbmc_project/logging/entry/50 created
Apr 07 23:47:35 spork-bmc phosphor-log-manager[234]: Event not found in PEL message registry: xyz.openbmc_project.Sensor.Device.Error.ReadFailure
Apr 07 23:47:35 spork-bmc phosphor-hwmon-readd[383]: Logging failing sysfs file
Apr 07 23:47:35 spork-bmc systemd[1]: xyz.openbmc_project.Hwmon@-ahb-apb-bus\x401e78a000-i2c\x2dbus\x40200-max31785\x4052.service: Main process exited, code=exited, status=1/FAILURE
Apr 07 23:47:35 spork-bmc systemd[1]: xyz.openbmc_project.Hwmon@-ahb-apb-bus\x401e78a000-i2c\x2dbus\x40200-max31785\x4052.service: Failed with result 'exit-code'.
Apr 07 23:47:36 spork-bmc systemd[1]: xyz.openbmc_project.Hwmon@-ahb-apb-bus\x401e78a000-i2c\x2dbus\x40200-max31785\x4052.service: Service RestartSec=1s expired, scheduling restart.
Apr 07 23:47:36 spork-bmc systemd[1]: xyz.openbmc_project.Hwmon@-ahb-apb-bus\x401e78a000-i2c\x2dbus\x40200-max31785\x4052.service: Scheduled restart job, restart counter is at 2.
Apr 07 23:47:36 spork-bmc systemd[1]: Stopped Phosphor Hwmon Poller.
Apr 07 23:47:36 spork-bmc systemd[1]: Started Phosphor Hwmon Poller.
```

When rebooting, there are many fan related errors:

```
Apr 07 23:53:40 spork-bmc systemd[1]: Starting MAX31785 hwmon helper...
Apr 07 23:53:40 spork-bmc max31785-hwmon-helper[752]: cp: can't stat '/etc/default/obmc/hwmon/ahb/apb/bus@1e78a000/i2c-bus@100/max31785@52_air.conf': No such file or directory
```

[journal.txt](https://github.com/openbmc/openbmc/files/4447492/journal.txt)

msbarth, 2020-04-08:
Thats the wrong i2c address of the max31785 that's used to load the correct hwmon config. Hwmon is configured to look at `i2c-bus@200` for the max31785's config.

https://github.com/openbmc/openbmc/blob/master/meta-ibm/meta-witherspoon/recipes-phosphor/sensors/phosphor-hwmon_%25.bbappend#L55-L57

Have seen this before and believe it was related to an i2c issue between the BMC and max31785.

msbarth, 2020-04-08:
Does the BMC build version in `/etc/os-release` show this is a Tacoma image? Witherspoon had the `i2c-bus@100` location, so wondering if something happened with the build you're using.

msbarth, 2020-04-21:
The issue deals with tacoma getting the witherspoon specific overridden `phosphor-fan-control@.service` installed instead of the base service file. Found that the tacoma machine type picks up witherspoon machine recipe overrides which is what's causing this to occur. I have found a possible solution that will provide which i2c address to be used within this overridden service file from the recipe at build time.

msbarth, 2020-04-22:
Resolved by https://gerrit.openbmc-project.xyz/31627

closed 2020-11-22 by stale[bot]