openbmc/openbmc #3654: ast2400 not found phy

wws66, 2020-04-09:
Palmetto code is used for ast2400,We need help in the following problems:
ftgmac100 1e660000. Ethernet (unnamed net_device) (uninitialized) : eth%d: no PHY found
ftgmac100 1e660000.ethernet : MII probe failed!

We use the lastest code and query the solution of similar problems. They have been solved,but we still encounter the above problems.

mdmillerii, 2020-04-13:
Palmetto should be using a NCSI connection and doesn't need a phy.
https://github.com/openbmc/linux/blob/fdc60468f3e452364d432f1a7c3f83d58bba1b84/arch/arm/boot/dts/aspeed-bmc-opp-palmetto.dts#L138
Perhaps you need to enable ncsi configs in your kernel?

wws66, 2020-04-14:
Now our hardware doesn't support ncsi function, so we choose to disable the ncsi function. May I ask if the palmetto code supports the function of scanning phy, we need to change those parameters or change another code?

mdmillerii, 2020-04-14:
The kernel does, but you will need to describe the phy hardware you have in the device tree for your board.  

For u-boot, you will need to change the config to disable NCSI mode (its compile time especially in the code that supports 2400).

Obviously you are not using a palmetto board.  If you are not using a POWER processor on your board, then you may not want the meta-openpower layer; the x86 cpus tend to have different platform control interfaces (eg OpenPOWER uses a GPIO state to determine desired power while some x86 platforms toggle a GPIO to simulate pushing the power button).

wws66, 2020-04-17:
We have modified the configuration of the kernel and Uboot, now the network of Uboot is normal, but for the kernel, it prompts some errors, could you please help us?

The following is the phy information we added in the device tree(The phy we used is RTL8211E, but the result of the scan is a universal phy.):
&mac0{
        
       status = "okay";
	
	//use-ncsi;
	phy-mode = "rgmii";
	phy-handle = <&ethphy1>;	

	pinctrl-names = "default";
	pinctrl-0 = <&pinctrl_rgmii1_default>;
	
        fixed-link {
		speed = <1000>;
		full-duplex;
	};

	ethphy1: phy@0{
		compatible = "ethernet-phy-id001c.c915",
				"ethernet-phy-ieee802.3-c22";
		reg = <0>;
	};
Here are the errors we encountered:

[    4.926495] libphy: Fixed MDIO Bus: probed
[    4.931991] ftgmac100 1e660000.ethernet: Error applying setting, reverse things back
[    4.940272] ftgmac100 1e660000.ethernet: Generated random MAC address ba:4c:8c:8a:fe:b7
[    4.949343] Generic PHY fixed-0:00: attached PHY driver [Generic PHY] (mii_bus:phy_addr=fixed-0:00, irq=POLL)
[    4.961111] ftgmac100 1e660000.ethernet eth0: irq 19, mapped at b862ec3c

root@palmetto:~# [  144.083761] ------------[ cut here ]------------
[  144.088467] WARNING: CPU: 0 PID: 146 at net/sched/sch_generic.c:447 dev_watchdog+0x270/0x290
[  144.097056] NETDEV WATCHDOG: eth0 (ftgmac100): transmit queue 0 timed out
[  144.103979] CPU: 0 PID: 146 Comm: rngd Not tainted 5.3.15-c008a29 #1
[  144.110350] Hardware name: Generic DT based system
[  144.115247] Backtrace: 
[  144.117779] [<80107754>] (dump_backtrace) from [<801079a8>] (show_stack+0x20/0x24)
[  144.125495]  r7:80a12b20 r6:00000000 r5:8d7dbe04 r4:00000009
[  144.131234] [<80107988>] (show_stack) from [<8061d300>] (dump_stack+0x20/0x28)
[  144.138621] [<8061d2e0>] (dump_stack) from [<80114008>] (__warn.part.0+0xb4/0xdc)
[  144.146258] [<80113f54>] (__warn.part.0) from [<801140a4>] (warn_slowpath_fmt+0x74/0x9c)
[  144.154466]  r6:8051ff3c r5:000001bf r4:8080c0ec
[  144.159136] [<80114030>] (warn_slowpath_fmt) from [<8051ff3c>] (dev_watchdog+0x270/0x290)
[  144.167439]  r3:8c117800 r2:8080c0b0
[  144.171047]  r6:00000000 r5:8c102680 r4:8c117a2c
[  144.175807] [<8051fccc>] (dev_watchdog) from [<8015b33c>] (call_timer_fn+0x40/0x124)
[  144.183596]  r9:00000122 r8:ffffc318 r7:00000100 r6:8051fccc r5:8c117a2c r4:8c117a2c
[  144.191485] [<8015b2fc>] (call_timer_fn) from [<8015b4c8>] (expire_timers+0xa8/0xec)
[  144.199368]  r8:ffffc318 r7:80a13840 r6:8d7dbecc r5:8d7dbecc r4:8c117a2c
[  144.206214] [<8015b420>] (expire_timers) from [<8015b5b4>] (run_timer_softirq+0xa8/0x1b8)
[  144.214521]  r10:00000002 r9:00000100 r8:80a12b20 r7:80a13840 r6:80a13840 r5:8d7dbecc
[  144.222375]  r4:8d7dbecc r3:ffffc319
[  144.226100] [<8015b50c>] (run_timer_softirq) from [<80102284>] (__do_softirq+0xe4/0x328)
[  144.234321]  r8:00000000 r7:8d7da000 r6:80a5b50c r5:00000001 r4:00000002
[  144.241074] [<801021a0>] (__do_softirq) from [<80118144>] (irq_exit+0xac/0xe4)
[  144.248444]  r10:00000001 r9:00000000 r8:00000000 r7:8d40f600 r6:00000001 r5:80a45908
[  144.256393]  r4:00000000
[  144.259003] [<80118098>] (irq_exit) from [<8014d128>] (__handle_domain_irq+0x60/0xb4)
[  144.266998] [<8014d0c8>] (__handle_domain_irq) from [<80102184>] (avic_handle_irq+0x58/0x74)
[  144.275588]  r9:00000000 r8:00053177 r7:0005317f r6:ffffffff r5:8d7dbfb0 r4:8d402e40
[  144.283377] [<8010212c>] (avic_handle_irq) from [<80101df8>] (__irq_usr+0x58/0x80)
[  144.291067] Exception stack(0x8d7dbfb0 to 0x8d7dbff8)
[  144.296250] bfa0:                                     535777d3 00000000 00000000 00000000
[  144.304548] bfc0: 00000001 00000000 00000000 00000000 012734e0 00000000 00000001 76f62d14
[  144.312756] bfe0: 00000001 76f62c50 48f90bf0 48f90f00 80000010 ffffffff
[  144.319492]  r5:80000010 r4:48f90f00
[  144.323087] ---[ end trace fb6457412bc1020f ]---

mdmillerii, 2020-04-17:
In the future, please consider using ` ``` ` tripple backticks or single backticks before and after your code block to avoid accidently triggering markup. There is a M and down arrow on the edit box that will give you help on [Markdown](https://guides.github.com/features/mastering-markdown/)

```
[ 4.926495] libphy: Fixed MDIO Bus: probed
[ 4.931991] ftgmac100 1e660000.ethernet: Error applying setting, reverse things back
```
I'm not sure if Fixed MDIO bus is right, it might mean your phy node needs adjustment.  I thought Fixed PHY was for direct *MII connections to other devices on backplanes (eg backplane switch) but I don't dig into this.

```
[ 4.940272] ftgmac100 1e660000.ethernet: Generated random MAC address ba:4c:8c:8a:fe:b7
```
This is informationand and expected for aspeed.  The the SOC does not have any storage to set the MAC by default.  The driver will use a MAC if it is set by u-boot and not a multicast address.  The u-boot code in the default ast2400 bsp will write a MAC to the phy if the `ethaddr` is found in the u-boot environment (see https://github.com/openbmc/u-boot/blob/66f534ae1d7d364d88abf18ad2d86b02cf9eda47/doc/README.enetaddr#L19 )

There is code that will take a MAC address from inventory and write it to the u-boot environment ; custom board code will need to populate the inventory from the system specific storage location (usually a i2c eeprom somewhere on the system).

```
[ 4.949343] Generic PHY fixed-0:00: attached PHY driver [Generic PHY] (mii_bus:phy_addr=fixed-0:00, irq=POLL)
```
The fact that the PHY is fixed-0:00 tells me Linux did not find your RTL8211E.

```
[ 4.961111] ftgmac100 1e660000.ethernet eth0: irq 19, mapped at b862ec3c
```
```
root@palmetto:~# [ 144.083761] ------------[ cut here ]------------
[ 144.088467] WARNING: CPU: 0 PID: 146 at net/sched/sch_generic.c:447 dev_watchdog+0x270/0x290
[ 144.097056] NETDEV WATCHDOG: eth0 (ftgmac100): transmit queue 0 timed 
```
This trace and the output notifies you that the device is not sending a packet that was queued for transmit.  Normally this is blocked until link up is detected but that won't happen with a working PHY driver.

wws66, 2020-04-17:
Ok, we'll check further. Thank you very much for your help and wish you a happy life.

closed 2020-04-23 by wws66