openbmc/openbmc #3656: Build OpenBMC, meet some errors.

zzhuang-cancon, 2020-04-17:
Hey,
As a beginner of OpenBMC, I used time bitbake obmc-phosphor-image to compile the image,
Has anyone encountered such a problem?
**ERROR: Task (/home/zhanghzuang/openbmc/meta/recipes-devtools/cmake/cmake-native_3.15.3.bb:do_compile) failed with exit code '1'**

Below are the details：

| {standard input}: Assembler messages:
| {standard input}:45553: Warning: end of file not at end of a line; newline inserted
| {standard input}:45840: Error: unbalanced parenthesis in operand 1.
| {standard input}: Error: open CFI at the end of file; missing .cfi_endproc directive
| g++: internal compiler error: Killed (program cc1plus)
| Please submit a full bug report,
| with preprocessed source if appropriate.
| See <file:///usr/share/doc/gcc-5/README.Bugs> for instructions.
| Source/CMakeFiles/CMakeLib.dir/build.make:1053: recipe for target 'Source/CMakeFiles/CMakeLib.dir/cmGeneratorTarget.cxx.o' failed
| make[2]: *** [Source/CMakeFiles/CMakeLib.dir/cmGeneratorTarget.cxx.o] Error 4
| make[2]: *** Waiting for unfinished jobs....
| make[2]: Leaving directory '/home/zhanghzuang/openbmc/build/tmp/work/x86_64-linux/cmake-native/3.15.3-r0/build'
| CMakeFiles/Makefile2:1559: recipe for target 'Source/CMakeFiles/CMakeLib.dir/all' failed
| make[1]: *** [Source/CMakeFiles/CMakeLib.dir/all] Error 2
| make[1]: Leaving directory '/home/zhanghzuang/openbmc/build/tmp/work/x86_64-linux/cmake-native/3.15.3-r0/build'
| Makefile:165: recipe for target 'all' failed
| make: *** [all] Error 2
| WARNING: /home/zhanghzuang/openbmc/build/tmp/work/x86_64-linux/cmake-native/3.15.3-r0/temp/run.do_compile.104763:1 exit 1 from 'exit 1'
| 
### **ERROR: Task (/home/zhanghzuang/openbmc/meta/recipes-devtools/cmake/cmake-native_3.15.3.bb:do_compile) failed with exit code '1'**
NOTE: Tasks Summary: Attempted 854 tasks of which 804 didn't need to be rerun and 1 failed.

Summary: 1 task failed:
  /home/zhanghzuang/openbmc/meta/recipes-devtools/cmake/cmake-native_3.15.3.bb:do_compile
Summary: There were 2 ERROR messages shown, returning a non-zero exit code.

mdmillerii, 2020-04-17:
> 
```
| {standard input}: Assembler messages:
| {standard input}:45553: Warning: end of file not at end of a line; newline inserted
| {standard input}:45840: Error: unbalanced parenthesis in operand 1.
| {standard input}: Error: open CFI at the end of file; missing .cfi_endproc directive
| g++: internal compiler error: Killed (program cc1plus)
```

The fact that your compiler was killed after reporting input that died mid-line probably indicates your system ran out of memory.  Assuming you are building under a Linux kernel check  `dmesg` for `OOM` (out-of-memory) reports.

If so bitbake can limit the number of tasks running at the same time to less than the number of your cpus.

zzhuang-cancon, 2020-04-17:
First of all, thank you very much,This is my first contact with OpenBMC.
I also suspected that there was not enough memory，but I allocated 200G of disk space to my virtual machine. Is this not enough? 
 And according to your description, I remember to configure the compile options in the local.conf directory before the compilation is executed, look:
[image]
  maybe, i can change this conf.

mdmillerii, 2020-04-17:
You ran out of Memory, not  disk space.

zzhuang-cancon, 2020-04-18:
Ok, I understand, thank you very much, I have now found the more comprehensive and comprehensive tutorial provided by the official OpenBMC, I will learn step by step, thanks again

closed 2020-11-19 by stale[bot]