openbmc/openbmc #3657: dunfell: gcc-cross_9.3.bb: sed: -e expression #1, char 111: unknown option to `s'

geissonator, 2020-04-28:
I've been seeing this intermittently in testing. I'm not sure if it's related to the new dunfell release or something else.

```
19:13:58 | configure: creating ./config.status
19:13:58 | config.status: creating as
19:13:58 | config.status: creating collect-ld
19:13:58 | config.status: creating nm
19:13:58 | config.status: creating Makefile
19:13:58 | config.status: creating ada/gcc-interface/Makefile
19:13:58 | config.status: creating ada/Makefile
19:13:58 | config.status: creating auto-host.h
19:13:58 | config.status: executing depdir commands
19:13:58 | mkdir -p -- .deps
19:13:58 | config.status: executing gccdepdir commands
19:13:58 | mkdir -p -- build/.deps
19:13:58 | mkdir -p -- ada/.deps
19:13:58 | mkdir -p -- brig/.deps
19:13:58 | mkdir -p -- c/.deps
19:13:58 | mkdir -p -- cp/.deps
19:13:58 | mkdir -p -- d/.deps
19:13:58 | mkdir -p -- fortran/.deps
19:13:58 | mkdir -p -- go/.deps
19:13:58 | mkdir -p -- jit/.deps
19:13:58 | mkdir -p -- lto/.deps
19:13:58 | mkdir -p -- objc/.deps
19:13:58 | mkdir -p -- objcp/.deps
19:13:58 | mkdir -p -- c-family/.deps
19:13:58 | mkdir -p -- common/.deps
19:13:58 | config.status: executing default commands
19:13:58 | sed: -e expression #1, char 111: unknown option to `s'
19:13:58 | WARNING: exit code 1 from a shell command.
19:13:58 | 
19:13:58 ERROR: Task (/home/jenkins-slave/workspace/run-openbmc-ci/distro/ubuntu/label/builder/target/zaius/openbmc/meta/recipes-devtools/gcc/gcc-cross_9.3.bb:do_compile) failed with exit code '1'
19:13:58 Waiting for 1 running tasks to finish:
19:14:00 Waiting for 1 running tasks to finish:
19:14:01 Waiting for 0 running tasks to finish:
19:14:04 NOTE: [29mTasks Summary: Attempted 2894 tasks of which 932 didn't need to be rerun and 1 fa
19:14:04 Waiting for 0 running tasks to finish:
```

Not finding anything obvious in web searches. Some mentions that usually this error is a sign of an inefficient sed command.

https://openpower.xyz/job/build-openbmc/54902/console wont be around long but it's a link to the fail. 

Trying to build this bump - https://gerrit.openbmc-project.xyz/c/openbmc/openbmc/+/31716/1

zaius machine, builder3

geissonator, 2020-04-28:
Saw this pop up in just a bump as well (unrelated to Dunfell)

```
16:55:01 ERROR: net-snmp-5.8-r0 do_populate_sysroot: Error executing a python function in exec_python_func() autogenerated:
16:55:01 
16:55:01 The stack trace of python calls that resulted in this exception/failure was:
16:55:01 File: 'exec_python_func() autogenerated', lineno: 2, function: <module>
16:55:01      0001:
16:55:01  *** 0002:do_populate_sysroot(d)
16:55:01      0003:
16:55:01 File: '/home/jenkins-slave/workspace/run-meta-ci/distro/ubuntu/label/builder/target/witherspoon/openbmc/meta/classes/staging.bbclass', lineno: 99, function: do_populate_sysroot
16:55:01      0095:python do_populate_sysroot () {
16:55:01      0096:    bb.build.exec_func("sysroot_stage_all", d)
16:55:01      0097:    bb.build.exec_func("sysroot_strip", d)
16:55:01      0098:    for f in (d.getVar('SYSROOT_PREPROCESS_FUNCS') or '').split():
16:55:01  *** 0099:        bb.build.exec_func(f, d)
16:55:01      0100:    pn = d.getVar("PN")
16:55:01      0101:    multiprov = d.getVar("MULTI_PROVIDER_WHITELIST").split()
16:55:01      0102:    provdir = d.expand("${SYSROOT_DESTDIR}${base_prefix}/sysroot-providers/")
16:55:01      0103:    bb.utils.mkdirhier(provdir)
16:55:01 File: '/home/jenkins-slave/workspace/run-meta-ci/distro/ubuntu/label/builder/target/witherspoon/openbmc/poky/bitbake/lib/bb/build.py', lineno: 249, function: exec_func
16:55:01      0245:    with bb.utils.fileslocked(lockfiles):
16:55:01      0246:        if ispython:
16:55:01      0247:            exec_func_python(func, d, runfile, cwd=adir)
16:55:01      0248:        else:
16:55:01  *** 0249:            exec_func_shell(func, d, runfile, cwd=adir)
16:55:01      0250:
16:55:01      0251:    try:
16:55:01      0252:        curcwd = os.getcwd()
16:55:01      0253:    except:
16:55:01 File: '/home/jenkins-slave/workspace/run-meta-ci/distro/ubuntu/label/builder/target/witherspoon/openbmc/poky/bitbake/lib/bb/build.py', lineno: 450, function: exec_func_shell
16:55:01      0446:    with open(fifopath, 'r+b', buffering=0) as fifo:
16:55:01      0447:        try:
16:55:01      0448:            bb.debug(2, "Executing shell function %s" % func)
16:55:01      0449:            with open(os.devnull, 'r+') as stdin, logfile:
16:55:01  *** 0450:                bb.process.run(cmd, shell=False, stdin=stdin, log=logfile, extrafiles=[(fifo,readfifo)])
16:55:01      0451:        finally:
16:55:01      0452:            os.unlink(fifopath)
16:55:01      0453:
16:55:01      0454:    bb.debug(2, "Shell function %s finished" % func)
16:55:01 File: '/home/jenkins-slave/workspace/run-meta-ci/distro/ubuntu/label/builder/target/witherspoon/openbmc/poky/bitbake/lib/bb/process.py', lineno: 182, function: run
16:55:01      0178:        if not stderr is None:
16:55:01      0179:            stderr = stderr.decode("utf-8")
16:55:01      0180:
16:55:01      0181:    if pipe.returncode != 0:
16:55:01  *** 0182:        raise ExecutionError(cmd, pipe.returncode, stdout, stderr)
16:55:01      0183:    return stdout, stderr
16:55:01 Exception: bb.process.ExecutionError: Execution of '/home/jenkins-slave/workspace/build-openbmc@2/build/work/arm1176jzs-openbmc-linux-gnueabi/net-snmp/5.8-r0/temp/run.net_snmp_sysroot_preprocess.36389' failed with exit code 1:
16:55:01 sed: -e expression #1, char 65: unknown option to `s'
16:55:01 WARNING: exit code 1 from a shell command.
```
witherspoon machine, builder5, building a power bump - https://gerrit.openbmc-project.xyz/c/openbmc/meta-phosphor/+/31713

https://openpower.xyz/job/build-openbmc/54871/

mdmillerii, 2020-04-28:
The log doesn't echo the command passed to sed, so you will either need to have the exception echo the command (case 1) or track down which config failed (2).

geissonator, 2020-04-28:
Saw this again on two runs last night:

gsj machine, builder4

```
23:22:30 Exception: bb.process.ExecutionError: Execution of '/home/jenkins-op/workspace/build-openbmc@2/build/work/armv7a-openbmc-linux-gnueabi/net-snmp/5.8-r0/temp/run.net_snmp_sysroot_preprocess.8303' failed with exit code 1:
23:22:30 sed: -e expression #1, char 62: unknown option to `s'
23:22:30 WARNING: exit code 1 from a shell command.
23:22:30 
23:22:30 
23:22:30 Currently  4 running tasks (5098 of 5163)  98% ||

23:22:30 0: phosphor-ipmi-flash-1.0+gitAUTOINC+01593f9658-r1 do_compile - 19s (pid 1977)
23:22:30 1: net-snmp-5.8-r0 do_populate_sysroot - 0s (pid 8303)
23:22:30 ERROR: Logfile of failure stored in: /home/jenkins-op/workspace/build-openbmc@2/build/work/armv7a-openbmc-linux-gnueabi/net-snmp/5.8-r0/temp/log.do_populate_sysroot.8303
```

geissonator, 2020-05-04:
Turns out this was because our jenkins node was putting "@" symbols in the build paths (when a jenkins node has multiple jobs, it adds a @1, @2, @3, ... to the workspace path). I fixed this by having the jenkins job build locally within the docker contain (/tmp/openbmc/) and not in the provided WORKSPACE.

closed 2020-05-04 by geissonator