openbmc/openbmc #3658: systemd:u-boot: /var/lock and /run/lock usage

geissonator, 2020-05-04:
Need to look into an upstream change to u-boot that changes fw_setenv/fw_printenv to either create the /var/lock directory when it does not exist or look into alternatives when not available.

This commit was put in as a workaround:
https://gerrit.openbmc-project.xyz/c/openbmc/meta-phosphor/+/31860

https://github.com/systemd/systemd/issues/15668 was the discussion with systemd on this. In general it seems like the usage of /var/lock should be deprecated.

Need to open up some discussion with upstream u-boot team.

geissonator, 2022-12-08:
This came up recently in discord because of https://gerrit.openbmc.org/c/openbmc/openbmc/+/57952, which removed /var/lock, and caused the original issue documented here. We still need a solution. Just create the directory in the u-boot-fw-utils-aspeed-sdk_2019.04.bb recipe?

geissonator, 2023-08-08:
Our Witherspoon system appears to still be using u-boot packages from 2016 so it is not getting the fixes:
-  https://github.com/u-boot/u-boot/commit/aeb40f1166e072856f865d26d42a4bea1ec3a514
-  https://github.com/openbmc/u-boot/commit/d51f71c1d7d5b23a9a27e3ebaefdb4dda78cacb2

So after a dediprog or factory reset, a witherspoon system will not boot correctly with openbmc master. Need to figure out if we can update witherspoon to more recent u-boot.

williamspatrick, 2023-09-25:
@geissonator - Is there any plan here?

geissonator, 2023-09-25:
No, for now we just have it documented in our process that if you factory reset a witherspoon (running master), you have to manually create /var/lock (similar to what was done in 315698e53df for emmc). I don't think we're going to do anything else here.

closed 2023-09-25 by geissonator