openbmc/openbmc #3659: ast2600:emmc: Got data interrupt 0x00100000 even though no data operation was in progress

geissonator, 2020-05-04:
During a code update on one of our witherspoon-tacoma systems, we hit this. It just continued to spew out to the serial console until we power cycled it.

```
May 04 21:16:40 witherspoon-tacoma systemd[1]: dropbear@4-9.3.86.86:22-9.3.62.28:42662.service: Succeeded.
May 04 21:16:43 witherspoon-tacoma systemd[1]: systemd-hostnamed.service: Succeeded.
[  393.182270] mmc0: Timeout waiting for hardware interrupt.
[  393.188313] mmc0: sdhci: ============ SDHCI REGISTER DUMP ===========
[  393.195497] mmc0: sdhci: Sys addr:  0x00000008 | Version:  0x00000002
[  393.202680] mmc0: sdhci: Blk size:  0x00007200 | Blk cnt:  0x00000008
[  393.209862] mmc0: sdhci: Argument:  0x0027ede1 | Trn mode: 0x0000003b
[  393.217044] mmc0: sdhci: Present:   0x01f70206 | Host ctl: 0x00000017
[  393.224218] mmc0: sdhci: Power:     0x0000000f | Blk gap:  0x00000000
[  393.231401] mmc0: sdhci: Wake-up:   0x00000000 | Clock:    0x00000107
[  393.238583] mmc0: sdhci: Timeout:   0x0000000e | Int stat: 0x00000000
[  393.245766] mmc0: sdhci: Int enab:  0x03ff008b | Sig enab: 0x03ff008b
[  393.252948] mmc0: sdhci: ACmd stat: 0x00000000 | Slot int: 0x00000000
[  393.260122] mmc0: sdhci: Caps:      0x01f80080 | Caps_1:   0x00000007
[  393.267307] mmc0: sdhci: Cmd:       0x0000123a | Max curr: 0x001f0f08
[  393.274490] mmc0: sdhci: Resp[0]:   0x00000900 | Resp[1]:  0xffffffff
[  393.281672] mmc0: sdhci: Resp[2]:   0x320f5913 | Resp[3]:  0x00000900
[  393.288846] mmc0: sdhci: Host ctl2: 0x0000008b
[  393.293799] mmc0: sdhci: ADMA Err:  0x00000000 | ADMA Ptr: 0xbe040200
[  393.300980] mmc0: sdhci: ============================================
[  393.408165] mmc0: Reset 0x4 never completed.
[  393.412924] mmc0: sdhci: ============ SDHCI REGISTER DUMP ===========
[  393.420106] mmc0: sdhci: Sys addr:  0x00000008 | Version:  0x00000002
[  393.427281] mmc0: sdhci: Blk size:  0x00007200 | Blk cnt:  0x00000008
[  393.434464] mmc0: sdhci: Argument:  0x0027ede1 | Trn mode: 0x0000003b
[  393.441646] mmc0: sdhci: Present:   0x01f70000 | Host ctl: 0x00000017
[  393.448820] mmc0: sdhci: Power:     0x0000000f | Blk gap:  0x00000000
[  393.456002] mmc0: sdhci: Wake-up:   0x00000000 | Clock:    0x00000107
[  393.463176] mmc0: sdhci: Timeout:   0x0000000e | Int stat: 0x00208002
[  393.470358] mmc0: sdhci: Int enab:  0x03ff008b | Sig enab: 0x03ff008b
[  393.477532] mmc0: sdhci: ACmd stat: 0x00000000 | Slot int: 0x00000001
[  393.484714] mmc0: sdhci: Caps:      0x01f80080 | Caps_1:   0x00000007
[  393.491888] mmc0: sdhci: Cmd:       0x0000123a | Max curr: 0x001f0f08
[  393.499070] mmc0: sdhci: Resp[0]:   0x00000900 | Resp[1]:  0xffffffff
[  393.506251] mmc0: sdhci: Resp[2]:   0x320f5913 | Resp[3]:  0x00000900
[  393.513424] mmc0: sdhci: Host ctl2: 0x0000008b
[  393.518376] mmc0: sdhci: ADMA Err:  0x00000000 | ADMA Ptr: 0xbe040200
[  393.525557] mmc0: sdhci: ============================================
[  393.632975] mmc0: Reset 0x4 never completed.
[  393.637747] mmc0: sdhci: ============ SDHCI REGISTER DUMP ===========
[  393.644930] mmc0: sdhci: Sys addr:  0x00000008 | Version:  0x00000002
[  393.652112] mmc0: sdhci: Blk size:  0x00007200 | Blk cnt:  0x00000008
[  393.659296] mmc0: sdhci: Argument:  0x00000000 | Trn mode: 0x00000033
[  393.666478] mmc0: sdhci: Present:   0x01f70000 | Host ctl: 0x00000017
[  393.673651] mmc0: sdhci: Power:     0x0000000f | Blk gap:  0x00000000
[  393.680825] mmc0: sdhci: Wake-up:   0x00000000 | Clock:    0x00000107
[  393.688008] mmc0: sdhci: Timeout:   0x0000000e | Int stat: 0x00108000
[  393.695189] mmc0: sdhci: Int enab:  0x03ff008b | Sig enab: 0x03ff008b
[  393.702371] mmc0: sdhci: ACmd stat: 0x00000000 | Slot int: 0x00000001
[  393.709553] mmc0: sdhci: Caps:      0x01f80080 | Caps_1:   0x00000007
[  393.716735] mmc0: sdhci: Cmd:       0x00000c1a | Max curr: 0x001f0f08
[  393.723909] mmc0: sdhci: Resp[0]:   0x00000000 | Resp[1]:  0xffffffff
[  393.731092] mmc0: sdhci: Resp[2]:   0x320f5913 | Resp[3]:  0x00000900
[  393.738275] mmc0: sdhci: Host ctl2: 0x0000008b
[  393.743227] mmc0: sdhci: ADMA Err:  0x00000000 | ADMA Ptr: 0xbe040200
[  393.750408] mmc0: sdhci: ============================================

```

Also got some kernel backtraces during it intermittently
```
[  416.546465] rcu: INFO: rcu_sched self-detected stall on CPU
[  416.552694] rcu: 	0-....: (18 ticks this GP) idle=462/1/0x40000002 softirq=26686/26687 fqs=11 
[  416.562301] 	(t=2279 jiffies g=56549 q=38)
[  416.566869] NMI backtrace for cpu 0
[  416.570761] CPU: 0 PID: 50 Comm: irq/48-mmc0 Not tainted 5.4.32-be03941 #1
[  416.578428] Hardware name: Generic DT based system
[... 38 lines cut ...]
[  416.875950] 9e60: bcb25ce4 60000013 00000000 00004ac0 bcb25b80 bc5f4eb0 60000013 bcb25ce4
[  416.885076] 9e80: 00000062 00000000 00000000 bcb39ebc bcb39ec0 bcb39eb0 8068f4d8 80864cc8
[  416.894198] 9ea0: 60000013 ffffffff
[  416.898088]  r9:bcb38000 r8:00000062 r7:bcb39e94 r6:ffffffff r5:60000013 r4:80864cc8
[  416.906736] [<80864c94>] (_raw_spin_unlock_irqrestore) from [<8068f4d8>] (sdhci_request_done+0x124/0x288)
[  416.917415] [<8068f3b4>] (sdhci_request_done) from [<8068f65c>] (sdhci_thread_irq+0x20/0x7c)
[  416.926832]  r10:00000000 r9:8017a24c r8:bcb22580 r7:bcb38000 r6:bcb225a4 r5:bcb0a600
[  416.935565]  r4:bcb25b80
[  416.938389] [<8068f63c>] (sdhci_thread_irq) from [<8017a278>] (irq_thread_fn+0x2c/0x88)
[  416.947320]  r7:bcb38000 r6:bcb225a4 r5:bcb0a600 r4:bcb22580
[  416.953636] [<8017a24c>] (irq_thread_fn) from [<8017a5dc>] (irq_thread+0x12c/0x220)
[  416.962179]  r7:bcb38000 r6:bcb225a4 r5:bcb0a600 r4:00000000
[  416.968497] [<8017a4b0>] (irq_thread) from [<801475f8>] (kthread+0x154/0x168)
[  416.976451]  r10:bd0dda3c r9:8017a4b0 r8:bcb22580 r7:bcb38000 r6:00000000 r5:bcb22540
[  416.985184]  r4:bcb22c80
[  416.988007] [<801474a4>] (kthread) from [<801010e8>] (ret_from_fork+0x14/0x2c)
[  416.996061] Exception stack(0xbcb39fb0 to 0xbcb39ff8)
[  417.001687] 9fa0:                                     00000000 00000000 00000000 00000000
[  417.010812] 9fc0: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[  417.019937] 9fe0: 00000000 00000000 00000000 00000000 00000013 00000000
[  417.027316]  r10:00000000 r9:00000000 r8:00000000 r7:00000000 r6:00000000 r5:801474a4
[  417.036049]  r4:bcb22540
```

I was using an image I had generated off of our downstream release. Ping me for details if needed.

geissonator, 2020-05-04:
Noticed that later on the error message in serial changed to this:
```
[  470.057115] mmc0: Got data interrupt 0x00100000 even though no data operation was in progress.
[  470.066722] mmc0: sdhci: ============ SDHCI REGISTER DUMP ===========
[  470.073895] mmc0: sdhci: Sys addr:  0x00000008 | Version:  0x00000002
[  470.081078] mmc0: sdhci: Blk size:  0x00007200 | Blk cnt:  0x00000008
[  470.088259] mmc0: sdhci: Argument:  0x00000000 | Trn mode: 0x00000033
[  470.095441] mmc0: sdhci: Present:   0x01f70000 | Host ctl: 0x00000017
[  470.102615] mmc0: sdhci: Power:     0x0000000f | Blk gap:  0x00000000
[  470.109798] mmc0: sdhci: Wake-up:   0x00000000 | Clock:    0x00000107
[  470.116980] mmc0: sdhci: Timeout:   0x0000000e | Int stat: 0x00108000
[  470.124162] mmc0: sdhci: Int enab:  0x03ff008b | Sig enab: 0x03ff008b
[  470.131344] mmc0: sdhci: ACmd stat: 0x00000000 | Slot int: 0x00000001
[  470.138518] mmc0: sdhci: Caps:      0x01f80080 | Caps_1:   0x00000007
[  470.145700] mmc0: sdhci: Cmd:       0x00000c1a | Max curr: 0x001f0f08
[  470.152874] mmc0: sdhci: Resp[0]:   0x00000000 | Resp[1]:  0xffffffff
[  470.160056] mmc0: sdhci: Resp[2]:   0x320f5913 | Resp[3]:  0x00000900
[  470.167228] mmc0: sdhci: Host ctl2: 0x0000008b
[  470.172180] mmc0: sdhci: ADMA Err:  0x00000000 | ADMA Ptr: 0xbe040200
[  470.179353] mmc0: sdhci: ============================================
[  470.186535] mmc0: Got data interrupt 0x00100000 even though no data operation was in progress.
[  470.196132] mmc0: sdhci: ============ SDHCI REGISTER DUMP ===========
[  470.203313] mmc0: sdhci: Sys addr:  0x00000008 | Version:  0x00000002
[  470.210488] mmc0: sdhci: Blk size:  0x00007200 | Blk cnt:  0x00000008
[  470.217670] mmc0: sdhci: Argument:  0x00000000 | Trn mode: 0x00000033
[  470.224843] mmc0: sdhci: Present:   0x01f70000 | Host ctl: 0x00000017
[  470.232025] mmc0: sdhci: Power:     0x0000000f | Blk gap:  0x00000000
[  470.239208] mmc0: sdhci: Wake-up:   0x00000000 | Clock:    0x00000107
[  470.246389] mmc0: sdhci: Timeout:   0x0000000e | Int stat: 0x00108000
[  470.253572] mmc0: sdhci: Int enab:  0x03ff008b | Sig enab: 0x03ff008b
[  470.260754] mmc0: sdhci: ACmd stat: 0x00000000 | Slot int: 0x00000001
[  470.267927] mmc0: sdhci: Caps:      0x01f80080 | Caps_1:   0x00000007
[  470.275109] mmc0: sdhci: Cmd:       0x00000c1a | Max curr: 0x001f0f08
[  470.282291] mmc0: sdhci: Resp[0]:   0x00000000 | Resp[1]:  0xffffffff
[  470.289473] mmc0: sdhci: Resp[2]:   0x320f5913 | Resp[3]:  0x00000900
[  470.296646] mmc0: sdhci: Host ctl2: 0x0000008b
[  470.301598] mmc0: sdhci: ADMA Err:  0x00000000 | ADMA Ptr: 0xbe040200
[  470.308779] mmc0: sdhci: ============================================
[  470.315961] mmc0: Got data interrupt 0x00100000 even though no data operation was in progress.
[  470.325566] mmc0: sdhci: ============ SDHCI REGISTER DUMP ===========
[  470.332748] mmc0: sdhci: Sys addr:  0x00000008 | Version:  0x00000002
[  470.339922] mmc0: sdhci: Blk size:  0x00007200 | Blk cnt:  0x00000008
[  470.347104] mmc0: sdhci: Argument:  0x00000000 | Trn mode: 0x00000033
[  470.354278] mmc0: sdhci: Present:   0x01f70000 | Host ctl: 0x00000017
[  470.361460] mmc0: sdhci: Power:     0x0000000f | Blk gap:  0x00000000
[  470.368633] mmc0: sdhci: Wake-up:   0x00000000 | Clock:    0x00000107
[  470.375814] mmc0: sdhci: Timeout:   0x0000000e | Int stat: 0x00108000
[  470.382988] mmc0: sdhci: Int enab:  0x03ff008b | Sig enab: 0x03ff008b
[  470.390170] mmc0: sdhci: ACmd stat: 0x00000000 | Slot int: 0x00000001
[  470.397344] mmc0: sdhci: Caps:      0x01f80080 | Caps_1:   0x00000007
[  470.404526] mmc0: sdhci: Cmd:       0x00000c1a | Max curr: 0x001f0f08
[  470.411700] mmc0: sdhci: Resp[0]:   0x00000000 | Resp[1]:  0xffffffff
[  470.418882] mmc0: sdhci: Resp[2]:   0x320f5913 | Resp[3]:  0x00000900
[  470.426055] mmc0: sdhci: Host ctl2: 0x0000008b
[  470.431007] mmc0: sdhci: ADMA Err:  0x00000000 | ADMA Ptr: 0xbe040200
[  470.438180] mmc0: sdhci: ============================================
```

gtmills, 2020-05-05:
Hit this as well, on the same system, different build but based on a similiar downstream release

closed 2020-12-05 by stale[bot]

amboar, 2020-12-05:
This was resolved by the patches exposing the SDHCI clk-phase properties in the devicetree and measuring the appropriate corrections with a scope.