openbmc/openbmc #3660: palmetto power control

wws66, 2020-05-08:
Hello, I'm a novice who just got to know openbmc. Now I have some problems. I hope they can be solved.

About Palmetto code, we want to complete the power control of the host. We modified the GPIO pin in gpio_defs.json, and we can see that it is useful, but we didn't find where to set the delay time of GPIO pin pull down operation, so we chose to release the occupation of the current GPIO pin, and wrote a shell script under BMC After pin operation, it can sleep manually for a certain period of time. It is effective and achieves the expected results. But we still want to get the location of modifying the sleep time from the code. I hope it can help us. Thank you very much.

mdmillerii, 2020-05-08:
OpenPOWER reference systems use a desired state model power control.   That means that a GPIO is in one state when the system is requested to be powered on, and a different state when it is desired to be powered off.

Palmetto is an OpenPOWER platform.

There have been numerous discussions of power gpio toggles where a GPIO is set and released, including several within the last two months on the mailing list.

wws66, 2020-05-11:
Hello, I'm sorry to disturb you again. In this period of time, I looked up the problems related to the mail in the mail server, but it was more about the GPIO pin settings, and solved the problems. It may be less about the delay setting of the pin related to the reset function of the switch machine. We have the correct goal about the pin, so I I haven't found the solution I want in the mail server. 
Can you give me a relevant solution or proposal here? Thank you very much for your help.

mdmillerii, 2020-05-11:
I'll ask that you search the archives at https://lists.ozlabs.org/pipermail/openbmc/ .

wws66, 2020-05-12:
Hello, I am looking for the answer at https://lists.ozlabs.org/pipermail/openbmc/ , but as mentioned above, I did not find the result I want.

mdmillerii, 2020-05-12:
I think its a different program than the one palmetto uses.  Did you read all power related messages?

feitiantutu, 2020-11-10:
> I think its a different program than the one palmetto uses. Did you read all power related messages?

I want to configure the power button for my machine now. My button is active low. Pull it down for three seconds and then pull it up to boot. How do I define the gpio trigger mode and the time to pull it down? I have configured it as initial in dts High and low effective trigger, do I need to configure gpio status in GPIO_def.JSON? Where is the pull-down time configured?

mdmillerii, 2020-11-11:
OpenPOWER systems including palmetto use a level for desired power state.  For a pulse to toggle power there are several x86 systems.  I don't know what the arm systems use.

closed 2021-07-08 by stale[bot]