openbmc/openbmc #3661: Addition of python3 (3.2.8) on swift - python fails with "Could not find platform independent libraries")

aminaminf, 2020-05-11:
I am aware that python support has been removed from openbmc images, but we have a requirement to include python3 to our platform BMC which has 128 MB flash (we don't have flash footprint limitation).   

I tried adding python3 by adding it as ```OBMC_IMAGE_EXTRA_INSTALL_append =" python3" ```.  
Booting the final image, observed the error described in **Unexpected behaviour you saw**.  

To be sure that it is not dependent to our platform,  add python3 to  "swift" as described below and boot it over qemu. The same error observed. 

**Unexpected behavior you saw**
```
swift login: root
Password:
root@swift:~# python
Could not find platform independent libraries <prefix>
Consider setting $PYTHONHOME to <prefix>[:<exec_prefix>]
Python path configuration:
  PYTHONHOME = (not set)
  PYTHONPATH = (not set)
  program name = 'python'
  isolated = 0
  environment = 1
  user site = 1
  import site = 1
  sys._base_executable = '/usr/bin/python'
  sys.base_prefix = '/usr'
  sys.base_exec_prefix = '/usr'
  sys.executable = '/usr/bin/python'
  sys.prefix = '/usr'
  sys.exec_prefix = '/usr'
  sys.path = [
    '/usr/lib/python38.zip',
    '/usr/lib/python3.8',
    '/usr/lib/python3.8/lib-dynload',
  ]
Fatal Python error: init_fs_encoding: failed to get the Python codec of the filesystem encoding
Python runtime state: core initialized
LookupError: no codec search functions registered: can't find encoding

Current thread 0x76f18010 (most recent call first):
<no Python frame>
```

**Expected behavior**
Being able to get python prompt and run simple python apps.

**OpenBMC Information:**
```
root@swift:~# cat /etc/os-release
ID=openbmc-witherspoon
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="2.8.0-dev"
VERSION_ID=2.8.0-dev-1808-g18f1293f3-dirty
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) 2.8.0-dev"
BUILD_ID="2.8.0-dev"
OPENBMC_TARGET_MACHINE="swift"

root@swift:~# uname -a
Linux swift 5.4.39-30079d6 #1 Mon May 11 19:31:49 UTC 2020 armv6l GNU/Linux
```
Added python3 to build:
```
git diff ../meta-ibm/
diff --git a/meta-ibm/recipes-phosphor/images/obmc-phosphor-image.bbappend b/meta-ibm/recipes-phosphor/images/obmc-phosphor-image.bbappend
index 562679222..0149540c3 100644
--- a/meta-ibm/recipes-phosphor/images/obmc-phosphor-image.bbappend
+++ b/meta-ibm/recipes-phosphor/images/obmc-phosphor-image.bbappend
@@ -1,3 +1,4 @@
 OBMC_IMAGE_EXTRA_INSTALL_append_ibm-ac-server = " mboxd max31785-msl phosphor-msl-verify liberation-fonts uart-render-controller first-boot-set-mac first-boot-set-hostname"
 OBMC_IMAGE_EXTRA_INSTALL_append_rainier = " mboxd"
 OBMC_IMAGE_EXTRA_INSTALL_append_mihawk = " mboxd liberation-fonts uart-render-controller first-boot-set-mac"
+OBMC_IMAGE_EXTRA_INSTALL_append_swift = " python3"
```

**Additional context**
Add any other context about the problem here.

aminaminf, 2020-05-11:
This is how the lib folder looks like (seems missing  .py(c) files)
```
root@swift:/# ls /usr/lib/python3.8/
LICENSE.txt                                 json
__pycache__                                 lib-dynload
_sysconfigdata__linux_arm-linux-gnueabi.py  lib2to3
asyncio                                     logging
collections                                 multiprocessing
concurrent                                  pydoc_data
config-3.8-arm-linux-gnueabi                site-packages
ctypes                                      sqlite3
curses                                      tkinter
dbm                                         turtledemo
distutils                                   unittest
email                                       urllib
encodings                                   venv
html                                        wsgiref
http                                        xml
idlelib                                     xmlrpc
importlib
```

geissonator, 2020-05-11:
We a bit of hackery in https://github.com/openbmc/meta-phosphor/blob/master/recipes-devtools/python/python3_%25.bbappend to ensure python can still fit on our 32MB systems. Maybe try removing the part that removes all of the .py files and see if it helps?

mdmillerii, 2020-05-11:
Also in python3 the pyc are under the `__pycache__` directory, not adjacent to the `.py` file.

aminaminf, 2020-05-11:
@geissonator thanks a lot. that helped!

```
root@swift:~# python
Python 3.8.2 (default, May 11 2020, 21:33:05)
[GCC 9.3.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import sys
>>> import os
>>> quit()
root@swift:~#
```
So i will close this for sure.
 
@geissonator @mdmillerii 
Just for curiosity and better understanding due to  my poor python knowledge : with the mentioned bbappend, how could a limited footprint system could run a python?

geissonator, 2020-05-12:
Your test seems to indicate that our removal of the .py files breaks python when installed into the image. The only testing we did, since we don't have any systems actually installing it, was that images built with the change so we had not tested this.

We better reopen this to track this issue.

closed 2020-12-19 by stale[bot]