openbmc/openbmc #3663: Failed to create a virtual USB media with AST2500 USB port A.

jns-git, 2020-05-14:
**What I want:**
In our schematic, there is a USB signal connect HOST CPU to BMC(AST2500 SOC) USB port A.
I want to use this USB channel to send file image from host side to BMC.
So I set the host CPU as a USB master. I set the USB port A of AST2500 BMC chip to a slave, (Virtual HUB).

**What I do:**
In the kernel DVT, I enable VHUB.
In the defconfig, I enable USB GADGET config.
meta-openbmc-bsp/meta-aspeed/recipes-kernel/linux/ast2500/defconfig
+CONFIG_USB_GADGET=y
+CONFIG_USB_ASPEED_VHUB=y
+CONFIG_USB_CONFIGFS=y
+CONFIG_USB_CONFIGFS_MASS_STORAGE=y
+CONFIG_USB_CONFIGFS_F_HID=y
+CONFIG_USB_MASS_STORAGE=y

**Now, I can see a usb hub under HOST CPU side. But these is no USB end point device**
root@localhost:/# lsusb
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 004: ID 04d8:00de Microchip Technology, Inc.
Bus 001 Device 003: ID 04d8:00dd Microchip Technology, Inc.
**Bus 001 Device 007: ID 1d6b:0107 Linux Foundation**
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
root@localhost:/# lsusb -vt
/:  Bus 02.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/4p, 5000M
/:  Bus 01.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/4p, 480M
    **|__ Port 2: Dev 7, If 0, Class=Hub, Driver=hub/5p, 480M**
    |__ Port 3: Dev 3, If 0, Class=Communications, Driver=cdc_acm, 12M
    |__ Port 3: Dev 3, If 1, Class=CDC Data, Driver=cdc_acm, 12M
    |__ Port 3: Dev 3, If 2, Class=Human Interface Device, Driver=i2c-mcp2221, M
    |__ Port 4: Dev 4, If 0, Class=Human Interface Device, Driver=mcp2210, 12M
root@localhost:/#

**Now I follow below steps to create a USB device, But I meet problem. When run "mkdir -p functions/rndis.usb0"** 
#!/bin/bash

cd /sys/kernel/config/usb_gadget

if [ ! -f "g1" ]; then
    mkdir g1
    cd g1

    echo 0x1d6b > idVendor  # Linux foundation
    echo 0x0104 > idProduct # Multifunction composite gadget
    mkdir -p strings/0x409
    echo "Linux" > strings/0x409/manufacturer
    echo "Ethernet/RNDIS gadget" > strings/0x409/product
    
    mkdir -p configs/c.1
    echo 100 > configs/c.1/MaxPower
    mkdir -p configs/c.1/strings/0x409
    echo "RNDIS" > configs/c.1/strings/0x409/configuration
    
    **mkdir -p functions/rndis.usb0** 
    
    ln -s functions/rndis.usb0 configs/c.1
    
    echo f0839000.udc > UDC

fi
exit 0

**Error message: mkdir: can't create directory 'functions/rndis.usb0': Function not implemented**

**How can I go ahead.**

**OpenBMC Information:**
root@bmc:/tmp# cat /etc/os-release
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="00.10"
VERSION_ID="00.10.00"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) 00.10"
BUILD_ID="0001"
root@bmc:/tmp#

root@bmc:/tmp# uname -a
Linux  4.17.11-db64579331df4ff9d9c2706601fdb41978192a09 #1 Thu May 14 03:41:x
root@bmc:/tmp#

Build Configuration:
BB_VERSION           = "1.38.0"
BUILD_SYS            = "x86_64-linux"
NATIVELSBSTRING      = "ubuntu-18.04"
TARGET_SYS           = "arm-openbmc-linux-gnueabi"
MACHINE              = "bmc"
DISTRO               = "openbmc-phosphor"
DISTRO_VERSION       = "0.1.0"
TUNE_FEATURES        = "arm armv6 thumb arm1176jzs"
TARGET_FPU           = "soft"

I know the kernel version I used is old, I don't want to update my Linux kernel.

jns-git, 2020-05-14:
Another point is that, In our schematic design, there is no storage device connected to BMC chip.
No emmc, no SSD, no SD card, no any storage device. 
Does the USB gadget support use DRAM for virtual media function?

mdmillerii, 2020-05-14:
We normally ask that questions be directed to our mailing list for the widest visibility, but there are a few of us that watch these issues and will offer quick advice.

Please use ` ``` block quotes ``` ` when posting listings and code consistently to avoid magic characters in markdown (and the `**` bold markup does not work in blockquotes).  You can toggle between the `Write` and `Preview` tabs to see the formatting of your issues.

Looking at [v4.17..dev-4.17 - click on **Files Changed** tab](https://github.com/torvalds/linux/compare/v4.17...openbmc:dev-4.17#diff-f5f8f6d7d0b5794fcf8d8d6fd0be24edR193) I see aspeed_g4_defconfig includes `CONFIG_USB_CONFIGFS_RNDIS` but your fragment does not.  Considering your error relates to rndis it may be related.

I also see in your script above you are testing for a file then making a directory. 
```
if [ ! -f "g1" ]; then
mkdir g1
cd g1
```

 If you want to avoid making a directory on top of a previous one, then use `-d` or `-e` argument to test.

jns-git, 2020-05-15:
Yes, I edit my first comments to fix the format problem.
I will try to set `CONFIG_USB_CONFIGFS_RNDIS` , then let you know result.

jns-git, 2020-05-15:
After set `CONFIG_USB_CONFIGFS_RNDIS`， I got pass result.

I also modify the shell script ,due to the UDC names.
modify `echo f0839000.udc > UDC`  to `echo 1e6a0000.usb-vhub:p1 > UDC`

```
root@bmc:~# ls /sys/class/udc/
1e6a0000.usb-vhub:p1  1e6a0000.usb-vhub:p3  1e6a0000.usb-vhub:p5
1e6a0000.usb-vhub:p2  1e6a0000.usb-vhub:p4
```

```
#!/bin/bash

cd /sys/kernel/config/usb_gadget

if [ ! -f "g1" ]; then
    mkdir g1
    cd g1

    echo 0x1d6b > idVendor  # Linux foundation
    echo 0x0104 > idProduct # Multifunction composite gadget
    mkdir -p strings/0x409
    echo "Linux" > strings/0x409/manufacturer
    echo "Ethernet/RNDIS gadget" > strings/0x409/product
    
    mkdir -p configs/c.1
    echo 100 > configs/c.1/MaxPower
    mkdir -p configs/c.1/strings/0x409
    echo "RNDIS" > configs/c.1/strings/0x409/configuration
    
    mkdir -p functions/rndis.usb0 
    
    ln -s functions/rndis.usb0 configs/c.1
    
    echo 1e6a0000.usb-vhub:p1 > UDC

fi
exit 0

```

**After run the shell script, I can see a USB to Ethernet device appear in BMC side.**

```
root@bmc:~# ./1.sh
[  579.479777] using random self ethernet address
[  579.484262] using random host ethernet address
[  579.505690] usb0: HOST MAC b6:8d:dc:f4:79:36
[  579.510214] usb0: MAC 42:68:0e:9e:6b:ac
root@bmc:~# [  580.033575] configfs-gadget gadget: high-speed config #1: c
```

```
root@bmc:~# ip addr show
1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue qlen 1000
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
    inet 127.0.0.1/8 scope host lo
       valid_lft forever preferred_lft forever
    inet6 ::1/128 scope host
       valid_lft forever preferred_lft forever
2: eth0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc pfifo_fast qlen 1000
    link/ether ea:66:55:3b:a4:b9 brd ff:ff:ff:ff:ff:ff
    inet 169.254.59.209/16 brd 169.254.255.255 scope link eth0
       valid_lft forever preferred_lft forever
    inet 10.116.108.59/24 brd 10.116.108.255 scope global dynamic eth0
       valid_lft 28262sec preferred_lft 28262sec
    inet6 fe80::e866:55ff:fe3b:a4b9/64 scope link
       valid_lft forever preferred_lft forever
3: eth1: <NO-CARRIER,BROADCAST,MULTICAST,UP> mtu 1500 qdisc pfifo_fast qlen 1000
[... 75 lines cut ...]
          RX packets:12924161 errors:0 dropped:0 overruns:0 frame:0
          TX packets:12924161 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:748104786 (713.4 MiB)  TX bytes:748104786 (713.4 MiB)

ma1       Link encap:Ethernet  HWaddr c8:de:c9:60:55:00
          inet addr:10.116.108.51  Bcast:10.116.108.255  Mask:255.255.255.0
          inet6 addr: fe80::cade:c9ff:fe60:5500/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:117624 errors:0 dropped:0 overruns:0 frame:0
          TX packets:2662 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:25902799 (24.7 MiB)  TX bytes:152635 (149.0 KiB)
          Memory:dcd00000-dcd7ffff

sit0      Link encap:IPv6-in-IPv4
          NOARP  MTU:1480  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)

root@localhost:/#

```

jns-git, 2020-05-22:
I give up to use USB to Ethernet.
Now I use usb gadget function with mass_storage.
Run below shell script to create a virtual USB media.
`./xx.sh start 0`
```
#!/bin/sh

if [ $# -ne 2 ]
then
    echo "usage: $0 <start|stop> <config>" >&2
    exit 1
fi

action=$1
config=$2

gadget_name=mass-storage
gadget_dir=/sys/kernel/config/usb_gadget/$gadget_name

case "$config" in
0)
    usb_device=/dev/shm/usb.img
    ;;
*)
    echo "invalid config $config" >&2
    exit 1
    ;;
esac

set -ex

case "$action" in
start)
    tr '\000' '\377' < /dev/zero | dd of=/dev/shm/usb.img bs=1024 count=50k
    mkdir -p $gadget_dir
    (
    cd $gadget_dir
    # http://www.linux-usb.org/usb.ids
    #    |-> 1d6b  Linux Foundation
    #          |-> 0104  Multifunction Composite Gadget
    echo "0x1d6b" > idVendor
    echo "0x0104" > idProduct
    mkdir -p strings/0x409
    echo "OpenBMC" > strings/0x409/manufacturer
    echo "Virtual Media Device" > strings/0x409/product
    mkdir -p configs/c.1/strings/0x409
    echo "config 1" > configs/c.1/strings/0x409/configuration
    mkdir -p functions/mass_storage.usb0
    ln -s functions/mass_storage.usb0 configs/c.1
    echo 1 > functions/mass_storage.usb0/lun.0/removable
    echo 0 > functions/mass_storage.usb0/lun.0/ro
    echo 0 > functions/mass_storage.usb0/lun.0/cdrom
    echo $usb_device > functions/mass_storage.usb0/lun.0/file
    echo "1e6a0000.usb-vhub:p1" > UDC
    )
    ;;
stop)
    (
    cd $gadget_dir
    rm configs/c.1/mass_storage.usb0
    rmdir functions/mass_storage.usb0
    rmdir configs/c.1/strings/0x409
    rmdir configs/c.1
    rmdir strings/0x409
	rm /dev/shm/usb.img
    )
    rmdir $gadget_dir
    ;;
*)
    echo "invalid action $action" >&2
    exit 1
esac

exit 0
```

Go to host CPU, there is USB popup.

```
root@localhost:~# lsusb -vt

/:  Bus 01.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/4p, 480M
    |__ Port 2: Dev 6, If 0, Class=Hub, Driver=hub/5p, 480M
        |__ Port 1: Dev 7, If 0, Class=Mass Storage, Driver=usb-storage, 480M

root@localhost:~#
root@localhost:~# fdisk -l
Disk /dev/sdb: 50 MiB, 52428800 bytes, 102400 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 512 bytes
I/O size (minimum/optimal): 512 bytes / 512 bytes
root@localhost:~#
```

Then, I format the /dev/sdb by using `mkfs.ext4 /dev/sdb` ,copy a file to the usb device.
Use the cmd `md5sum /dev/sdb` to get the md5 checksum of the USB device.

```
root@localhost:/mnt/onl/data# mkfs.ext4 /dev/sdb
mke2fs 1.42.12 (29-Aug-2014)
Creating filesystem with 51200 1k blocks and 12824 inodes
Filesystem UUID: 4b9ef3d3-c864-435a-9d7d-c0b5642493cd
Superblock backups stored on blocks:
        8193, 24577, 40961

Allocating group tables: done
Writing inode tables: done
Creating journal (4096 blocks): done
Writing superblocks and filesystem accounting information: done

root@localhost:/mnt/onl/data#
root@localhost:/mnt/onl/data# mkdir tmp
root@localhost:/mnt/onl/data# mount /dev/sdb tmp/
root@localhost:/mnt/onl/data# cd tmp/
root@localhost:/mnt/onl/data/tmp# cp ../stt.bin .
root@localhost:/mnt/onl/data/tmp# ls
lost+found  kkk.bin
root@localhost:/mnt/onl/data/tmp# cd ..
root@localhost:/mnt/onl/data# sync
root@localhost:/mnt/onl/data# umount tmp/

root@localhost:/mnt/onl/data/moun# md5sum /dev/sdb
e0367fb3d5ab7af9e609f987fcf52ff8  /dev/sdb
root@localhost:/mnt/onl/data/moun#
```

Go back to BMC side, I use `md5sum usb.img` get the same md5 checksum value. But when I try to mount it with ext4 format, I meet error. I also tried other formal likes ·vfat· ·msdos· but the same failed result got. How can go on? 
```
root@bmc:/dev/shm# md5sum usb.img
e0367fb3d5ab7af9e609f987fcf52ff8  usb.img
root@bmc:/dev/shm#

root@bmc:/dev/shm# mount -t ext4 usb.img tmp/
mount: /dev/shm/tmp: unknown filesystem type 'ext4'.
root@bmc:/dev/shm#
```

mine260309, 2020-05-25:
I tried to use ecm to create the virtual network, and it works fine.
Here is some changes needed:
1. kernel config:
```
+CONFIG_USB_U_ETHER=y
+CONFIG_USB_F_ECM=y
+CONFIG_USB_CONFIGFS_ECM=y
```
2. Create `ecm.xxx` dir in the functions:
```
mkdir functions/ecm.$name
```

After the ecm device created, the usb ethernet device is shown in both sides.

mdmillerii, 2020-06-03:
> Go back to BMC side, I use `md5sum usb.img` get the same md5 checksum value, it means the data transfer successful from host to BMC . But when I try to mount it with ext4 format, I meet error. I also tried other formal likes `vfat` `msdos` but the same failed result got. How can go on?

@jns-git check `/proc/filesystems` to see if it contains `ext4` (or `vfat` if you prefer), we don't use the file system for normal MTD flash storage (only eMMC would have it by default).  You'll likely need to add a kernel config fragment in your Yocto kernel build.

jns-git, 2020-06-03:
@mdmillerii  Yes, we add vfat supported in kernel build, now we can use the USB gadget successfully.
Thanks for your help.
@mine260309  we can't see USB to Ethernet device, due to host linux not support USBNET function, host linux need to enable USBNET related config. Thanks for your help.

closed 2020-06-03 by jns-git

mdmillerii, 2020-06-03:
@jns-git please review this series adding usb-ctrl from the Intel repository to phosphor-misc and adding use cases.  Consider adding a patch with your use case.

https://gerrit.openbmc-project.xyz/c/openbmc/phosphor-misc/+/32877
https://gerrit.openbmc-project.xyz/c/openbmc/phosphor-misc/+/33137
https://gerrit.openbmc-project.xyz/c/openbmc/phosphor-misc/+/32878

chineweff, 2020-08-17:
@jns-git  Hello

chineweff, 2020-08-17:
@jns-git 
Hello,
I encountered some problem when I try to use usb gadget function with mass_storage I ran the same script with you and it seemed that it didn't work.

chineweff, 2020-08-17:
@jns-git
Hello,
I encountered some problems when I try to use USB gadget function with mass_storage.
My host was windows and I ran the same script with you and it seemed that it didn't work.

I printed out vhub status with:
cat /sys/class/udc/1e6a0000.usb-vhub\:p1/state
It showed "not attached"

My dmesg only showed these two lines:
[ 8629.307286] Mass Storage Function, version: 2009/09/11
[ 8629.312465] LUN: removable file: (no medium)

What is your kernel config setting?
Is there any dependency that I have to notice?

Thanks

jns-git, 2020-08-17:
@chineweff  
root@bmc:~# ls /sys/class/udc/1e6a0000.usb-vhub\:p1/
a_alt_hnp_support  b_hnp_enable       device             is_a_peripheral    is_selfpowered     soft_connect       state              uevent
a_hnp_support      current_speed      function           is_otg             maximum_speed      srp                subsystem
root@bmc:~#

kernel config setting:
+CONFIG_USB_GADGET=y
+CONFIG_USB_ASPEED_VHUB=y
+CONFIG_USB_CONFIGFS=y
+CONFIG_USB_CONFIGFS_MASS_STORAGE=y
+CONFIG_USB_CONFIGFS_F_HID=y
+CONFIG_USB_MASS_STORAGE=y
+CONFIG_USB_CONFIGFS_ACM=y
+CONFIG_USB_CONFIGFS_OBEX=y
+CONFIG_USB_CONFIGFS_NCM=y
+CONFIG_USB_U_ETHER=y
+CONFIG_USB_F_ECM=y
+CONFIG_USB_CONFIGFS_ECM=y
+CONFIG_USB_CONFIGFS_ECM_SUBSET=y
+CONFIG_USB_CONFIGFS_RNDIS=y
+CONFIG_USB_CONFIGFS_EEM=y
+CONFIG_USB_CONFIGFS_F_LB_SS=y
+CONFIG_USB_CONFIGFS_F_FS=y
+CONFIG_FAT_FS=y
+CONFIG_VFAT_FS=y
+CONFIG_NLS_CODEPAGE_437=y
+CONFIG_NLS_ISO8859_1=y

In the DTS, should also enable vhub.
&vhub {
    status = "okay";
};

chineweff, 2020-08-17:
@jns-git 
Thanks for replying.
I used your setting, but it still showed "not attatched" now.

I open USB gadget debug mode my dmesg:
[  172.579745] Mass Storage Function, version: 2009/09/11
[  172.584927] LUN: removable file: (no medium)
[  172.619829] mass_storage.usb0/lun.0: read-only status set to 0
[  172.623152] mass_storage.usb0/lun.0: open backing file: /dev/shm/usb.img
[  172.624847] udc 1e6a0000.usb-vhub:p1: registering UDC driver [mass-storage]
[  172.624952] configfs-gadget gadget: adding 'Mass Storage Function'/a7cd8302 to config 'c'/71fee232
[  172.625252] configfs-gadget gadget: I/O thread pid: 585
[  172.625303] aspeed_vhub 1e6a0000.usb-vhub: port1 Match EP type 2
[  172.625325] aspeed_vhub 1e6a0000.usb-vhub: port1 Allocating gen EP 0 for addr 1
[  172.625649] aspeed_vhub 1e6a0000.usb-vhub: port1 Allocated epn#0 for port EP1
[  172.625684] aspeed_vhub 1e6a0000.usb-vhub: port1 Match EP type 2
[  172.625708] aspeed_vhub 1e6a0000.usb-vhub: port1 Allocating gen EP 1 for addr 2
[  172.625921] aspeed_vhub 1e6a0000.usb-vhub: port1 Allocated epn#1 for port EP2
[  172.625963] aspeed_vhub 1e6a0000.usb-vhub: port1 start
[  172.625984] aspeed_vhub 1e6a0000.usb-vhub: port1 pullup(1)
[  172.626005] aspeed_vhub 1e6a0000.usb-vhub: port1 port 1 status 0000 -> 0001 (C=1)

Do you have any ideas?

jns-git, 2020-08-17:
@chineweff 

root@bmc:#
root@bmc:# usb-gadget.sh stop 0
root@bmc:# cat /sys/class/udc/1e6a0000.usb-vhub\:p1/state
not attached
root@bmc:# usb-gadget.sh start 0
root@bmc:# cat /sys/class/udc/1e6a0000.usb-vhub\:p1/state
configured

I stop the usb-gadget script, then the state show "not attached"
I start the usb-gadget script, then the state show "configured"

the dmesg show as:
[  241.246379] Mass Storage Function, version: 2009/09/11
[  241.246405] LUN: removable file: (no medium)
[  241.728533] configfs-gadget gadget: high-speed config #1: c

You can follow the script step by step, run it manual to see what steps you meet error.

chineweff, 2020-08-28:
I found that my libcomposite probe failed

Pravee207, 2023-01-02:
We are trying to reproduce this issue on Qemu. Does it require hardware to reproduce this issue or can we perform it on Qemu?. Please do advise.