openbmc/openbmc #3665: TypeError: Cannot convert name to ASCII on really long paths

gtmills, 2020-05-28:
Build team was seeing this. 
Steps to recreate: 

`mkdir -p /tmp/longpath/1234567890/abcdefghij/0987654321/jihgfedcba/6060606060/7070707070/8080808080/909090909090/1001001001/`

`cd /tmp/longpath/1234567890/abcdefghij/0987654321/jihgfedcba/6060606060/7070707070/8080808080/909090909090/1001001001/`

` TEMPLATECONF=meta-ibm/meta-witherspoon/conf`

`. /esw/san5/`whoami`/obmc/openbmc/openbmc-env`

`nice -20 bitbake obmc-phosphor-image`
```
phosphor-webui-1.0+gitAUTOINC+4a16a026b6-r1 do_compile: Execution of '/tmp/longpath/1234567890/abcdefghij/0987654321/jihgfedcba/6060606060/7070707070/8080808080/909090909090/1001001001/build/tmp/work/all-openbmc-linux/phosphor-webui/1.0+gitAUTOINC+4a16a026b6-r1/temp/run.do_compile.4185' failed with exit code 7:
TypeError: Cannot convert name to ASCII
    at Url.parse (url.js:386:23)
    at Object.urlParse [as parse] (url.js:150:13)
    at Object.validateUrl [as validate] (/tmp/longpath/1234567890/abcdefghij/0987654321/jihgfedcba/6060606060/7070707070/8080808080/909090909090/1001001001/build/tmp/work/all-openbmc-linux/phosphor-webui/1.0+gitAUTOINC+4a16a026b6-r1/recipe-sysroot-native/usr/lib/node_modules/npm/node_modules/nopt/lib/nopt.js:172:13)
    at validate (/tmp/longpath/1234567890/abcdefghij/0987654321/jihgfedcba/6060606060/7070707070/8080808080/909090909090/1001001001/build/tmp/work/all-openbmc-linux/phosphor-webui/1.0+gitAUTOINC+4a16a026b6-r1/recipe-sysroot-native/usr/lib/node_modules/npm/node_modules/nopt/lib/nopt.js:221:24)
    at validate (/tmp/longpath/1234567890/abcdefghij/0987654321/jihgfedcba/6060606060/7070707070/8080808080/909090909090/1001001001/build/tmp/work/all-openbmc-linux/phosphor-webui/1.0+gitAUTOINC+4a16a026b6-r1/recipe-sysroot-native/usr/lib/node_modules/npm/node_modules/nopt/lib/nopt.js:187:11)
    at /tmp/longpath/1234567890/abcdefghij/0987654321/jihgfedcba/6060606060/7070707070/8080808080/909090909090/1001001001/build/tmp/work/all-openbmc-linux/phosphor-webui/1.0+gitAUTOINC+4a16a026b6-r1/recipe-sysroot-native/usr/lib/node_modules/npm/node_modules/nopt/lib/nopt.js:104:12
    at Array.map (<anonymous>)
    at /tmp/longpath/1234567890/abcdefghij/0987654321/jihgfedcba/6060606060/7070707070/8080808080/909090909090/1001001001/build/tmp/work/all-openbmc-linux/phosphor-webui/1.0+gitAUTOINC+4a16a026b6-r1/recipe-sysroot-native/usr/lib/node_modules/npm/node_modules/nopt/lib/nopt.js:70:15
    at Array.forEach (<anonymous>)
    at Function.clean (/tmp/longpath/1234567890/abcdefghij/0987654321/jihgfedcba/6060606060/7070707070/8080808080/909090909090/1001001001/build/tmp/work/all-openbmc-linux/phosphor-webui/1.0+gitAUTOINC+4a16a026b6-r1/recipe-sysroot-native/usr/lib/node_modules/npm/node_modules/nopt/lib/nopt.js:58:21)
/tmp/longpath/1234567890/abcdefghij/0987654321/jihgfedcba/6060606060/7070707070/8080808080/909090909090/1001001001/build/tmp/work/all-openbmc-linux/phosphor-webui/1.0+gitAUTOINC+4a16a026b6-r1/recipe-sysroot-native/usr/lib/node_modules/npm/lib/npm.js:59
      throw new Error('npm.load() required')
```

First saw this at 
https://github.com/openbmc/openbmc/commit/062316f13d6e9d71a14724e293241b6258c63876

Last good build: 
https://github.com/openbmc/openbmc/commit/9f17c411648a900222911d48cd4d36777acbb1aa

mdmillerii, 2020-05-30:
searching `TypeError: Cannot convert name to ASCII` yields errors about ICU not handling a valid path among others.

njpau, 2022-08-12:
We're seeing this same issue when trying to build chromium-x11 on our jenkins agents, but the builds work fine when building locally on our machines. Both builds happen in identical docker environments.

geissonator, 2023-04-13:
We modified our docker build scripts to utilize /tmp for building within the docker container (instead of the lengthy home directory of the jenkins job). This "fixed" it for us. The root issue of npm not handling long build directory paths has not been addressed to my knowledge.

williamspatrick, 2023-09-25:
Has this been reported upstream to npm?  I don't think we're the right place to hold an npm issue.

closed 2023-09-25 by williamspatrick