openbmc/openbmc #3666: ipmi: Unable to establish IPMI v2 / RMCP+ session

yanlinding, 2020-06-02:
Using master [https://github.com/openbmc/openbmc/commit/3b76fc97c12c5ba670e0f2802b777589637daf05](url)
on Romulus, I'm Unable to establish IPMI v2 / RMCP+ session

ipmitool -I lanplus -H 192.168.3.31 -P 0penBmc mc info
Error in open session response message : invalid authentication algorithm

Error: Unable to establish IPMI v2 / RMCP+ session
$ ipmitool -I lanplus -H 192.168.3.31 -U root -P 0penBmc mc info
Error in open session response message : invalid authentication algorithm

Error: Unable to establish IPMI v2 / RMCP+ session
$ ipmitool -I lanplus -H 192.168.3.31 -U root -P 0penBmc mc info -vvv

>> Sending IPMI command payload
>>    netfn   : 0x06
>>    command : 0x38
>>    data    : 0x8e 0x04 

BUILDING A v1.5 COMMAND
>> IPMI Request Session Header
>>   Authtype   : NONE
>>   Sequence   : 0x00000000
>>   Session ID : 0x00000000
>> IPMI Request Message Header
>>   Rs Addr    : 20
>>   NetFn      : 06
>>   Rs LUN     : 0
>>   Rq Addr    : 81
>>   Rq Seq     : 00
>>   Rq Lun     : 0
>>   Command    : 38
<< IPMI Response Session Header
<<   Authtype                : NONE
<<   Payload type            : IPMI (0)
<<   Session ID              : 0x00000000
<<   Sequence                : 0x00000000
<<   IPMI Msg/Payload Length : 16
<< IPMI Response Message Header
<<   Rq Addr    : 81
<<   NetFn      : 07
<<   Rq LUN     : 0
<<   Rs Addr    : 20
<<   Rq Seq     : 00
<<   Rs Lun     : 0
<<   Command    : 38
<<   Compl Code : 0x00
>> SENDING AN OPEN SESSION REQUEST

<<OPEN SESSION RESPONSE
<<  Message tag                        : 0x00
<<  RMCP+ status                       : invalid authentication algorithm
<<  Maximum privilege level            : Unknown (0x00)
<<  Console Session ID                 : 0xa0a2a3a4
Error in open session response message : invalid authentication algorithm

Error: Unable to establish IPMI v2 / RMCP+ session 

on board use netstat tool list info as following:
root@romulus:/etc# netstat -nlu
Active Internet connections (only servers)
.......
udp        0      0 :::623                  :::*
udp        0      0 :::427                  :::*

/etc/group info:
root@romulus:/etc# cat /etc/group
root:x:0:
daemon:x:1:
bin:x:2:
sys:x:3:
adm:x:4:
tty:x:5:
disk:x:6:
lp:x:7:
mail:x:8:
news:x:9:
uucp:x:10:
man:x:12:
proxy:x:13:
kmem:x:15:
input:x:19:
dialout:x:20:
fax:x:21:
voice:x:22:
cdrom:x:24:
floppy:x:25:
tape:x:26:
sudo:x:27:
audio:x:29:
dip:x:30:
www-data:x:33:
backup:x:34:
operator:x:37:
list:x:38:
irc:x:39:
src:x:40:
gnats:x:41:
shadow:x:42:
utmp:x:43:
video:x:44:
sasl:x:45:
plugdev:x:46:
kvm:x:47:
staff:x:50:
games:x:60:
shutdown:x:70:
users:x:100:
bmcweb:x:991:
systemd-bus-proxy:x:992:
systemd-timesync:x:993:
systemd-resolve:x:994:
systemd-network:x:995:
systemd-coredump:x:996:
systemd-journal:x:997:
avahi:x:998:
messagebus:x:999:
priv-admin:x:1000:root
priv-operator:x:1001:
priv-user:x:1002:
web:x:1003:root
redfish:x:1004:root
ipmi:x:1005:root
nogroup:x:65534:
wheel:x:990:
render:x:989:
nobody:x:988:

root@romulus:/etc# uname -a
Linux romulus 5.4.39-30079d6 #1 Sun May 31 11:05:16 UTC 2020 armv6l GNU/Linux

root@romulus:/etc# cat os-release
ID=openbmc-romulus
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="2.9.0-dev"
VERSION_ID=2.9.0-dev-88-g3b76fc9-dirty
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) 2.9.0-dev"
BUILD_ID="2.9.0-dev"
OPENBMC_TARGET_MACHINE="romulus"

geissonator, 2020-06-02:
I wonder if this is related to a security change where only certain cipher suites are allowed now? Try adding a "-C 17" to your ipmitool command line.

yanlinding, 2020-06-03:
Thanks for your suggestions, After added "-C 17", ipmitool can work well.
ipmitool -I lanplus -H 192.168.3.31 -C 17 -U root -P 0penBmc mc info
Device ID                 : 0
Device Revision           : 0
Firmware Revision         : 0.00
IPMI Version              : 2.0
Manufacturer ID           : 0
Manufacturer Name         : Unknown
Product ID                : 0 (0x0000)
Product Name              : Unknown (0x0)
Device Available          : yes
Provides Device SDRs      : yes
Additional Device Support :
    Sensor Device
    SEL Device
    FRU Inventory Device
    Chassis Device
Aux Firmware Rev Info     : 
    0x00
    0x00
    0x00
    0x00

closed 2021-01-09 by stale[bot]