openbmc/openbmc #3669: Issue on Virtual media redirection for HDD image.

edineshkumar, 2020-06-15:
**Note on Bugs**
HDD image can able to redirect, but can not able to mount in host side. Getting error like "cant't read superblock".

_**Note:  CD-ROM (.iso or .nrg ) images working fine.**_

**Unexpected behavior you saw**
Can not able to mount redirected HDD image .

**Expected behavior**
It should mount in host without any error.

**To Reproduce**
Steps to reproduce the behavior:
1. Flash OpenBMC image on ast_2600 EVB .
2. Login web-ui.
3. Goto Server Control --> Virtual Media.
4. Select Choose file, and Upload HDD (.iso or .img) image.
5. Redirected successfully.
6. Host side also can able to see with lsblk command.
7. While mounting, can't succeed.

**Screenshots**
Redirected successfully.
[image]

Issue with mounting the same image.

[image]

 

**OpenBMC Information:**
 - uname -a:  Linux AMI6e9d62ae397a 5.4.11-INVALID-dirty-b8ad0c7 #1 SMP Thu Jun 4 10:50:48 UTC 2020 armv7l GNU/Linux

edineshkumar, 2020-06-15:
**Correction in 4th reproduce step** 

Choose only .img or .ima file. Do not choose .iso file. CD-ROM (.iso) working fine.

mdmillerii, 2020-06-17:
Suggestions to help you isolate your problem were made in the referenced issue.  Also you can try to copy the image to  the host with dd.

edineshkumar, 2020-07-10:
Thanks mdmillerii , 
HDD image I can able to mount as a loop device only in host side.

Is this possible  to mount same like **"iso"** image ?

mdmillerii, 2020-07-10:
Not sure what your question is asking. 

The bmc usually doesn't include the filesystem drivers for iso images or maybe even disk images but like most Linux kernels you can compile them in but I don't know what that would prove.   
Did you copy the image through the USB drive to the file to the host and compare  it?   On a Linux host I would try to use `dd` to read contents to get selected blocks on loopback file and the usb virt media.   If they differ try dd on the bmc to the media file or nbd device.

edineshkumar, 2020-07-10:
HDD image file's content can read from mounted loop device.

Thanks.

closed 2020-07-10 by edineshkumar