openbmc/openbmc #3671: intermittent image/etc/ipmi_pass': No such file or directory" compile failure

geissonator, 2020-06-29:
We've been seeing this for the last month or so intermittently in CI. It will hit intermittent when we bring in some new upstream code. Then once we have the sstate built up we wont see it again.

```
[555A[J[1;31mERROR[0m: [31mLogfile of failure stored in: /tmp/openbmc/work/arm1176jzs-openbmc-linux-gnueabi/pam-ipmi/1.0+gitAUTOINC+f3919c432f-r1/temp/log.do_install.33337[0m
Log data follows:
| DEBUG: Executing python function autotools_aclocals
| DEBUG: SITE files ['endian-little', 'bit-32', 'arm-common', 'arm-32', 'common-linux', 'common-glibc', 'arm-linux', 'arm-linux-gnueabi', 'common']
| DEBUG: Python function autotools_aclocals finished
| DEBUG: Executing python function extend_recipe_sysroot
| NOTE: Direct dependencies are ['/home/jenkins-slave/workspace/ci-meta/distro/ubuntu/label/docker-builder/target/tiogapass/openbmc/meta/recipes-devtools/gcc/gcc-cross_10.1.bb:do_populate_sysroot', 'virtual:native:/home/jenkins-slave/workspace/ci-meta/distro/ubuntu/label/docker-builder/target/tiogapass/openbmc/meta/recipes-devtools/autoconf/autoconf_2.69.bb:do_populate_sysroot', 'virtual:native:/home/jenkins-slave/workspace/ci-meta/distro/ubuntu/label/docker-builder/target/tiogapass/openbmc/meta/recipes-devtools/autoconf-archive/autoconf-archive_2019.01.06.bb:do_populate_sysroot', '/home/jenkins-slave/workspace/ci-meta/distro/ubuntu/label/docker-builder/target/tiogapass/openbmc/meta/recipes-devtools/gcc/gcc-runtime_10.1.bb:do_populate_sysroot', '/home/jenkins-slave/workspace/ci-meta/distro/ubuntu/label/docker-builder/target/tiogapass/openbmc/meta/recipes-connectivity/openssl/openssl_1.1.1g.bb:do_populate_sysroot', '/home/jenkins-slave/workspace/ci-meta/distro/ubuntu/label/docker-builder/target/tiogapass/openbmc/meta/recipes-devtools/libtool/libtool-native_2.4.6.bb:do_populate_sysroot', '/home/jenkins-slave/workspace/ci-meta/distro/ubuntu/label/docker-builder/target/tiogapass/openbmc/meta/recipes-devtools/quilt/quilt-native_0.66.bb:do_populate_sysroot', 'virtual:native:/home/jenkins-slave/workspace/ci-meta/distro/ubuntu/label/docker-builder/target/tiogapass/openbmc/meta/recipes-devtools/gnu-config/gnu-config_git.bb:do_populate_sysroot', 'virtual:native:/home/jenkins-slave/workspace/ci-meta/distro/ubuntu/label/docker-builder/target/tiogapass/openbmc/meta/recipes-devtools/pseudo/pseudo_git.bb:do_populate_sysroot', 'virtual:native:/home/jenkins-slave/workspace/ci-meta/distro/ubuntu/label/docker-builder/target/tiogapass/openbmc/meta/recipes-devtools/patch/patch_2.7.6.bb:do_populate_sysroot', '/home/jenkins-slave/workspace/ci-meta/distro/ubuntu/label/docker-builder/target/tiogapass/openbmc/meta/recipes-devtools/libtool/libtool-cross_2.4.6.bb:do_populate_sysroot', '/home/jenkins-slave/workspace/ci-meta/distro/ubuntu/label/docker-builder/target/tiogapass/openbmc/meta/recipes-extended/pam/libpam_1.3.1.bb:do_populate_sysroot', 'virtual:native:/home/jenkins-slave/workspace/ci-meta/distro/ubuntu/label/docker-builder/target/tiogapass/openbmc/meta/recipes-devtools/automake/automake_1.16.2.bb:do_populate_sysroot', 'virtual:native:/home/jenkins-slave/workspace/ci-meta/distro/ubuntu/label/docker-builder/target/tiogapass/openbmc/meta/recipes-devtools/pkgconfig/pkgconfig_git.bb:do_populate_sysroot', '/home/jenkins-slave/workspace/ci-meta/distro/ubuntu/label/docker-builder/target/tiogapass/openbmc/meta/recipes-core/glibc/glibc_2.31.bb:do_populate_sysroot']
| NOTE: Installed into sysroot: []
| NOTE: Skipping as already exists in sysroot: ['gcc-cross-arm', 'autoconf-native', 'autoconf-archive-native', 'gcc-runtime', 'openssl', 'libtool-native', 'quilt-native', 'gnu-config-native', 'pseudo-native', 'patch-native', 'libtool-cross', 'libpam', 'automake-native', 'pkgconfig-native', 'glibc', 'gmp-native', 'xz-native', 'mpfr-native', 'linux-libc-headers', 'texinfo-dummy-native', 'binutils-cross-arm', 'libmpc-native', 'zlib-native', 'flex-native', 'm4-native', 'libgcc', 'opkg-utils', 'attr-native', 'flex', 'cracklib', 'libxcrypt', 'gettext-minimal-native', 'zlib']
| DEBUG: Python function extend_recipe_sysroot finished
| DEBUG: Executing shell function do_install
| NOTE: make -j40 DESTDIR=/tmp/openbmc/work/arm1176jzs-openbmc-linux-gnueabi/pam-ipmi/1.0+gitAUTOINC+f3919c432f-r1/image install
| make[1]: Entering directory '/tmp/openbmc/work/arm1176jzs-openbmc-linux-gnueabi/pam-ipmi/1.0+gitAUTOINC+f3919c432f-r1/build'
|  /tmp/openbmc/hosttools/mkdir -p '/tmp/openbmc/work/arm1176jzs-openbmc-linux-gnueabi/pam-ipmi/1.0+gitAUTOINC+f3919c432f-r1/image/etc'
|  /tmp/openbmc/hosttools/mkdir -p '/tmp/openbmc/work/arm1176jzs-openbmc-linux-gnueabi/pam-ipmi/1.0+gitAUTOINC+f3919c432f-r1/image/lib/security'
|  ./arm-openbmc-linux-gnueabi-libtool   --mode=install /tmp/openbmc/hosttools/install -c   pam_ipmicheck.la pam_ipmisave.la '/tmp/openbmc/work/arm1176jzs-openbmc-linux-gnueabi/pam-ipmi/1.0+gitAUTOINC+f3919c432f-r1/image/lib/security'
|  /tmp/openbmc/hosttools/install -c -m 644 ../git/key_file ../git/ipmi_pass '/tmp/openbmc/work/arm1176jzs-openbmc-linux-gnueabi/pam-ipmi/1.0+gitAUTOINC+f3919c432f-r1/image/etc'
| arm-openbmc-linux-gnueabi-libtool: install: /tmp/openbmc/hosttools/install -c .libs/pam_ipmicheck.so /tmp/openbmc/work/arm1176jzs-openbmc-linux-gnueabi/pam-ipmi/1.0+gitAUTOINC+f3919c432f-r1/image/lib/security/pam_ipmicheck.so
| arm-openbmc-linux-gnueabi-libtool: install: /tmp/openbmc/hosttools/install -c .libs/pam_ipmicheck.lai /tmp/openbmc/work/arm1176jzs-openbmc-linux-gnueabi/pam-ipmi/1.0+gitAUTOINC+f3919c432f-r1/image/lib/security/pam_ipmicheck.la
| arm-openbmc-linux-gnueabi-libtool: install: /tmp/openbmc/hosttools/install -c .libs/pam_ipmisave.so /tmp/openbmc/work/arm1176jzs-openbmc-linux-gnueabi/pam-ipmi/1.0+gitAUTOINC+f3919c432f-r1/image/lib/security/pam_ipmisave.so
| arm-openbmc-linux-gnueabi-libtool: install: /tmp/openbmc/hosttools/install -c .libs/pam_ipmisave.lai /tmp/openbmc/work/arm1176jzs-openbmc-linux-gnueabi/pam-ipmi/1.0+gitAUTOINC+f3919c432f-r1/image/lib/security/pam_ipmisave.la
| arm-openbmc-linux-gnueabi-libtool: warning: remember to run 'arm-openbmc-linux-gnueabi-libtool --finish /lib/security'
| make  install-data-hook
| make[2]: Entering directory '/tmp/openbmc/work/arm1176jzs-openbmc-linux-gnueabi/pam-ipmi/1.0+gitAUTOINC+f3919c432f-r1/build'
| chmod 0600 /tmp/openbmc/work/arm1176jzs-openbmc-linux-gnueabi/pam-ipmi/1.0+gitAUTOINC+f3919c432f-r1/image/etc/key_file
| chmod 0600 /tmp/openbmc/work/arm1176jzs-openbmc-linux-gnueabi/pam-ipmi/1.0+gitAUTOINC+f3919c432f-r1/image/etc/ipmi_pass
| chmod: cannot access '/tmp/openbmc/work/arm1176jzs-openbmc-linux-gnueabi/pam-ipmi/1.0+gitAUTOINC+f3919c432f-r1/image/etc/ipmi_pass': No such file or directory
| Makefile:947: recipe for target 'install-data-hook' failed
| make[2]: *** [install-data-hook] Error 1
| make[2]: Leaving directory '/tmp/openbmc/work/arm1176jzs-openbmc-linux-gnueabi/pam-ipmi/1.0+gitAUTOINC+f3919c432f-r1/build'
| Makefile:871: recipe for target 'install-data-am' failed
| make[1]: *** [install-data-am] Error 2
| make[1]: *** Waiting for unfinished jobs....
| make[1]: Leaving directory '/tmp/openbmc/work/arm1176jzs-openbmc-linux-gnueabi/pam-ipmi/1.0+gitAUTOINC+f3919c432f-r1/build'
| Makefile:816: recipe for target 'install-am' failed
| make: *** [install-am] Error 2
| ERROR: oe_runmake failed
| WARNING: exit code 1 from a shell command.
```

tomjoseph83, 2020-06-30:
@rthomaiy Any time run into this error?

rthomaiy, 2020-06-30:
No @tomjoseph83 I never faced such issue and i don't find any issue as well, i.e. When install-data-hook is called key_file & ipmi_pass is already in right place (No error seen for key_file right?). @vmauery have you seen any error on install-data-hook ?

geissonator, 2020-07-25:
Another upstream yocto bump and another recreate of this issue:
```
arm-openbmc-linux-gnueabi-libtool: install: /tmp/openbmc/hosttools/install -c .libs/pam_ipmicheck.so /tmp/openbmc/work/armv5e-openbmc-linux-gnueabi/pam-ipmi/1.0+gitAUTOINC+f3919c432f-r1/image/lib/security/pam_ipmicheck.so
arm-openbmc-linux-gnueabi-libtool: install: /tmp/openbmc/hosttools/install -c .libs/pam_ipmicheck.lai /tmp/openbmc/work/armv5e-openbmc-linux-gnueabi/pam-ipmi/1.0+gitAUTOINC+f3919c432f-r1/image/lib/security/pam_ipmicheck.la
arm-openbmc-linux-gnueabi-libtool: install: /tmp/openbmc/hosttools/install -c .libs/pam_ipmisave.so /tmp/openbmc/work/armv5e-openbmc-linux-gnueabi/pam-ipmi/1.0+gitAUTOINC+f3919c432f-r1/image/lib/security/pam_ipmisave.so
arm-openbmc-linux-gnueabi-libtool: install: /tmp/openbmc/hosttools/install -c .libs/pam_ipmisave.lai /tmp/openbmc/work/armv5e-openbmc-linux-gnueabi/pam-ipmi/1.0+gitAUTOINC+f3919c432f-r1/image/lib/security/pam_ipmisave.la
arm-openbmc-linux-gnueabi-libtool: warning: remember to run 'arm-openbmc-linux-gnueabi-libtool --finish /lib/security'
make  install-data-hook
make[2]: Entering directory '/tmp/openbmc/work/armv5e-openbmc-linux-gnueabi/pam-ipmi/1.0+gitAUTOINC+f3919c432f-r1/build'
chmod 0600 /tmp/openbmc/work/armv5e-openbmc-linux-gnueabi/pam-ipmi/1.0+gitAUTOINC+f3919c432f-r1/image/etc/key_file
chmod 0600 /tmp/openbmc/work/armv5e-openbmc-linux-gnueabi/pam-ipmi/1.0+gitAUTOINC+f3919c432f-r1/image/etc/ipmi_pass
chmod: cannot access '/tmp/openbmc/work/armv5e-openbmc-linux-gnueabi/pam-ipmi/1.0+gitAUTOINC+f3919c432f-r1/image/etc/ipmi_pass': No such file or directory
Makefile:947: recipe for target 'install-data-hook' failed
```

geissonator, 2020-11-12:
Still hitting this one intermittently.

geissonator, 2020-12-02:
Another yocto bump, another hit of this issue:

https://jenkins.openbmc.org/job/ci-meta/2005/distro=ubuntu,label=docker-builder,target=zaius/console

```
| make[1]: Entering directory '/home/jenkins-slave/workspace/ci-meta/distro/ubuntu/label/docker-builder/target/zaius/build/work/arm1176jzs-openbmc-linux-gnueabi/pam-ipmi/1.0+gitAUTOINC+f3919c432f-r1/build'
|  /home/jenkins-slave/workspace/ci-meta/distro/ubuntu/label/docker-builder/target/zaius/build/hosttools/mkdir -p '/home/jenkins-slave/workspace/ci-meta/distro/ubuntu/label/docker-builder/target/zaius/build/work/arm1176jzs-openbmc-linux-gnueabi/pam-ipmi/1.0+gitAUTOINC+f3919c432f-r1/image/etc'
|  /home/jenkins-slave/workspace/ci-meta/distro/ubuntu/label/docker-builder/target/zaius/build/hosttools/mkdir -p '/home/jenkins-slave/workspace/ci-meta/distro/ubuntu/label/docker-builder/target/zaius/build/work/arm1176jzs-openbmc-linux-gnueabi/pam-ipmi/1.0+gitAUTOINC+f3919c432f-r1/image/lib/security'
|  ./arm-openbmc-linux-gnueabi-libtool   --mode=install /home/jenkins-slave/workspace/ci-meta/distro/ubuntu/label/docker-builder/target/zaius/build/hosttools/install -c   pam_ipmicheck.la pam_ipmisave.la '/home/jenkins-slave/workspace/ci-meta/distro/ubuntu/label/docker-builder/target/zaius/build/work/arm1176jzs-openbmc-linux-gnueabi/pam-ipmi/1.0+gitAUTOINC+f3919c432f-r1/image/lib/security'
|  /home/jenkins-slave/workspace/ci-meta/distro/ubuntu/label/docker-builder/target/zaius/build/hosttools/install -c -m 644 ../git/key_file ../git/ipmi_pass '/home/jenkins-slave/workspace/ci-meta/distro/ubuntu/label/docker-builder/target/zaius/build/work/arm1176jzs-openbmc-linux-gnueabi/pam-ipmi/1.0+gitAUTOINC+f3919c432f-r1/image/etc'
| arm-openbmc-linux-gnueabi-libtool: install: /home/jenkins-slave/workspace/ci-meta/distro/ubuntu/label/docker-builder/target/zaius/build/hosttools/install -c .libs/pam_ipmicheck.so /home/jenkins-slave/workspace/ci-meta/distro/ubuntu/label/docker-builder/target/zaius/build/work/arm1176jzs-openbmc-linux-gnueabi/pam-ipmi/1.0+gitAUTOINC+f3919c432f-r1/image/lib/security/pam_ipmicheck.so
| arm-openbmc-linux-gnueabi-libtool: install: /home/jenkins-slave/workspace/ci-meta/distro/ubuntu/label/docker-builder/target/zaius/build/hosttools/install -c .libs/pam_ipmicheck.lai /home/jenkins-slave/workspace/ci-meta/distro/ubuntu/label/docker-builder/target/zaius/build/work/arm1176jzs-openbmc-linux-gnueabi/pam-ipmi/1.0+gitAUTOINC+f3919c432f-r1/image/lib/security/pam_ipmicheck.la
| arm-openbmc-linux-gnueabi-libtool: install: /home/jenkins-slave/workspace/ci-meta/distro/ubuntu/label/docker-builder/target/zaius/build/hosttools/install -c .libs/pam_ipmisave.so /home/jenkins-slave/workspace/ci-meta/distro/ubuntu/label/docker-builder/target/zaius/build/work/arm1176jzs-openbmc-linux-gnueabi/pam-ipmi/1.0+gitAUTOINC+f3919c432f-r1/image/lib/security/pam_ipmisave.so
| arm-openbmc-linux-gnueabi-libtool: install: /home/jenkins-slave/workspace/ci-meta/distro/ubuntu/label/docker-builder/target/zaius/build/hosttools/install -c .libs/pam_ipmisave.lai /home/jenkins-slave/workspace/ci-meta/distro/ubuntu/label/docker-builder/target/zaius/build/work/arm1176jzs-openbmc-linux-gnueabi/pam-ipmi/1.0+gitAUTOINC+f3919c432f-r1/image/lib/security/pam_ipmisave.la
| arm-openbmc-linux-gnueabi-libtool: warning: remember to run 'arm-openbmc-linux-gnueabi-libtool --finish /lib/security'
| make  install-data-hook
| make[2]: Entering directory '/home/jenkins-slave/workspace/ci-meta/distro/ubuntu/label/docker-builder/target/zaius/build/work/arm1176jzs-openbmc-linux-gnueabi/pam-ipmi/1.0+gitAUTOINC+f3919c432f-r1/build'
| chmod 0600 /home/jenkins-slave/workspace/ci-meta/distro/ubuntu/label/docker-builder/target/zaius/build/work/arm1176jzs-openbmc-linux-gnueabi/pam-ipmi/1.0+gitAUTOINC+f3919c432f-r1/image/etc/key_file
| chmod 0600 /home/jenkins-slave/workspace/ci-meta/distro/ubuntu/label/docker-builder/target/zaius/build/work/arm1176jzs-openbmc-linux-gnueabi/pam-ipmi/1.0+gitAUTOINC+f3919c432f-r1/image/etc/ipmi_pass
| chmod: cannot access '/home/jenkins-slave/workspace/ci-meta/distro/ubuntu/label/docker-builder/target/zaius/build/work/arm1176jzs-openbmc-linux-gnueabi/pam-ipmi/1.0+gitAUTOINC+f3919c432f-r1/image/etc/ipmi_pass': No such file or directory
```

williamspatrick, 2020-12-09:
I've pushed up a fix at https://gerrit.openbmc-project.xyz/c/openbmc/pam-ipmi/+/38913

closed 2020-12-09 by bradbishop